nextcloud/nextcloudpi #463: NCP recently stopped being accessible

cottoa00, 2018-03-28:
[NCP-report-0328-2018.txt](https://github.com/nextcloud/nextcloudpi/files/1857148/NCP-report-0328-2018.txt)

nachoparker, 2018-03-30:
If you activate SSH through ncp-web it will force you to change the password next time you log in. It is a security meassure, nothing to worry about.

Note that nc-passwd is the password for ncp-web, not for SSH

ncp-web doesn't use a database to store its settings.

nachoparker, 2018-03-30:
I'm reading again... do you have access to NC? do you have access to ncp-web? do you have ping or SSH access?

closed 2018-03-30 by nachoparker

cottoa00, 2018-04-02:
Sorry for delay... I can ping and can SSH in. That was how I get ncp-report output (which is very handy by the way). I cannot get to either the main ncp of the ncp-web (admin GUI)... for both of these I simply get a timeout. I tried coming in from outside (no-ip) and looking at the browser console I get a  500 Server Error instead of a timeout.

From the ncp-report output there is a line at the top that says "Nextcloud check    error" but all the services look to be up and running.

The next block shows a DB access error " Access denied for user 'ncadmin'@'localhost' "

The next block shows a repeating group of lines re: apache not reliably determining servers fqdn... and a couple more lines then a graceful restart?

DB block looks ok? 

[ncp-report-output.txt](https://github.com/nextcloud/nextcloudpi/files/1869273/ncp-report-output.txt)

nachoparker, 2018-04-03:
seems like the db password is wrong? let's get the output of

```
systemctl status nc-provisioning
```

cottoa00, 2018-04-03:
Here is output from above command;

 nc-provisioning.service - Randomize passwords on first boot
   Loaded: loaded (/usr/lib/systemd/system/nc-provisioning.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2018-04-02 14:52:57 EDT; 1 day 1h ago
  Process: 1612 ExecStart=/bin/bash /usr/local/bin/ncp-provisioning.sh (code=exited, status=0/SUCCESS)
 Main PID: 1612 (code=exited, status=0/SUCCESS)

Apr 02 14:52:57 ecnextcloudpi systemd[1]: Started Randomize passwords on first boot.
Apr 02 14:52:57 ecnextcloudpi bash[1612]: Updating NextCloud config with Redis password
Apr 02 14:52:57 ecnextcloudpi bash[1612]: Updating NextCloud config with MariaDB password

nachoparker, 2018-04-05:
what happens if you launch

```
systemctl start nc-provisioning
```
and try again

cottoa00, 2018-04-15:
Sorry for delay in responding... runningthe above and retrying did not change anything.
I'm now at the point where I cannot get at ncp-web (admin UI) without using the IP address.
https://ecnextcloudpi:4443 fails but https://192.168.1.151:4443 prompts for creds and gets in.
System Info still shows "Nextcloud Check" error and "NextCloud configuration" shows trusted domain info including internal and external domain names in addition to IP.

ncp-report output (attached) still shows db access failure for 'ncadmin'@'localhost'

[image]
[image]
[NCP-report-0415-2018.txt](https://github.com/nextcloud/nextcloudpi/files/1913468/NCP-report-0415-2018.txt)

cottoa00, 2018-04-16:
Update... got ncp-web working with domain name... just needed to update DNS table in router. So ncp-web is fine. Sill no change on ncp... ncp-report still shows DB failure... tried nc-provisioning again but no go. Think we're close?
[NCP-report-0416-2018.txt](https://github.com/nextcloud/nextcloudpi/files/1917225/NCP-report-0416-2018.txt)

cottoa00, 2018-04-17:
Update... have been digging thru any and all logs looking for something that stands out. 
Only thing that is consistent is DBfailure for access to ncadmin@localhost 
AND 
excerpt from ncp.log #52 {main} Error reading data directory. Is Nextcloud running and configured?
Sys Info says datadirectory is /media/USBdrive/ncdata.... 
but Maria 50-Server.cnf says datadir = /media/USBdrive/ncdatabase?
Tried to put config.php back to dbpassword='ownyourbits' and on reboot comes back to provisioned key listed above in "Nextcloud configuration"
Would like to start config from scratch... nc-init? 
Not sure what else to do....

nachoparker, 2018-04-18:
> If I run nc-init... will that leave my data alone and just reset my config? By data, I mean the files I've loaded, the comments on those files, my calendars and my contacts?

nc-init will remove everything you have.

For some reason, NC doesn't have your db password updated. Try comparing the value in `/root/.my.cnf` and the value in NC config ( the screenshot above )

cottoa00, 2018-04-18:
Looks like they match up... screenshot of .my.cnf attached. 
Thanks for helping me... don't want to let this get away from me.
[image]

nachoparker, 2018-04-19:
Well, the password is ok everywhere, so all that is left is to make sure it is also set in the database.

Try this

```
DBADMIN=ncadmin
DBPASSWD=$( grep password /root/.my.cnf | sed 's|password=||' )
mysql <<EOF
GRANT USAGE ON *.* TO '$DBADMIN'@'localhost' IDENTIFIED BY '$DBPASSWD';
DROP USER '$DBADMIN'@'localhost';
CREATE USER '$DBADMIN'@'localhost' IDENTIFIED BY '$DBPASSWD';
GRANT ALL PRIVILEGES ON nextcloud.* TO $DBADMIN@localhost;
EXIT
EOF
```

cottoa00, 2018-04-19:
Really getting interesting now... ran cmds above... try to access... NADA
BUT... when run ncp-report showed "Nextcloud check" ok! ... however, looks like datadir privs corrupt? (see attached ncp_report)... but no DB access failure!
To be sure tried to login at mysql with provisioned cred... got in successfully!
[image]

Checked datadir privs and got this;
[image]

[NCP-report-04-19-2018.txt](https://github.com/nextcloud/nextcloudpi/files/1929140/NCP-report-04-19-2018.txt)

nachoparker, 2018-04-20:
makes no sense... the password in config.php is the one NC uses to access the database

cottoa00, 2018-04-20:
I believe the DB access issue is now fixed... just having datadir access privelege issue as below excerpt form ncp-report shows....
----------------------------------------------
Cannot write into "config" directory!
This can usually be fixed by giving the webserver write access to the config directory
See https://docs.nextcloud.com/server/12/go.php?to=admin-dir_permissions
----------------------------------------------
Went to doc link but is just talking about using the config routine to setup datadir, privs, etc. 

I believe all I need now is the proper privs for www-data against datadir... 755? Is currently 750? I'm going to try adjusting datadir privs to see if can resolve. Not clear on relationship between 'ncdata' and 'ncdatabase'... seems like both need to be active... ncdata has launch point for database?

nachoparker, 2018-04-20:
[Permissions](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#what-userpermissions-should-i-have-to-the-external-usb-drive-mount-point-the-ncdata-and-ncdatabase-directory)

You are close. I don't know what happened to your system but you are running on a very old image. I suggest you fix this, then backup, then install latest image (wait a couple days, I am building one right now), and then restore on the latest image

cottoa00, 2018-04-20:
I believe I had one of the first images that I installed on the NCP "Box" that I got from Wester Digital. It was initially provisioned with Ubuntu Snap which I was not happy with Philosophically. When I found you that was what I was looking for. I seem to remeber having to make some manual tweaks to webserver config files to get ncp-web working so know I have an older "base". I run the updates but not sure how to transition to a moden base w/o losing my stuff.

cottoa00, 2018-04-20:
Bummer... checked your privs link and looks like my privs are what they are supposed to be... so not sure why I'm getting "config" directory access failure... need to dig up my notes on occ to see if I can get sorted

nachoparker, 2018-04-21:
nice! I am glad :)

I am generating the image right now and writing the announcement. Today or tomorrow will post it.

Keep tuned to ownyourbits ;)

cottoa00, 2018-04-24:
Got your new NCPlus image burnt onto a fresh SD card. Tested it in a PI that was connected to a monitor so I could do some prelim setup (SSH and Raspi creds). Got SSH setup and tried web access with IP... brought up activation screen. Saved creds in text file as well as printed activation.pdf.

Shutdown and moved SD to NextCloud Box... went to activation screen BUT did not notice creds had changed until I hit the activate button w/o saving/printing creds!

Is there a way to reset the activation process and start over or do I need to re-image the card and start fresh?

nachoparker, 2018-04-25:
You did a backup of your original instance through `nc-backup`, right?

cottoa00, 2018-04-25:
Yes I did nc-backup and nc-export-ncp  but output is saved on my USB drive which is unaccessible on a fresh start? Config wizard appears to go thru  nc-automount (to get at USBdrive) then nc-format-USB as ext4... obviously wiping my backup files.. so skipped config wizard just tried to access my backup on USBdrive... that's where got tangled up. 
Can ssh in and get to my old /media/USBdrive which apparently was a symlink to /media/ecCoudDrive... so all my stuff is still there... most importantly my backup (.tar) files (.bkup and .config). Will see if I can fit them onto a thumb drive so I can get at them from another fresh start.
Then I should be able to do nc-restore...

cottoa00, 2018-04-25:
My backup was around 50GB so no thumbdrive... was able to sftp them to a network share just in case.
Went back and re-imaged my card and started fresh... this time went into config wizard to do automount but skipped USB format (did not notice could skip before).
Then went to nc-import-ncp and was able to point at my old location /media/ecCloudDrive/ncp-config_20180424.tar which completed successfully (see attached log file)
Then went to nc-restore again pointing at my old location /media/ecCloudDrive/nextcloud-bkup_20180420.tar... took quite a while but ended with Redis exception (see attached log) and did not complete successfully.
[NCPlus-config-restore.txt](https://github.com/nextcloud/nextcloudpi/files/1948920/NCPlus-config-restore.txt)
[NCPlus-restore.txt](https://github.com/nextcloud/nextcloudpi/files/1948921/NCPlus-restore.txt)

Tried to  access but no go.

nachoparker, 2018-04-26:
mmm weird... the restore process updates your redis credentials. What is the output of `sudo ncp-report` right now? 

you are close

cottoa00, 2018-04-26:
Agreed... thanks for your help

[NCPlus-Report-04262018.txt](https://github.com/nextcloud/nextcloudpi/files/1951601/NCPlus-Report-04262018.txt)

nachoparker, 2018-04-27:
mmm compare the redis pasword in `/etc/redis/redis.conf` and `/var/www/nextcloud/config/config.php` (or ncp-web)

cottoa00, 2018-04-27:
Is password in redis.conf ... BUT no passwd in config.php.... just the following;
      'redis' => 
      array (
        'host' => '/var/run/redis/redis.sock',
        'port' => 0,
        'timeout' => 0.0,
      ),

nachoparker, 2018-04-27:
lol that's why the restore procedure didn't work... you didn't have that option for whatever reason.

Just add the line

```
'password' => , 'xxxxx',
```

where `xxxx` is the password from redis.conf

cottoa00, 2018-04-27:
Added redis pwd... then was stuck in maint mode. Used suggestion from shadowstreik from earlier issue to clear that... Now I'm good to go!

Thanks for hanging in there with me! ;-)

nachoparker, 2018-04-27:
awesome!

no probs

:D

cottoa00, 2018-04-28:
Bugger... have one last tweak... my letsencrypt cert seems to be broken (if it was ever active?)
I tried to check external access via my no-ip account ecnextcloud.ddns.net but fails on self-signed cert. 
I went to ncp-web and letsencrypt did not show a check mark (active)... but did have my correct domain (above) and my email. I reran... log is attached

So I went back to rerun no-ip in case it had expired?... it seemed to run thru... suggests I do have theat domain name... not sure whats tangled up?
[letsencrypt.log](https://github.com/nextcloud/nextcloudpi/files/1958077/letsencrypt.log)
[no-ip.log](https://github.com/nextcloud/nextcloudpi/files/1958078/no-ip.log)

nachoparker, 2018-04-29:
letsencrypt couldn't connect to your NCP. Probably port forwarding is not properly set up. You can check this with `nc-info`, `sudo ncp-report` or ncp-web. 

If you don't know how to do this, there's some info in the wiki

cottoa00, 2018-04-29:
nc-report shows both ports (80 & 443) open. Went to no-ip and used their port checking tool. Shows port 443 open from outside... not sure what else to do. Realize this is not an NCP issue... will work with involved third parties to sort out and post my results.
Cheers

[image]

nachoparker, 2018-04-29:
well, if your 80 port is open and your 443 port is open ( you can check from your phone, disconnected from wifi ) then it has to work, as long as you type the right domain to letsencrypt

cottoa00, 2018-04-30:
After some digging and testing have come to the following conclusion;
Running "letsencrypt" from ncp-web logs the following...
Domain: ecnextcloud.ddns.net
Type: connection
Detail: Fetching
http://ecnextcloud.ddns.net/.well-known/acme-challenge/s5KVzkA7g3ctlEHnGH8cdtn_U_KnpQmZ0FfMbrjwm6k:
Timeout during connect (likely firewall problem)

On digging into letsencrypt website suggests can install cert from shell access using certbot and that is already a part of Apache on Stretch but may need some "plugins"?
I tried "certbot certificates" and does not appear to be installed on my device. Not sure what your scripts are using but failure point is when acme challenge is trying to access http://ecnextcloud.ddns.net... my Firefox will not allow access since website forces https but there is no cert behind it... so chicken and egg?

Maybe I'll see if can turn https off until get cert installed?

nachoparker, 2018-05-01:
you have some networking problem somewhere... normally it's enough to open ports 80 and 443

don't disable https, it's better to accept security exceptions

cottoa00, 2018-05-03:
Doh!! All I needed was to forward port 80 on my router! Was under the false impression that should not open up port 80 since only using https (443). 
Can I shut it down now that cert is in place?

nachoparker, 2018-05-03:
haha there you go

> Can I shut it down now that cert is in place?

yessir

cottoa00, 2018-05-03:
Thinking might put on my calendar to open up again a day or two before cert due to expire. Will I be notified by NCP that cert has been updated?

nachoparker, 2018-05-03:
> Will I be notified by NCP that cert has been updated?

no, but actually that's a good idea! ;)

for now you have to check manually with the certbot tool

piamay, 2018-09-28:
> Well, the password is ok everywhere, so all that is left is to make sure it is also set in the database.
>
> Try this
>
> ```
> DBADMIN=ncadmin
> DBPASSWD=$( grep password /root/.my.cnf | sed 's|password=||' [...]

This saved my last weekend and my ncp is up again. A running watchtower container updated my ncp container to the last version and caused some trouble, like connection to database failed. So Thank you