nextcloud/nextcloudpi #464: no synchronization from laptop to iPad

user-jan, 2018-03-29:
Hi all,

I have recently updated my NCP to stable v. 13.0.1 and had no problems before.

For example, I work on a text document on my laptop (MacBook), and then save my work to NCP using the macOS Desktop Client. Then, I wish to continue working on that document (using a word processing app) on my iPad. The syncing info (e.g. 50 minutes ago) on the iOS Mobile client is there, but when I open the text document in the Mobile client on my iPad the document is not updated. However, when I add some text to the text document (using a wordprocessing app) on my iPad and save it to NCP, it is synchronized to my laptop's Desktop Client.

Any ideas what the problem could be?

Cheers,
Jan

Paste the output of `sudo ncp-report`

[ nc-info ]
Gathering information...
NextCloudPi version  v0.53.6
NextCloudPi image    NextCloudPi_11-12-17
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      4.3G/15G
rootfs usage         4.3G/15G
swapfile
Nextcloud check      ok
Nextcloud version    13.0.1.1
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.2
gateway              192.168.1.1
interface            eth0
certificates         xxxx
certs due            42 days 
NAT loopback         no
uptime               13min

You should enable dnsmasq to use your domain inside home

nachoparker, 2018-03-30:
There's some other people having trouble with iOS and NC13.0.1, I'll point you to the issue in the forums.

https://help.nextcloud.com/t/desktop-client-s-not-connecting-to-nextcloudpi/29663/

Doesn't seem specific to NCP

closed 2018-03-30 by nachoparker

user-jan, 2018-03-31:
Hi Nacho,
Thanks for your untiring and constructive support!!!
Please don’t forget to always make sure to get a good night’s sleep 😊
Best,
Jan

nachoparker, 2018-04-01:
Haha, will do, thanks!

take care

Tupi42, 2018-07-20:
Hi there, I had to make s fresh install of my NextCloudPi and I encountered the same problem described here on mac OSX and Linux : my desktop client can't connect anymore. Any updates on this issue ?

tks a lot;)