nextcloud/nextcloudpi #465: Docker image NAS data dir

crouthamela, 2018-03-30:
I am just starting out with Nextcloud, and am trying to get NextCloudPi running via Docker on my RaspberryPi. I have it running fine with the default setup, everything works great.

I have a home NAS running SMB/CIFS/NFS that I would like to mount as my default data dir. How do I accomplish this? I see there is the "External storages" app, but that puts it in a separate area. I want my users to dump their data onto my large NAS by default (the "Files" area). On a normal Linux system I would just set up the mount in my fstab. I'm not sure how best to go about this with the Docker image.

I would greatly appreciate any help on this.

crouthamela, 2018-03-30:
It looks like I need to mount it to the host via /etc/fstab as I am used to, but I need to pass it via --volume to the docker container. What is the path to the Nextcloud data dir? And is there anything I need to do besides define that path? Will it just override the existing default directory like a normal mount command would do?

carlhung, 2018-03-30:
he’s suggested to move whole docker to the external storage. Also how I did:
https://ownyourbits.com/2017/06/08/nextcloudpi-docker-for-raspberry-pi/

carlhung, 2018-03-30:
And if you mount on external device like usb HDD, etc.On my ubuntu x86, I have found that the docker will be executed before the system automatically mounts your usb storage(if your system will automatically mounts everytime it reboots.). Therefore, you have to disable docker auto-run.

nachoparker, 2018-03-30:
You can also do [this](https://github.com/nextcloud/nextcloudpi/issues/437#issuecomment-373059376)

closed 2018-03-30 by nachoparker

crouthamela, 2018-03-30:
@nachoparker Thank you, that is what I was looking for! :)