nextcloud/nextcloudpi #467: MariaDB Issue At Boot

robbiemacg, 2018-04-01:
I've succssfully run/maintained an ownCloud instance for several years. Recently, I've been experimenting with NextCloud on a RaspberryPi Zero W with some external storage attached. 

So far so good… until recently,  MariaDB is no longer starting correctlly at boot. When I check the service with 'systemctl status mysql' I see the following output:

```
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Thu 2018-03-22 13:28:49 UTC; 8min ago
  Process: 781 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=exited, status=1/FAILURE)
  Process: 705 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 435 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP_S
  Process: 397 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 364 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 781 (code=exited, status=1/FAILURE)
   Status: "MariaDB server is down"

Mar 22 13:28:27 nextcloudpi systemd[1]: Starting MariaDB database server...
Mar 22 13:28:49 nextcloudpi mysqld[781]: 2018-03-22 13:28:49 3062385376 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 781 ...
Mar 22 13:28:49 nextcloudpi mysqld[781]: 2018-03-22 13:28:49 3062385376 [Warning] Can't create test file /media/InvisibleDrive/ncdatabase/nextcloudpi.lower-test
Mar 22 13:28:49 nextcloudpi mysqld[781]: [108B blob data]
Mar 22 13:28:49 nextcloudpi mysqld[781]: 2018-03-22 13:28:49 3062385376 [ERROR] Aborting
Mar 22 13:28:49 nextcloudpi systemd[1]: mariadb.service: Main process exited, code=exited, status=1/FAILURE
Mar 22 13:28:49 nextcloudpi systemd[1]: Failed to start MariaDB database server.
Mar 22 13:28:49 nextcloudpi systemd[1]: mariadb.service: Unit entered failed state.
Mar 22 13:28:49 nextcloudpi systemd[1]: mariadb.service: Failed with result 'exit-code'.
lines 1-20/20 (END)
```

I'm able to start the service without issue from the command line:

```
sudo systemctl start mysql.service

systemctl status mysql
● mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: active (running) since Thu 2018-03-22 13:25:01 UTC; 17s ago
  Process: 17392 ExecStartPost=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 17390 ExecStartPost=/etc/mysql/debian-start (code=exited, status=0/SUCCESS)
  Process: 17321 ExecStartPre=/bin/sleep 10 (code=exited, status=0/SUCCESS)
  Process: 17223 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0 ]   && systemctl set-environment _WSREP
  Process: 17219 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 17215 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 17352 (mysqld)
   Status: "Taking your SQL requests now..."
   CGroup: /system.slice/mariadb.service
           └─17352 /usr/sbin/mysqld

Mar 22 13:24:46 nextcloudpi systemd[1]: Starting MariaDB database server...
Mar 22 13:24:58 nextcloudpi mysqld[17352]: 2018-03-22 13:24:58 3062459104 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1) starting as process 17352 ...
Mar 22 13:25:01 nextcloudpi systemd[1]: Started MariaDB database server.
lines 1-17/17 (END)
```

Any thoughts on what's happening and how to remedy the issue would be very welcome. This little machine isn't on all the time, and I'd rather not b SSHing in to start the DB manually everytime it boots. Only recent change to teh system is that I installed emacs, which seems unlikely to cause a problem like this in my mind.

Nextcloud version : 12.0.4.3
Operating system and version : Linux nextcloudpi 4.9.66+ (So, basically up to date Raspbian)
Apache or nginx version : Apache/2.4.25 
PHP version : PHP 7.0.27-0+deb9u1

config.php file:

```
<?php
$CONFIG = array (
  'passwordsalt' => 'xxxxxxx',
  'secret' => 'xxxxxxx',
  'trusted_domains' => 
  array (
[... 152 lines cut ...]
```

{"reqId":"WrvPXX8AAQEAAEm9LvIAAABK","level":2,"time":"2018-03-28T17:22:37+00:00","remoteAddr":"5.188.9.100","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"5.188.9.100\" tried to access using \"24.138.57.199\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"Wry2Rn8AAQEAAFas5JcAAAAB","level":2,"time":"2018-03-29T09:47:50+00:00","remoteAddr":"119.241.130.27","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"119.241.130.27\" tried to access using \"24.138.57.199\" as host.","userAgent":"curl\/7.43.0","version":"12.0.4.3"}
{"reqId":"Wry9Y38AAQEAAFatNjAAAABA","level":2,"time":"2018-03-29T10:18:12+00:00","remoteAddr":"77.72.85.108","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"77.72.85.108\" tried to access using \"24.138.57.199\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"Wrz3HX8AAQEAAFas5M4AAAAI","level":2,"time":"2018-03-29T14:24:29+00:00","remoteAddr":"60.191.38.78","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"60.191.38.78\" tried to access using \"24.138.57.199:443\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.11; rv:47.0) Gecko\/20100101 Firefox\/47.0","version":"12.0.4.3"}
{"reqId":"Wr0BN38AAQEAAFas5M8AAAAK","level":2,"time":"2018-03-29T15:07:36+00:00","remoteAddr":"184.105.139.70","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.139.70\" tried to access using \"24.138.57.199\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"Wr1e2n8AAQEAAFatN08AAABQ","level":2,"time":"2018-03-29T21:47:06+00:00","remoteAddr":"71.6.146.130","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"71.6.146.130\" tried to access using \"24.138.57.199\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/34.0.1847.137 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"Wr1e9H8AAQEAAFatN1AAAABV","level":3,"time":"2018-03-29T21:47:32+00:00","remoteAddr":"71.6.146.130","user":"--","app":"PHP","method":"GET","url":"\/sitemap.xml","message":"Exception: The requested uri(\/sitemap.xml) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"--","version":"12.0.4.3"}
{"reqId":"Wr1e938AAQEAAFatN1EAAABW","level":3,"time":"2018-03-29T21:47:35+00:00","remoteAddr":"71.6.146.130","user":"--","app":"PHP","method":"GET","url":"\/.well-known\/security.txt","message":"Exception: The requested uri(\/.well-known\/security.txt) cannot be processed by the script '\/core\/templates\/404.php') at \/var\/www\/nextcloud\/lib\/private\/AppFramework\/Http\/Request.php#729","userAgent":"--","version":"12.0.4.3"}
{"reqId":"Wr3S4n8AAQEAAFas5lAAAAAK","level":2,"time":"2018-03-30T06:02:10+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"24.138.57.199\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"Wr4Rln8AAQEAAGoEZB0AAAAA","level":2,"time":"2018-03-30T10:29:42+00:00","remoteAddr":"164.132.91.13","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"164.132.91.13\" tried to access using \"24.138.57.199\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 6.1) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/41.0.2228.0 Safari\/537.36","version":"12.0.4.3"}
{"reqId":"Wr4u4H8AAQEAAGoFV7YAAABB","level":2,"time":"2018-03-30T12:34:41+00:00","remoteAddr":"216.218.206.67","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"216.218.206.67\" tried to access using \"24.138.57.199\" as host.","userAgent":"--","version":"12.0.4.3"}
[... cut ...] ng \"24.138.57.199\" as host.","userAgent":"--","version":"12.0.4.3"}
{"reqId":"WsEYkn8AAQEAABPmdYsAAAAG","level":2,"time":"2018-04-01T17:36:18+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"24.138.57.199\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"12.0.4.3"}

```
</details>

nachoparker, 2018-04-05:
> I installed emacs,

I refuse to help an emacs user!! Just kiddin ;P

Maybe it has to do with the timing of automount, being the zero very slow. I would first try to add it manually to fstab

nachoparker, 2018-04-05:
Welcome!

Well, I was thinking that if mounting with fstab works, then it's likely a timing issue.

If that's the case, we can try a couple things if you are up to it

nachoparker, 2018-04-09:
;) cool, I am glad