nextcloud/nextcloudpi #470: Preview Generator App Feedback on NCP @ Rpi2

Iolaum, 2018-04-07:
I 've enabled the [Preview Generator App](https://apps.nextcloud.com/apps/previewgenerator) and have been testing it on my raspberry pi 2 NCP installation.

**What's the problem?**

Generating preview images is too hard from raspberry pi's and therefore performance downgrades when viewing a folder with images.

**Possible solution:**
Installing the Preview Generator on the raspberry pi allows the creation of those preview images ahead of time so the rpi doesn't have to compute them when you browse.

Unfortunately depending on your usage this may not be straightforward. On a load of about 135 Gb of photos my rpi2 has taken two weeks to generate around 2/3 of previews needed ... :( 

This issue is about exploring the issue, considering how we can enable Preview Generator on NCP and some tips and tricks on people who want to enable it on their own.

Steps to enable Preview Generator  on your own.

1. Go to the apps page of your local NextCloud instance and enable Preview Generator.
2. Log into your your machine hosting your NCP instance, preferable through SSH.
3. The commands available from the app after it is installed are listed in the [projects README page](https://github.com/rullzer/previewgenerator/#commands). The first things you need to do is generate a preview of all existing images and the set up a periodic run to generate previews for new images.

Because the generation of the preview images takes ages !! you need to to it in a slightly special way by using the `screen` command. It enables you to create a session that will continue to run even after you log out - because you don't want to have to stay logged in for 1 week/50GbPhotos. 

More specifically for the raspberry pi image of NextCloudPi:

```
# become root user
$ sudo -i
# create a new screen 
$ screen
# Click enter if needed after the interactive information that pops up
# Go to the Nextcloud directory
$ cd /var/www/nextcloud
# Start the preview generation process:
sudo -u www-data php ./occ preview:generate-all
# Get out of the screen session:
Ctrl+A+D # Just hit those 3 keys together and you get detached from the screen
```

Now your rpi has started creating the preview images you may want to see what's it doing. Here's how to do it:

```
# First log in and get a terminal up
# Use top to see the running process
$ top # php having 100% usage is normal
# become root
$ sudo -i
# See available screen sessions
$ screen -list
# Assuming you have only one screen session running, connect to it
$ screen -r
# You can now see how fat the generate-all command has proceeded.
Ctrl+a+d # detach from session to let it keep running
# Lastly you may want to monitor your rpi temperature - result are in mC
$ cat /sys/class/thermal/thermal_zone0/temp
46002 # This means 46 degrees Celcius
```

Potential problems:
 - If your rpi2 reboots expect the process to terminate abruptly without having completed.
- Auto updates for ncp may fail (not always) during this process.

I haven't decided on the optimum frequency of new photos preview generation. It depends on your usage. I am thinking 1/week to ensure that a new preview doesn't start while an old is running. That may not be a problem however. I need to investigate more.

Feedback from people trying this is welcome. We are looking on enabling the app on a later update of NCP and we would like to find a way to do it with as little bad experiences as possible.

nachoparker, 2018-04-10:
thanks for this @Iolaum! I think that @nero120 will also be interested. He did some nice testing in #406

> The plugin already checks for existing previews and only generates missing ones. In that case we could set up the (quick and dirty) solution to kill the process after some time (e. g. by a second cron job a few hours after starting it).

I also have doubts about this point. If I understand correctly, you need to do the complete pregeneration one before it can be run "incrementally". Maybe @rullzer can clarify this.

In any case, the best solution would be to get the desired behaviour / features in the app itself

> I haven't decided on the optimum frequency of new photos preview generation.

IIRC it won't overlap, so it won't run if it detects that it is already running.

rullzer, 2018-04-10:
The app can only 'incrementally' generate previews since you enabled to app.

So if you also want generated previews from files uploaded before that you need to run the whole batch.

The app is basically just requesting previews for the images it encounters. So the preview generation is still done in the server part. So if a preview exists it will not regenerate anything.

nachoparker, 2018-04-10:
@rullzer good, so if the pre-generation (of the pics that were there before the app started) is interrupted and resumed, it doesn't start from zero, right? at least that's what I thought I saw when I tried it.

btw, thanks for the app ;)

rullzer, 2018-04-10:
It does start from zero because there is no order in that sense. But the images it encounters that have already previews are basically done instantly ;)

nachoparker, 2018-04-13:
I have been playing around with this. On my pi3, the first generation is taking a long time for 80GB of pictures. Probably it will take several days to finish.

Only one core is being used though, so the system is still completely usable and almost as responsive.

If the process is interrupted, it takes a while to get to where it was but it doesn't actually generate so it doesn't start from zero. It takes some time though to check all the previously processed files.

I wonder why your pi2 is so sluggish...

nachoparker, 2018-04-30:
There's some interesting benchmarking and tunning in [this thread](https://github.com/rullzer/previewgenerator/issues/107#issuecomment-380618890) for those that are interested

nachoparker, 2018-10-27:
let's move discussion to [this PR](https://github.com/nextcloud/nextcloudpi/pull/662)

closed 2018-10-27 by nachoparker