nextcloud/nextcloudpi #471: Cloudflare support

Haones, 2018-04-07:
First I want to say, you did great work, installation was very easy and nice :) Now using this on my pretty new Pi3B+. I want to suggest one enhancement. 

Add a support for Cloudflare to NCP-Admin dashboard. I did installation manually (it still shows me in system info that I don't have any certificates, but it working fine with cloudflare), but would be great to see this an option. May be it will be better to integrate all methods to get free SSL by putting all types of free SSL services (let's encrypt, no-ip and others) in one module, something like 'free-ssl'.

nachoparker, 2018-04-09:
what is cloudflare exactly? I thought it was some ISP DDOS attack protection

theCalcaholic, 2018-04-09:
Cloud flare is not related to ISPs. If I'm not mistaken it is some kind of CDN/content cache with DDOS protection but also provides some kind of 'middleware' page which checks a client (and judges whether it might be a bot) and forwards the request to the target site if the client passes the sanity check.

nachoparker, 2018-04-09:
> Cloud flare is not related to ISPs. If I'm not mistaken it is some kind of CDN/content cache with DDOS protection but also provides some kind of 'middleware' page which checks a client (and judges [...]

Yes, that's what I thought, even though I didn't express it very well. I don't see how it fits here...

Haones, 2018-04-09:
> Yes, that's what I thought, even though I didn't express it very well. I don't see how it fits here...

DDoS protection is just additional feature. Main purpose of this enhancement is an ability to get free root SSL certificate from Cloudflare. You can get it for up to 15 years. It's just an additional way to get SSL certificate, like from No-IP or Let's encrypt.

nachoparker, 2018-04-10:
Sadly, we already have LE, which works perfectly, and we don't have a lot of manpower to implement new things, so we have to be selective.

That being said, if anyone really wants this, PRs are welcome ;)

My personal problem with cloudflare is that it gets constantly in my way when browsing with Tor. It is really annoying, but hey, this is about choice.

Again, thanks for the feedback!

closed 2018-04-10 by nachoparker