nextcloud/nextcloudpi #474: Error 500 - mysqld won't start

kevquirk, 2018-04-09:
So my NCP instance went down whilst I was in work today. I've done some troubleshooting, and it seems that mysql won't start up.

It seems that when the Pi boots, mysql will no longer start. When I try, I get the following error:

$ sudo /etc/init.d/mysql start
Starting mysql (via systemctl): mysql.serviceJob for mariadb.service failed because a fatal signal was delivered to the control process.
See "systemctl status mariadb.service" and "journalctl -xe" for details.
 failed!

**Output of "systemctl status mariadb.service":**
mariadb.service - MariaDB database server
   Loaded: loaded (/lib/systemd/system/mariadb.service; enabled; vendor preset: enabled)
   Active: activating (start-pre) since Mon 2018-04-09 17:38:49 UTC; 8s ago
  Process: 3869 ExecStart=/usr/sbin/mysqld $MYSQLD_OPTS $_WSREP_NEW_CLUSTER $_WSREP_START_POSITION (code=killed, signal=ABRT)
  Process: 3902 ExecStartPre=/bin/sh -c [ ! -e /usr/bin/galera_recovery ] && VAR= ||   VAR=`/usr/bin/galera_recovery`; [ $? -eq 0
  Process: 3898 ExecStartPre=/bin/sh -c systemctl unset-environment _WSREP_START_POSITION (code=exited, status=0/SUCCESS)
  Process: 3895 ExecStartPre=/usr/bin/install -m 755 -o mysql -g root -d /var/run/mysqld (code=exited, status=0/SUCCESS)
 Main PID: 3869 (code=killed, signal=ABRT); Control PID: 4000 (sleep)
   CGroup: /system.slice/mariadb.service
           â””â”€control
             â””â”€4000 /bin/sleep 20

Apr 09 17:38:49 nextcloudpi systemd[1]: Starting MariaDB database server...

**Output of "journalctl -xe":**
-- Unit mariadb.service has begun starting up.
Apr 09 17:39:20 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:20 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information
Apr 09 17:39:23 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:23 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information
Apr 09 17:39:28 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:28 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information
Apr 09 17:39:34 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:34 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information
Apr 09 17:39:38 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:38 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information
Apr 09 17:39:38 nextcloudpi mysqld[4257]: 2018-04-09 17:39:38 1988325376 [Note] /usr/sbin/mysqld (mysqld 10.1.23-MariaDB-9+deb9u1
Apr 09 17:39:39 nextcloudpi systemd[1]: mariadb.service: Main process exited, code=killed, status=6/ABRT
Apr 09 17:39:39 nextcloudpi systemd[1]: Failed to start MariaDB database server.
-- Subject: Unit mariadb.service has failed
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit mariadb.service has failed.
-- 
-- The result is failed.
Apr 09 17:39:39 nextcloudpi systemd[1]: mariadb.service: Unit entered failed state.
Apr 09 17:39:39 nextcloudpi systemd[1]: mariadb.service: Failed with result 'signal'.
Apr 09 17:39:44 nextcloudpi systemd[1]: mariadb.service: Service hold-off time over, scheduling restart.
Apr 09 17:39:44 nextcloudpi systemd[1]: Stopped MariaDB database server.
-- Subject: Unit mariadb.service has finished shutting down
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit mariadb.service has finished shutting down.
Apr 09 17:39:44 nextcloudpi systemd[1]: Starting MariaDB database server...
-- Subject: Unit mariadb.service has begun start-up
-- Defined-By: systemd
-- Support: https://www.debian.org/support
-- 
-- Unit mariadb.service has begun starting up.
Apr 09 17:39:48 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:48 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information
Apr 09 17:39:55 nextcloudpi dhcpcd[621]: eth0: Router Advertisement from fe80::ee08:6bff:fee7:d602
Apr 09 17:39:55 nextcloudpi dhcpcd[621]: eth0: requesting DHCPv6 information

nachoparker, 2018-04-10:
what's the output of `sudo ncp-report`?

kevquirk, 2018-04-10:
pi@nextcloudpi:~$ sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.53.27
NextCloudPi image    NextCloudPi_03-18-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      14G/112G
rootfs usage         1.9G/29G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      down
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.0.3
gateway              192.168.0.1
interface            eth0
certificates         carrot.kevq.uk
NAT loopback         no
uptime               23:36

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] No such file or directory in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(628): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1511): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(332): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#18 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\\Authenticati...')
#19 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#20 /var/www/nextcloud/lib/private/Server.php(336): OC\ServerContainer->query('OC\\Authenticati...')
#21 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#22 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\\Authenticati...')
#23 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
[... 76 lines cut ...]
```

{"reqId":"Wsqas38AAQEAAAO@d1IAAABY","level":2,"time":"2018-04-08T22:41:55+00:00","remoteAddr":"77.72.85.108","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"77.72.85.108\" tried to access using \"212.159.22.187\" as host.","userAgent":"--","version":"13.0.1.1"}
{"reqId":"Wsqci38AAQEAAAO9zQoAAAAX","level":2,"time":"2018-04-08T22:49:47+00:00","remoteAddr":"77.72.85.108","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"77.72.85.108\" tried to access using \"212.159.22.187\" as host.","userAgent":"--","version":"13.0.1.1"}
{"reqId":"WsqgfH8AAQEAAAO91QcAAAAQ","level":2,"time":"2018-04-08T23:06:36+00:00","remoteAddr":"77.72.85.108","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"77.72.85.108\" tried to access using \"212.159.22.187\" as host.","userAgent":"--","version":"13.0.1.1"}
[... cut ...] : Sabre\\\\DAV\\\\Server->invokeMethod(Object(Sabre\\\\HTTP\\\\Request), Object(Sabre\\\\HTTP\\\\Response))\\n#6 \\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/appinfo\\\/v1\\\/webdav.php(80): Sabre\\\\DAV\\\\Server->exec()\\n#7 \\\/var\\\/www\\\/nextcloud\\\/remote.php(164): require_once('\\\/var\\\/www\\\/nextcl...')\\n#8 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/apps\\\/dav\\\/lib\\\/Connector\\\/Sabre\\\/Auth.php\",\"Line\":163}","userAgent":"--","version":"13.0.1.1"}

```
</details>

nachoparker, 2018-04-11:
The database crashed... not looking good.

There is a big chance of SD card corruption. Check the NCP wiki FAQ on how to test this.

Maybe there's more detail in `/var/log/mysql/error.log`

closed 2018-04-11 by nachoparker

kevquirk, 2018-04-11:
Thought that may have been the answer. Oh well, at least it's easy to reinstall. Note to self - keep the DB on the SD. :)

Thanks for the help.

nachoparker, 2018-04-11:
no problem :)

One clarification... it is not that the database itself is corrupt, but the database binaries, this is independent of you moving the database or not.

kevquirk, 2018-04-11:
What would have caused that then? Is it something that I can prevent in the future?

nachoparker, 2018-04-11:
Good quality power supply is the most important thing

https://github.com/nextcloud/nextcloudpi/wiki/FAQ#why-is-nextcloud-so-slow--failing-randomly--getting-server-errors-even-with-a-fresh-image

kevquirk, 2018-04-11:
Thanks. I have the official Raspberry Pi one. However, it's very low down, so maybe it was knocked or something. I'll move it.

Thanks for all the help, really do appreciate it.

K