nextcloud/nextcloudpi #476: Port Forwarding no longer working

rickswan, 2018-04-11:
After returning home from a weekend trip, I plugged my NCP back in and port forwarding no longer worked.  DHCP assigned a different IP than before I went away.  I rebuilt NCP from a clean Stretch build and using curl to get the NCP components. While using the installation wizard things die in the letsencrypt.  Everything else seems to be configured and working fine with the exception of Port Forwarding.  It looks like the port mapping array is somehow incomplete or corrupt, but not being a networking guy I'm not entirely sure where that file lives.  It looks like it still wants to map to an older IP.55 rather than the current IP.53.  Below are the results from nc-forward-ports as well as a few things directly from miniupnp.  I've masked a few parts for privacy but they do match (are correct).  Any help would be great.

Thanks ... and thanks again for all your work on this.

______________________________________________________________________________________
Launching nc-forward-ports
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:service:Layer3Forwarding:1

 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.7.1:1900/igd.xml
 st: uuid:fcdb9233-a63f-41da-b42c-7cfeb99c8adf

 desc: http://192.168.7.1:1900/igd.xml
 st: upnp:rootdevice

Found valid IGD : http://192.168.7.1:1900/ipc
Local LAN ip address : 192.168.7.53
UPNP_DeletePortMapping() returned : 714
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:service:Layer3Forwarding:1

 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.7.1:1900/igd.xml
 st: uuid:fcdb9233-a63f-41da-b42c-7cfeb99c8adf

 desc: http://192.168.7.1:1900/igd.xml
 st: upnp:rootdevice

Found valid IGD : http://192.168.7.1:1900/ipc
Local LAN ip address : 192.168.7.53
UPNP_DeletePortMapping() returned : 714
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:service:Layer3Forwarding:1

 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.7.1:1900/igd.xml
 st: uuid:fcdb9233-a63f-41da-b42c-7cfeb99c8adf

 desc: http://192.168.7.1:1900/igd.xml
 st: upnp:rootdevice

Found valid IGD : http://192.168.7.1:1900/ipc
Local LAN ip address : 192.168.7.53
ExternalIPAddress = **Masked Ext IP x.x.x.92 **
AddPortMapping(443, 443, 192.168.7.53) failed with code 718 (ConflictInMappingEntry)
InternalIP:Port = 192.168.7.55:443
external 100.15.190.92:443 TCP is redirected to internal 192.168.7.55:443 (duration=0)
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:service:Layer3Forwarding:1

 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.7.1:1900/igd.xml
 st: uuid:fcdb9233-a63f-41da-b42c-7cfeb99c8adf

 desc: http://192.168.7.1:1900/igd.xml
 st: upnp:rootdevice

Found valid IGD : http://192.168.7.1:1900/ipc
Local LAN ip address : 192.168.7.53
ExternalIPAddress = **Masked Ext IP x.x.x.92 **
AddPortMapping(80, 80, 192.168.7.53) failed with code 718 (ConflictInMappingEntry)
InternalIP:Port = 192.168.7.55:80
external 100.15.190.92:80 TCP is redirected to internal 192.168.7.55:80 (duration=0)
Done. Press any key...

pi@nextcloudpi:~ $ curl -ks https://testcloud.MASKED.com
pi@nextcloudpi:~ $ nslookup testcloud.MASKED.com
Server:		127.0.0.1
Address:	127.0.0.1#53

Non-authoritative answer:
Name:	testcloud.MASKED.com
Address: **Masked Ext IP x.x.x.92 **

pi@nextcloudpi:~ $ curl -ks http://checkip.dyndns.org
<html><head><title>Current IP Check</title></head><body>Current IP Address: **Masked Ext IP x.x.x.92 **</body></html>

pi@nextcloudpi:~ $ upnpc -l
upnpc : miniupnpc library test client. (c) 2005-2014 Thomas Bernard
Go to http://miniupnp.free.fr/ or http://miniupnp.tuxfamily.org/
for more information.
List of UPNP devices found on the network :
 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:service:Layer3Forwarding:1

 desc: http://192.168.7.1:1900/igd.xml
 st: urn:schemas-upnp-org:device:InternetGatewayDevice:1

 desc: http://192.168.7.1:1900/igd.xml
 st: uuid:fcdb9233-a63f-41da-b42c-7cfeb99c8adf

 desc: http://192.168.7.1:1900/igd.xml
 st: upnp:rootdevice

Found valid IGD : http://192.168.7.1:1900/ipc
Local LAN ip address : 192.168.7.53
Connection Type : IP_Routed
Status : Connected, uptime=245208s, LastConnectionError : ERROR_NONE
  Time started : Sun Apr  8 18:10:19 2018
GetLinkLayerMaxBitRates failed.
ExternalIPAddress = **Masked Ext IP x.x.x.92 **
 i protocol exPort->inAddr:inPort description remoteHost leaseTime
 0 TCP    80->192.168.7.55:80    'user port mapping' '' 0
 1 UDP  4500->192.168.7.52:4500  'user port mapping' '' 0
 2 TCP   443->192.168.7.55:443   'user port mapping' '' 0
GetGenericPortMappingEntry() returned 713 (SpecifiedArrayIndexInvalid)
pi@nextcloudpi:~ $

nachoparker, 2018-04-11:
Port forwarding is in your router to configure. As a quick thing to try, why don't you use a static IP? you can set it from ncp-web or ncp-config

rickswan, 2018-04-11:
It seemed logical that the mapping configuration is on the router, but I can't find anything that talks about it.  I kind of wanted to have things working with DHCP rather than a static IP.  Wondering if I need to just factory reset my EERO and try again.  Oh well, thank you much.

closed 2018-04-12 by nachoparker