nextcloud/nextcloudpi #477: Many "locale" errors in install.sh when run on armbian

Tanarri, 2018-04-12:
Hi,

I installed a cubietruck (https://www.armbian.com/cubietruck/) and a clean armbian Debian 9 (https://dl.armbian.com/cubietruck/Debian_stretch_next.7z) on an SD.

Then I downloaded and started "install.sh" to install nextcloudPi as recommended here: https://github.com/nextcloud/nextcloudpi/issues/442 with the command
```
sudo curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash
```

The installation started and everything went well - except for the many warnings in the installation process via LANGUAGE, LC_ALL:
```
Get:13 http://cdn-fastly.deb.debian.org/debian stretch/main armhf python3-virtualenv all 15.1.0+ds-1[57.7 kB][German only
Get:14 http://cdn-fastly.deb.debian.org/debian stretch/main armhf virtualenv all 15.1.0+ds-1[19.7 kB]
fetched 30.8 MB in 13s (2289 kB/s)                                
Perl: Warning: Locale setting failed.                             
perl: Warning: Please check if your locale settings are correct:                                                                               
        LANGUAGE = (not set),                                       
        LC_ALL = (not set),                                         
        LANG ="en_EN.UTF-8".                                      
    are supported and installed on your system.                   
perl: Warning: Use the default locale ("C").         
locale: LC_CTYPE cannot be set to default locale: No such file or directory
locale: LC_MESSAGES cannot be set to default locale: No such file or directory
locale: LC_ALL cannot be set to default locale: No such file or directory**
Selection of previously unselected Augeas lenses.                                                                               
(Read database ..... 36330 files and directories currently installed.)
Preparation for unpacking .../00-augeas-lenses_1.8.0-1+deb9u1_all.deb ....
Unpacking Augeas Lenses (1.8.0-1+deb9u1) .... 
```

Additional information:
```
env | grep LANG
LANG=en_EN.UTF-8
```

It would be cool if the installation process sets a correct LANGUAGE, LC_ALL, LANG and the warning disappears.

regards
TanarRi

nachoparker, 2018-04-13:
Hi, 

I just tested on a clean image and didn't get this. Did you change locale settings?

I don't know if changing the locales for users without telling them would be a good idea, they might get mad at us :P

closed 2018-04-13 by nachoparker

Tanarri, 2018-04-13:
I got my failure...
I did a fresh install of armbian on my cubietruck for 2 times to find the nasty locale error and got it (but I don't know why)

A fresh installed armbian 5.38 Debian stretch has the correct locale settings
```
root@cubietruck:~# locale
LANG=en_US.UTF-8
LANGUAGE=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES=en_US.UTF-8
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

But then I changed the SSH config:
```
vim /etc/ssh/sshd_config
# old
#UsePAM yes
# new setting!
UsePAM no
service ssh restart
```

Now I logged in new with SSH _from my kubuntu desktop (LANG=de_DE.UTF-8)_ to my armbian cubietruck... and there where the error in the locale at armbian
```
root@cubietruck:~# locale        
locale: Cannot set LC_CTYPE to default locale: No such file or directory                                                             
locale: Cannot set LC_MESSAGES to default locale: No such file or directory                                                          
locale: Cannot set LC_ALL to default locale: No such file or directory                                                               
LANG=de_DE.UTF-8                 
LANGUAGE=                        
LC_CTYPE="de_DE.UTF-8"           
LC_NUMERIC="de_DE.UTF-8"         
LC_TIME="de_DE.UTF-8"            
LC_COLLATE="de_DE.UTF-8"         
LC_MONETARY="de_DE.UTF-8"        
LC_MESSAGES="de_DE.UTF-8"        
LC_PAPER="de_DE.UTF-8"           
LC_NAME="de_DE.UTF-8"            
LC_ADDRESS="de_DE.UTF-8"         
LC_TELEPHONE="de_DE.UTF-8"       
LC_MEASUREMENT="de_DE.UTF-8"     
LC_IDENTIFICATION="de_DE.UTF-8"  
LC_ALL=                   
```

LANGUAGE isn't set and the error/warnings show up!

There is no error in nextcoudPi 

Greetings
Tanarri

nachoparker, 2018-04-14:
great, thanks for the confirmation 

cheers

JohnAtl, 2022-09-23:
Having the same problem.
NextCloudPi 1.50.0

```
pi@nextcloudpi:~ $ locale
locale: Cannot set LC_CTYPE to default locale: No such file or directory
locale: Cannot set LC_MESSAGES to default locale: No such file or directory
locale: Cannot set LC_ALL to default locale: No such file or directory
LANG=en_GB.UTF-8
LANGUAGE=
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```