nextcloud/nextcloudpi #479: NCP Docker Upgrade Process

crouthamela, 2018-04-14:
What is the recommended upgrade process for the docker images?

If I simply to a `docker stop nextcloudpi`, `docker pull nextcloudpi`, `docker rm nextcloudpi`, then re-run my original `docker run` command, to create a container of the latest image, I always get database errors, certificate errors, etc.

In my example, I use the following command: `docker run -d --network=host --restart unless-stopped -v nextcloudpidata:/data --mount type=bind,source=/mnt/users/data,target=/data/app/data --mount type=bind,source=/mnt/media,target=/media --mount type=bind,source=/mnt/backups,target=/backups --name nextcloudpi ownyourbits/nextcloudpi-armhf:march mydomain.com`

The `:march` tag I use is just an alias to a pull I did on March 17th, before the NextCloudPlus branding and new activation page.

I just tested this with a duplicated volume, and when I boot a new container from `nextcloudpi-armhf:latest` pointing to the same /data volume, my Panel (:4443) never loads, I received a certificate mismatch error on mydomain.com:443 page, and I received a MySQL error in the docker logs:
`2018-04-14  4:56:14 1996398592 [Note] mysqld (mysqld 10.1.26-MariaDB-0+deb9u1) starting as process 49 ...
2018-04-14  4:56:14 1996398592 [Warning] Can't create test file /data/database/server.lower-test
mysqld: Can't change dir to '/data/database/' (Errcode: 2 "No such file or directory")
2018-04-14  4:56:14 1996398592 [ERROR] Aborting`

Any help would be appreciated in ensuring I can upgrade my docker containers going forward.

nachoparker, 2018-04-14:
Hello,

The docker container is work in progress. Last week I pushed a bunch of improvements ( from [9th April](https://github.com/nextcloud/nextcloudpi/commits/master) ). I still haven't published the info with the changes. Let me explain the current situation.

- Up until recently the way of upgrading the container was to backup your NC instance, then remove the container and volume, and then pulling the new version and restoring the instance.

- Now we have `nc-update` and `nc-autoupdate-ncp` in the container, so we can (auto) upgrade in-container just like with the SD card version. 

- At this moment upgrading through pulling is not supported yet. It will come, but first we need to do [some separation between code and config](#380 ). This is the main reason why pulling the latest container and reusing the old volume doesn't work reliably: most of the code for ncp-apps lives in the persistent volume. Database and certificates should work in principle though.

I hope it makes more sense now. It goes without saying that any help would be much appreciated to make our goals be accomplished faster. ;)

nachoparker, 2018-04-16:
Yes, good points.

Well, I am trying to bring the docker container to a better shape. Hopefully soon we will be able to upgrade also via docker pull.

`nc-update` and `nc-autoupdate-ncp` are pretty new additions to the docker container, so you might have to pull the latest one ( and run in an empty volume/folder ) to see them.

Thanks for the feedback!

closed 2018-04-19 by nachoparker

lucmos, 2021-01-11:
> Yes, good points.
>
> Well, I am trying to bring the docker container to a better shape. Hopefully soon we will be able to upgrade also via docker pull.
>
> `nc-update` and `nc-autoupdate-ncp` are pretty [...]

Hello,

Just stumbled here, what is the recommended way to update a nextcloudpi installed with docker?

ovpc, 2021-01-11:
https://docs.nextcloudpi.com/en/staying-up-to-date/ works for docker too, for most parts afaik.

lucmos, 2021-01-11:
> https://docs.nextcloudpi.com/en/staying-up-to-date/ works for docker too, for most parts afaik.

Ok, thank you! I read that page, but since it didn't explicitly mention docker I thought it wouldn't apply