nextcloud/nextcloudpi #482: HTTP 500 error when trying to access https://<my ip address>

glennb13, 2018-04-16:
Hi, newbie here. I am not able to access the nextcloud via https://<my ip address>
webpage is showing HTTP 500 error
It is a fresh install via Etcher, did the pi password change, ssh, entered country locale, and enabled wifi.

https://<my ip address>:4443 and  https://<my ip address>:4443/wizard is showing properly though.

output of `sudo ncp-report`
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.53.27
NextCloudPi image    NextCloudPi_03-18-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.9G/118G
rootfs usage         1.9G/118G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       no }
port check 80        open
port check 443       open
IP                   192.168.0.155
gateway              192.168.0.1
interface            wlan0
certificates         none
NAT loopback         no
uptime               11min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES) in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(628): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /var/www/nextcloud/lib/private/Server.php(1511): OC\ServerContainer->query('DatabaseConnect...')
#16 /var/www/nextcloud/lib/private/Server.php(332): OC\Server->getDatabaseConnection()
#17 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
[... 89 lines cut ...]
2018-04-16 12:49:01 1988702208 [Note] Server socket created on IP: '127.0.0.1'.
2018-04-16 12:49:01 1988702208 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"D49yStIUofSIu5BPw4JN","level":4,"time":"2018-03-18T20:30:23+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"6hum7Mc00NWMhxp9PcrW","level":4,"time":"2018-03-18T20:45:16+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"0pGslTZa49CMdLXvcchQ","level":4,"time":"2018-03-18T21:00:15+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"QjDmMDD3iahD8pMaAFP8","level":4,"time":"2018-03-18T21:15:13+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"71jTTe2yIaOKIUUbutpc","level":4,"time":"2018-03-18T21:30:17+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"NsMw6FKjigDIfYUwYKCY","level":4,"time":"2018-03-18T21:45:14+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"Bx8ujWZN1DHx3NGOxtaI","level":4,"time":"2018-03-18T22:00:15+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"Kye7bXpvbCdrgp6nTa6v","level":4,"time":"2018-03-18T22:15:14+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"9RHG3Kjh6F4dFTUJevgp","level":4,"time":"2018-03-18T22:30:22+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"fNP4SWER3cq6MoGyG4lg","level":4,"time":"2018-03-18T22:45:25+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":"13.0.1.1"}

```
</details>

glennb13, 2018-04-19:
Please help point me to the right direction

chiyi4488, 2018-04-19:
Can you use HTTPS?

glennb13, 2018-04-19:
that is actually what is weird. I am using https:// but it gives me;
HTTP 500 error
That’s odd... the website can’t display this page
That's in I.E.  In Opera it gives me even less informative error of;
This page isn’t working
192.168.0.155 is currently unable to handle this request.

If I try http:// it shows the standard self signed certificate warning, after accepting to continue with self signed cert it gives me the same error.

nachoparker, 2018-04-20:
It's weird that only you are seeing this. I would look for [hardware issues](https://github.com/nextcloud/nextcloudpi/wiki/FAQ#why-is-nextcloud-so-slow--failing-randomly--getting-server-errors-even-with-a-fresh-image)

chiyi4488, 2018-04-21:
When I reboot, I also showed 500.

glennb13, 2018-04-21:
Thank you nachoparker!

That was an expensive problem.  I bought an new 128gb SD card, formatted to ext4 with minitool partition, and flashed with etcher, did the country locale, change passwords and enabled wifi. And this time it worked!  I'll reformat my SD hopefully it's not totally broken.

P.S. I ran into that "trusted domain" issue asking to add my IP to the config.php from /var/www/nextcloud/config/config.php and I saw the config.php has gotten a different IP;
  array (
    0 => 'localhost',
    5 => 'nextcloudpi.local',
    1 => '192.168.0.132',
Not sure if this was hardcoded, anyways I changed it to my IP and everything got fixed.

Thanks again

closed 2018-04-21 by glennb13

nachoparker, 2018-04-22:
great!