nextcloud/nextcloudpi #484: NC-INIT ERROR

cargasua, 2018-04-20:
Hi,
I try to use nc-init but I can't. It start the process but ends with with Device "eth0
wlan0" does not exist.

The whole text is: 
[ nc-init ]
Setting up a clean Nextcloud instance... wait until message 'NC init done'
Setting up database...
Setting up Nextcloud...
Nextcloud is not installed - only a limited number of commands are available
Nextcloud was successfully installed
Set mode for background jobs to 'cron'
System config value mysql.utf8mb4 set to boolean true
Device "eth0
wlan0" does not exist.

The result of ncp-report is:
details>
<summary>NextCloudPi diagnostics</summary>
                                                                                                                                                                                                                                                                               
```                                                                                                                                                                                                                                                                            
                                                                                                                                                                                                                                                                               
Device "eth0                                                                                                                                                                                                                                                                   
wlan0" does not exist.                                                                                                                                                                                                                                                         
NextCloudPi version  v0.53.28                                                                                                                                                                                                                                                  
NextCloudPi image    NextCloudPi_03-18-18                                                                                                                                                                                                                                      
distribution         Raspbian GNU/Linux 9 \n \l                                                                                                                                                                                                                                
automount            yes                                                                                                                                                                                                                                                       
USB devices          sda                                                                                                                                                                                                                                                       
datadir              /var/www/nextcloud/data                                                                                                                                                                                                                                   
data in SD           yes                                                                                                                                                                                                                                                       
data filesystem      ext2/ext3                                                                                                                                                                                                                                                 
data disk usage      2.0G/30G                                                                                                                                                                                                                                                  
rootfs usage         2.0G/30G                                                                                                                                                                                                                                                  
swapfile             /var/swap                                                                                                                                                                                                                                                 
Nextcloud check      error                                                                                                                                                                                                                                                     
HTTPD service        up                                                                                                                                                                                                                                                        
PHP service          up                                                                                                                                                                                                                                                        
MariaDB service      up                                                                                                                                                                                                                                                        
Redis service        up                                                                                                                                                                                                                                                        
Postfix service      up                                                                                                                                                                                                                                                        
internet check       ok                                                                                                                                                                                                                                                        
port check 80        open
port check 443       open
IP
gateway              192.168.1.1
192.168.1.1
interface            eth0
wlan0
certificates         none
NAT loopback         yes
uptime               1day

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": [
            "localhost"
        ],
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost",
[... 51 lines cut ...]
```
</details>

<details>
<summary>Database logs</summary>

```

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"WtgmWH8AAQEAAARFxLYAAEAY","level":2,"time":"2018-04-19T05:17:13+00:00","remoteAddr":"192.168.1.36","user":"--","app":"core","method":"POST","url":"\/index.php\/heartbeat","message":"Trusted domain error. \"192.168.1.36\" tried to access using \"192.168.1.33\" as host.","userAgent":"Mozilla\/5.0 (Macintosh; Intel Mac OS X 10.12; rv:59.0) Gecko\/20100101 Firefox\/59.0","version":"13.0.1.1"}
{"reqId":"Wtl0Y38AAQEAAB9XdaQAAAAA","level":2,"time":"2018-04-20T05:02:28+00:00","remoteAddr":"192.168.1.36","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"192.168.1.36\" tried to access using \"192.168.1.33\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:59.0) Gecko\/20100101 Firefox\/59.0","version":"13.0.1.1"}
{"reqId":"Wtl0ZX8AAQEAAB9XdaUAAAEB","level":2,"time":"2018-04-20T05:02:29+00:00","remoteAddr":"192.168.1.36","user":"--","app":"core","method":"GET","url":"\/index.php\/js\/core\/merged-template-prepend.js?v=c5e5e710-0","message":"Trusted domain error. \"192.168.1.36\" tried to access using \"192.168.1.33\" as host.","userAgent":"Mozilla\/5.0 (X11; Linux x86_64; rv:59.0) Gecko\/20100101 Firefox\/59.0","version":"13.0.1.1"}

```
</details>

What can I do? Thank you
Carlos

nachoparker, 2018-04-20:
Hi,

What's the output of

```
IFACE=$( ip r | grep "default via" | awk '{ print $5 }' )
IP=$( ip a show dev "$IFACE" | grep global | grep -oP '\d{1,3}(.\d{1,3}){3}' | head -1 )
echo "IFACE $IFACE"
echo "IP $IP"
ip r
ip a
```

cargasua, 2018-04-20:
Hi,
The output of these commands is:
pi@nextcloudpi:~ $ IFACE=$( ip r | grep "default via" | awk '{ print $5 }' )
pi@nextcloudpi:~ $ IP=$( ip a show dev "$IFACE" | grep global | grep -oP '\d{1,3}(.\d{1,3}){3}' | head -1 )
Device "eth0
wlan0" does not exist.
pi@nextcloudpi:~ $ echo "IFACE $IFACE"
IFACE eth0                                                                                                                           
wlan0                                                                                                                                
pi@nextcloudpi:~ $ echo "IP $IP"                                                                                                     
IP                                                                                                                                   
pi@nextcloudpi:~ $ ip r                                                                                                              
default via 192.168.1.1 dev eth0 src 192.168.1.33 metric 202                                                                         
default via 192.168.1.1 dev wlan0 src 192.168.1.44 metric 303                                                                        
192.168.1.0/24 dev eth0 proto kernel scope link src 192.168.1.33 metric 202                                                          
192.168.1.0/24 dev wlan0 proto kernel scope link src 192.168.1.44 metric 303

Thanks!

nachoparker, 2018-04-21:
I see, you are connected through wifi and ethernet at the same time. Let's try this

```
IFACE=$( ip r | grep "default via" | awk '{ print $5 }' | head -1 )
IP=$( ip a show dev "$IFACE" | grep global | grep -oP '\d{1,3}(.\d{1,3}){3}' | head -1 )
echo "IFACE $IFACE"
echo "IP $IP"
```

cargasua, 2018-04-21:
Hi
I've typed the proposed commands. The problem remains the same.
I don't know if it is related, but the same day that this problem began there was a kernel update

nachoparker, 2018-04-22:
those commands were not meant to solve the problem, but to verify that the fix will work

anyway, I have pushed the fix now. Update with `sudo update` and try again, it should work now

closed 2018-04-22 by nachoparker

cargasua, 2018-04-22:
Hi, 
I've have updated but it doesn't work. I'm thinking about reinstalling
Thx