nextcloud/nextcloudpi #489: can`t enter wizard

FredMed, 2018-04-25:
Hey everyone,

first time for me searching for help at github :-)

I just installed nexcloudplus on my RPI 3. Everything seems to work fine. 

I got that page showing User and Password:
"""NextCloudPlus Actvasion
Your NextCloudPlus user is
ncp
Your NextCloudPlus password is
***
"""

..and i managed to enter my nextcloud and create a new User.
But i cant enter the wizard under : https://myipadress:4443/wizard/

"ncp-web login" is asking for usrname and pwd.. but its not accepting my credentials... 

anybody got an idee what the problem could be? or is there any file in which i can try to change ncp pwd?

not sure if thats an issue for github but i couldn*t find any forum.. so here i am.

Thanks in advance,
fredmed

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPlus diagnostics</summary>

```

nachoparker, 2018-04-26:
Hi,

You can ask in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

closed 2018-04-26 by nachoparker

FredMed, 2018-04-26:
thank u :-)