nextcloud/nextcloudpi #492: Offer default values to revert to in settings

giddyhup, 2018-04-27:
Many configurations (e.g. move database to USB, move data dir to USB, enable ram logging) make reverting back default hard since they lack a 'revert-to-default' option. I had to wade through Nextcloud source code to find where the database was originally stored.

nachoparker, 2018-04-27:
Hi,

we need to redesign ncp-web to account for this and other extra functionality. I will add this to the list of things to do in #380 

thanks for your feedback

closed 2018-04-27 by nachoparker

giddyhup, 2018-04-27:
@nachoparker Thanks. Apparently my data storage is now broken. Even though I moved it back to the SD card I see errors in the log pointing to USB storage:
`filesize(): stat failed for /media/USBdrive/ncdata/appdata_ocmx0enqff4e/avatar/admin/avatar.32.png at /var/www/nextcloud/lib/private/LargeFileHelper.php#171`
I don't mind the files but I need to keep the database (which works) because of addresses and appointments. How can I get a clean slate?

nachoparker, 2018-04-27:
better ask in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)