nextcloud/nextcloudpi #494: ncp-config not working through SSH

Sweenu, 2018-04-28:
I have been having a problem with ncp-config: 
It works flawlessly when I plug my raspberryPi to a display and launch it from there. Works great as well through the web-ui. However, for some reasons, I can't get it to work through ssh.

The tui opens and I can select different items but it will always stop at step n-1 with n being the last step. For example, when I select 'letsencrypt' it will give me the warning (step 1) but after selecting yes, it just doesn't do anything and goes back to the main menu instead of going to the second and final step of entering the domain name and mail.
For items that have only one step, it just doesn't do anything.

I have no ideas where it comes from and if there are any logs I can look at for the ncp-config program.
I guess it has to do with ssh configuration?

Output of `sudo ncp-report` :
<details>
<summary>NextCloudPlus diagnostics</summary>

```

NextCloudPlus version  v0.54.1
NextCloudPlus image    NextCloudPlus_04-21-18
distribution           Raspbian GNU/Linux 9 \n \l
automount              no
USB devices            none
datadir                /var/www/nextcloud/data
data in SD             yes
data filesystem        ext2/ext3
data disk usage        1.9G/29G
rootfs usage           1.9G/29G
swapfile               /var/swap
Nextcloud check        ok
Nextcloud version      13.0.1.1
HTTPD service          up
PHP service            up
MariaDB service        up
Redis service          up
Postfix service        up
internet check         ok
port check 80          open
port check 443         open
IP                     192.168.1.98
gateway                192.168.1.1
interface              eth0
certificates           none
NAT loopback           no
uptime                 10:44

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudplus.local",
            "1": "192.168.1.98"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "13.0.1.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
[... 77 lines cut ...]
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"7oWusGz5voZTr24iwFFe","level":4,"time":"2018-04-21T11:15:42+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"LhAOWosookHQwkgLRBkt","level":4,"time":"2018-04-21T11:30:16+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"OICTazOGVM1n4O0WYwnv","level":4,"time":"2018-04-21T11:45:17+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"vbo3G5olGIJiqpoIMbYf","level":4,"time":"2018-04-21T12:00:16+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"6IlOKGqwMDGz0yMJsRCq","level":4,"time":"2018-04-21T12:15:19+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"6MDixblPJh4CCmuIwikE","level":4,"time":"2018-04-21T12:30:13+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"v9WoqTRYCgFFcD9NlLPk","level":4,"time":"2018-04-21T12:45:14+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"z1FChrOqoErkyM9RIrAq","level":4,"time":"2018-04-21T13:00:16+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"KqjCr9XE1qe5qXFprWJW","level":4,"time":"2018-04-21T13:15:17+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"5tgUofBog3BqSxEj2PSK","level":4,"time":"2018-04-21T13:30:18+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"88TuTxiXVVGuEVmMRunT","level":4,"time":"2018-04-21T13:45:19+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":"13.0.1.1"}
{"reqId":"s5FreASCSnqv6UHHWcCW","level":3,"time":"2018-04-27T22:50:46+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"Redis::connect(): connect() failed: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/RedisFactory.php#84","userAgent":"--","version":"13.0.1.1"}

```
</details>

Sweenu, 2018-04-28:
Okay so I know where the problem comes from. It wasn't ssh but my terminal.
I'm using termite as a terminal emulator and apparently the interface has some kind of bug with it because it works fine with xterm.

Do you have any idea what could cause that and how I could fix it?

nachoparker, 2018-04-29:
no idea, you could try to replace [this line](https://github.com/nextcloud/nextcloudpi/blob/master/etc/library.sh#L153) for a simple `clear`

Sweenu, 2018-04-29:
That didn't work but I found the problem (and solution, kinda).

The problem is that for ncurses based app to fully work on termite, it needs `/lib/terminfo/t/termite` and `$TERM` set to `termite` on the host.

New versions of ncurses include `/lib/terminfo/t/termite` so an ncurses update would solve part of the problem. The rest of the problem comes from termite which would need to accept [this pull request](https://github.com/thestinger/termite/pull/591) I just made in order to use ncurses' terminfo file instead of their own they were using to make up for ncurses not having one until recently.

closed 2018-04-29 by Sweenu

nachoparker, 2018-04-29:
cool, good job!