nextcloud/nextcloudpi #497: docker-armhf Config Page On Port 4443 Not Accessible

msd252, 2018-05-01:
Currently running the docker-armhf on a Odroid HC1 .

The following command is used to initialise the NCP container:

docker run -d -p 4443:4443 -p 443:443 -p 8080:80 -v ncdata:/data --name nextcloudplus ownyourbits/nextcloudplus-armhf $DOMAIN

This is nearly identical to the command listed on https://ownyourbits.com/nextcloudplus/ under "Docker ARM" (as I have another service currently bound to port 80, I have used port 8080).

Both port 80 (port 8080) and 443 work as expected showing the NextCloudPlus Activation page (port 80 redirects to port 443 under TLS). Port 4443, which should be the NextCloudPlus Activation and Config page does not respond. 

Going into the container and running lsof yields the following:

root@2e701e4485c2:/# lsof -i :80,443,4443
COMMAND PID USER   FD   TYPE DEVICE SIZE/OFF NODE NAME
apache2  44 root    3u  IPv4  52479      0t0  TCP *:http (LISTEN)
apache2  44 root    4u  IPv4  52481      0t0  TCP *:https (LISTEN)

Am I missing something or is docker image misconfigured and the config page not available?

I will test this on a fresh system when I can but would be interested if anyone else can replicate?

nachoparker, 2018-05-02:
Hello,

That is to be expected. Once you hit activate, you will have Nextcloud on port 443 and ncp-web on port 4443.

But you need to activate NCP first. You will be automatically redirected to ncp-web when you do so.

closed 2018-05-02 by nachoparker

msd252, 2018-05-02:
Thanks for the clarification.