nextcloud/nextcloudpi #501: WebUI not available

bernd-wechner, 2018-05-12:
Just installed NCP on a Pi2. All good so far. 

I set up ssh access using TUI fine, and I changed the hostname and renamed the default user.

Followed this for the user:

http://unixetc.co.uk/2016/01/07/how-to-rename-the-default-raspberry-pi-user/#comment-78852

and just "sudo hostname Nephele" for the host name.

Standard moves I make for Pis on my LAN. 

TUI works fine. Used it to move the cdp data dir and database to a USB drive. Would like to move logs to the USB drive too but can only see a TUI command to move them to RAM (which is not what I want, yes I want to save the SSD from undue wear, but I want persistent rolling logs too, not RAM logs, but that is a digression).

So I try the WebUI. But cannot connect.

Firefox reports:

> Unable to connect
> 
> Firefox can’t establish a connection to the server at 192.168.0.14:4443.
> 
>     The site could be temporarily unavailable or too busy. Try again in a few moments.
>     If you are unable to load any pages, check your computer’s network connection.
>     If your computer or network is protected by a firewall or proxy, make sure that Firefox is permitted to access the Web.

Bummer. Checked the apache logs and this is the error log:

```
[Sat May 12 22:10:19.452202 2018] [ssl:warn] [pid 469:tid 1992712192] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sat May 12 22:10:19.453854 2018] [ssl:error] [pid 469:tid 1992712192] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: AE8EA77112056528 / notbefore: Apr 20 14:05:43 2018 GMT / notafter: Apr 17 14:05:43 2028 GMT]
[Sat May 12 22:10:19.453977 2018] [ssl:error] [pid 469:tid 1992712192] AH02604: Unable to configure certificate localhost:4443:0 for stapling
[Sat May 12 22:10:19.456367 2018] [ssl:warn] [pid 469:tid 1992712192] AH01909: localhost:443:0 server certificate does NOT include an ID which matches the server name
[Sat May 12 22:10:19.457421 2018] [ssl:error] [pid 469:tid 1992712192] AH02217: ssl_stapling_init_cert: can't retrieve issuer certificate! [subject: CN=raspberrypi / issuer: CN=raspberrypi / serial: AE8EA77112056528 / notbefore: Apr 20 14:05:43 2018 GMT / notafter: Apr 17 14:05:43 2028 GMT]
[Sat May 12 22:10:19.457566 2018] [ssl:error] [pid 469:tid 1992712192] AH02604: Unable to configure certificate localhost:443:0 for stapling
[Sat May 12 22:10:19.468773 2018] [unique_id:alert] [pid 469:tid 1992712192] (EAI 2)Name or service not known: AH01564: unable to find IPv4 address of "Nephele"
AH00016: Configuration Failed
```
Taking a look at listeners:

```
# netstat -ltupe
Active Internet connections (only servers)
Proto Recv-Q Send-Q Local Address           Foreign Address         State       User       Inode      PID/Program name    
tcp        0      0 localhost:mysql         0.0.0.0:*               LISTEN      mysql      12152      869/mysqld          
tcp        0      0 0.0.0.0:ssh             0.0.0.0:*               LISTEN      root       14422      516/sshd            
tcp        0      0 0.0.0.0:smtp            0.0.0.0:*               LISTEN      root       12745      863/master          
tcp6       0      0 [::]:ssh                [::]:*                  LISTEN      root       14424      516/sshd            
udp        0      0 0.0.0.0:bootpc          0.0.0.0:*                           root       9880       435/dhcpcd          
udp        0      0 0.0.0.0:60525           0.0.0.0:*                           avahi      8802       301/avahi-daemon: r 
udp        0      0 0.0.0.0:mdns            0.0.0.0:*                           avahi      8800       301/avahi-daemon: r 
udp6       0      0 [::]:dhcpv6-client      [::]:*                              root       985        435/dhcpcd          
udp6       0      0 [::]:36401              [::]:*                              avahi      8803       301/avahi-daemon: r 
udp6       0      0 [::]:mdns               [::]:*                              avahi      8801       301/avahi-daemon: r 
```

Apache not listening at all alas. 

Seems it's bombing because of certificates. Sheesh. The cert is written for "raspberrypi" it seems. 

```
# openssl x509 -in /etc/ssl/certs/ssl-cert-snakeoil.pem -text -noout
Certificate:
    Data:
        Version: 3 (0x2)
        Serial Number:
            ae:8e:a7:71:12:05:65:28
    Signature Algorithm: sha256WithRSAEncryption
        Issuer: CN = raspberrypi
        Validity
            Not Before: Apr 20 14:05:43 2018 GMT
            Not After : Apr 17 14:05:43 2028 GMT
        Subject: CN = raspberrypi
        Subject Public Key Info:
            Public Key Algorithm: rsaEncryption
                Public-Key: (2048 bit)
                Modulus:
                    00:a2:43:36:22:7a:3c:6b:65:d4:cb:62:29:a8:b9:
                    7b:ee:d9:d8:59:70:64:c1:9d:bf:74:15:53:66:16:
                    73:1b:9f:b0:96:e1:94:39:45:aa:d5:76:df:06:b4:
                    8c:55:3f:99:5b:8e:88:ae:44:8c:41:8d:6e:b2:62:
                    5d:f3:48:80:99:af:b4:1c:d8:28:67:9e:4e:b5:b9:
                    e7:20:2c:c6:a1:8a:e7:04:e4:1e:89:d7:a2:cd:5b:
                    3a:89:ab:be:f7:f5:3f:c9:8b:ea:fa:04:a4:29:d8:
                    4c:9f:ff:87:15:d4:02:48:b2:41:f4:bc:8f:02:35:
                    65:8c:6a:a6:a3:36:b6:2b:5e:3b:51:6d:cf:28:07:
                    6a:4b:b5:8d:3f:3d:58:29:a8:b5:a4:bb:db:57:7e:
                    c3:77:0c:0c:3f:2c:3b:93:f7:74:7b:0e:7b:66:70:
                    d0:55:4a:be:52:cc:ec:db:44:25:87:66:79:22:1d:
                    84:17:b1:57:96:7f:a6:59:76:9f:58:7c:44:77:f4:
                    ff:90:dd:65:68:a4:67:4e:14:4e:64:ad:88:71:91:
                    ca:a4:a3:c5:ec:4c:e0:3c:a2:2a:88:9b:ab:ee:b3:
                    b3:46:0d:75:ff:65:58:58:43:b1:56:75:75:ff:4a:
                    5d:56:1e:c1:b5:44:eb:38:4a:f6:02:6a:da:c3:73:
                    cb:c9
                Exponent: 65537 (0x10001)
        X509v3 extensions:
            X509v3 Basic Constraints: 
                CA:FALSE
            X509v3 Subject Alternative Name: 
                DNS:raspberrypi
    Signature Algorithm: sha256WithRSAEncryption
         82:9d:3b:74:7e:a1:11:86:5d:66:56:e4:6c:35:c8:5b:3d:09:
         d2:95:7f:e0:3d:d3:ca:7d:e7:fc:50:1a:d4:d2:fb:52:28:34:
         40:06:ce:89:7a:4c:e5:6c:6b:79:ff:a5:8f:ea:c7:8a:b6:1a:
         a1:a6:4f:a3:5b:ae:be:dc:05:89:c3:6c:31:b3:58:83:97:f3:
         ea:a4:9c:40:a9:43:4a:ef:ec:4d:8e:8a:de:56:04:eb:bb:04:
         56:fd:dd:f4:95:90:f9:34:fd:19:f1:07:66:78:70:82:10:8b:
         e8:2f:65:57:f6:5d:fa:21:28:25:37:25:89:01:64:7c:9f:db:
         8d:e7:f4:e4:f9:c2:94:9b:61:86:21:28:01:e6:d6:2d:2d:70:
         82:18:d7:66:da:25:e1:8b:12:a8:45:e1:06:38:7d:f9:a1:9d:
         52:84:dd:5e:07:39:33:47:e3:fe:84:07:fb:bc:03:d2:36:a8:
         f8:35:27:a2:e4:27:0a:84:ee:74:dd:19:40:71:24:58:f3:f3:
         7f:e2:01:af:0e:c2:1c:cd:ad:6d:21:8e:90:b8:b4:30:88:42:
         9e:de:57:19:24:6f:a4:ef:52:8e:e5:d9:f9:8f:ea:f3:90:99:
         65:99:ef:4b:db:6f:ab:70:53:2f:99:12:95:59:5a:eb:d9:8e:
         67:25:77:84
```

I guess I have a way forward to fix that, get my own certs (certbot here I come), but I post an issue here for the following reasons:

1. To get some confirmation of my diagnosis. I'm a tad blown away that apache refuses to even listen on account of a cert and hostname mismatch, but hey (I typically use lighttpd and it better than that checking certs only after receiving a request and moreover I serve many certs for different hostnames from one box using it as I host a number of domains on the one box, hence am a tad surprised with this apparent behaviour of apache's but rather naive on apache configs - been two decades since I last administered it ;-).

2. To suggest that ncp perhaps consider adding TUI options to do these fairly commonly (IMO) desired configs: change hostname and change default user name. Both are eminently scriptable and could in so doing warn that WebUI serving won't work unless ... saving suckers like me this evenings diagnostic efforts ;-). I don't mind, if I find time taking a crack at scripting those as a service if ncp takes PRs in this space, though am not sure I'll find/make time for it.

3. To solicit advice on what I need to do to fix it. My guess is get a cert for my host, and edit `/etc/apache2/sites-enabled/ncp.conf` to point to them. Is that all? Will that see Apache listening and the WebUI available? Guess I'll find out soon when I try ...

bernd-wechner, 2018-05-13:
OK, so tried with TUI to install letsencrypt. By selecting in the TUI:

letsencrypt         Automatic signed SSL certificates   

it bombs with:

```
Launching letsencrypt
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
Obtaining a new certificate
Performing the following challenges:
http-01 challenge for nephele.thumbs.place
Using the webroot path /var/www/nextcloud for all unmatched domains.
Waiting for verification...
Cleaning up challenges
Failed authorization procedure. nephele.thumbs.place (http-01): urn:acme:error:unauthorized :: The client lacks sufficient authorization :: Invalid response from http://nephele.thumbs.place/.well-known/acme-challenge/e06t_Pua38OmigFBtBZ92hN7XNEzlnqiedT4H73EDko: "<?xml version="1.0" encoding="iso-8859-1"?>
<!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
         "http://www."
IMPORTANT NOTES:
 - The following errors were reported by the server:

   Domain: nephele.thumbs.place
   Type:   unauthorized
   Detail: Invalid response from
   http://nephele.thumbs.place/.well-known/acme-challenge/e06t_Pua38OmigFBtBZ92hN7XNEzlnqiedT4H73EDko:
   "<?xml version="1.0" encoding="iso-8859-1"?>
   <!DOCTYPE html PUBLIC "-//W3C//DTD XHTML 1.0 Transitional//EN"
            "http://www."

   To fix these errors, please make sure that your domain name was
   entered correctly and the DNS A/AAAA record(s) for that domain
   contain(s) the right IP address.
 - Your account credentials have been saved in your Certbot
   configuration directory at /etc/letsencrypt. You should make a
   secure backup of this folder now. This configuration directory will
   also contain certificates and private keys obtained by Certbot so
   making regular backups of this folder is ideal.
Done. Press any key...
```
and the attached log file.

[letsencrypt.log](https://github.com/nextcloud/nextcloudpi/files/1998172/letsencrypt.log)

Odd references to nginx in there, not least give I can't see after this, nginx or certbot installed. Hmmm, the effort to diagnose this continues.

nachoparker, 2018-05-13:
ncp-web is not available on port 4443 initially, until you activate your instance ( by accessing port 80 or 443 ). 

Then, letsencrypt should work as well because available ports will be changed after activation to the normal ones.

bernd-wechner, 2018-05-14:
Hmmm, this is puzzling and the price I pay for wanting to individualise my hostname methinks.

Cut a long story as short as I can, I did this:

It turned SSLEngine off in:

/etc/apache2/sites-enabled/ncp.conf
/etc/apache2/sites-enabled/ncp-activation.conf

and started apache and noted it did start! And was listening.Progress.

Note Apache was not even starting prior to this complaining as noted in the OP that "server certificate does NOT include an ID which matches the server name"

I then used TUI to install letsencrypt. And voila that worked, I have certs. At least for the full domain name not the .lan name. But the upshot is that I turned SSLEngine back on and restarted and apache is now running and listening. Progress. And I have certs.

But there are further issues.

Now if I navigate to:

https://192.168.0.14

This produces a popup asking "https://192.168.0.14 is requesting your username and password. The site says: “ncp-web login”"

if I click OK I get:
```
Internal Server Error

The server encountered an internal error or misconfiguration and was unable to complete your request.

Please contact the server administrator at [no address given] to inform them of the time this error occurred, and the actions you performed just before this error.

More information about this error may be available in the server error log.
```

If I repeat and this time  I don't get the login pop just the error. Grrr.

If I navigate here:

https://192.168.0.14/index.php/activate

I get:

```
The page isn’t redirecting properly

Firefox has detected that the server is redirecting the request for this address in a way that will never complete.

    This problem can sometimes be caused by disabling or refusing to accept cookies.
```

Man this is proving difficult. 

For what tit's worth in config/config.php I've added the IP and all the possible domain names to trusted_domains. Had to do that to get this far.

Methinks NextCloudPlus is seriously configured assuming you just keep the standard host name and accounts ... and I'm tossing up continuing this (as it is useful in terms of learning too I admit, albeit a huge hassle) and just reinstalling the image and starting again.

nachoparker, 2018-05-15:
Mmm there should be no need to disable SSL. In any case, we need the output of `sudo ncp-report` in order to help.

Also, please test that everything works before changing anything manually so we can make sure if the problem is in our end.

If the problem persists, please reopen. Questions/conversation on configuration is better to happen in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)

closed 2018-05-15 by nachoparker

bernd-wechner, 2018-05-16:
I doubt hardware issues. It's a clean Pi. The issue is certainly to do with the fact I changed the hostname and the default user name. I think I'm stuck with either starting anew with a clean NetxCloudPlus image and seeing where I get to, or ditching it and just installing NextCloud on an Ubuntu server I'm running, I thought it would be a great use for the Pi with it's USB hard drive.

bernd-wechner, 2018-06-17:
An update. Two salient things since I last checked in here:

1) Yes, it cost me some frustration and time, but there most definitely was a hardware issue somewhere along the way. Namely the 16GB SD card I was using flaked. I learned a lot about flaking SD cards in the process and how to check them (with enormous thanks to the F3 utils). To summarise the failure mode is transparent like you might not even notice if for some while and when you do you end up scratching your head wondering if you're in the twilight zone. Because the way this one (and it's not alone as I only learned this was possible because of abundant reports of same and recommendations to try F3) fails is that it would not accept writes. It was read only. Now you think you'd notice that ordinarily but the way it doesn't accept writes is almost evil in its wicked invisibility, namely it does accept and hold writes. You can do what you want, and all is good for as long as you want, but come reboot time the drive is back to where it was at lockdown (when writes first failed). I was literally rubbing my eyes, installing a package with apt, using it, rebooting and it was gone, rinse repeat do it a few times and check sobriety then search web and find lots of people reporting similar. And this mode of SD card failure being known and not uncommon. Grrrr.

2) I did take a more cautious approach and install form a new image, then test as is. And all worked. So took a snapshot of the filesystem and then tinkered, with my certs, and options and account creation etc, and it all still worked. The main things I like doing on a Pi which I did before testing it when I reported this issue, are a) rename the host to suit my LAN, b) rename the pi user, c) set up sshkeys for both my user and root, then disable root password login. This time I was a bit more cautious and did these backwards, but kept the pi user, created my own user account as a sudoer and then disabled the pi account, then changed the hostname with raspi-config. Testing between each change. And all is sweet and works well. One salient difference being I'm on a new SD card ;-). 

Anyhow started from scratch, tested stepwise and took backups between possibly risk changes, but have a working NCP system now. Thanks for an awesome tool!

nachoparker, 2018-06-17:
great! I am glad you sorted it out