nextcloud/nextcloudpi #502: Wifi problems

Ricardosgeral, 2018-05-15:
Hi
My NC instance is working ok with LAN but if I use wifi i have a problem.

After I configure WIFI using nextcloudpi-config everything works, but if i reboot the PI for some reason, the raspberry can't find the network that was connected previously. It shows others networks visible in the range.

The only way i can put it working again via WIFI is to say NO in the configuration of the wifi via nextcloudpi-config, and then say YES. This way the network reconnects automatically (since i have selected the connect automatic).

In sum, I have something in the wifi configuration in the nextcloudpi that should not be ok.

Any Ideas?

Paste the output of `sudo ncp-report`

Ricardosgeral, 2018-05-15:
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPlus diagnostics</summary>

```

NextCloudPlus version  v0.54.7
NextCloudPlus image    NextCloudPi_12-04-17
distribution           Raspbian GNU/Linux 9 \n \l
automount              yes
USB devices            sda
datadir                /media/myCloudDrive/ncdata
data in SD             no
data filesystem        btrfs
data disk usage        1.3T/2.8T
rootfs usage           2.6G/15G
swapfile               /var/swap
Nextcloud check        ok
Nextcloud version      13.0.2.1
HTTPD service          up
PHP service            up
MariaDB service        up
Redis service          up
Postfix service        up
internet check         ok
port check 80          open
port check 443         open
IP                     192.168.1.3
gateway                192.168.1.1
interface              eth0
certificates           ricarcloud.duckdns.org
NAT loopback           no
uptime                 18:03

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.6",
            "3": "ricarcloud.ddns.net",
            "4": "ricarcloud.duckdns.org"
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/ricarcloud.duckdns.org",
        "dbtype": "mysql",
        "version": "13.0.2.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "overwriteprotocol": "https",
        "maintenance": false,
        "mail_smtpmode": "smtp",
        "mail_smtpauthtype": "LOGIN",
[... 47 lines cut ...]
```

{"reqId":"WvJ@zn8AAQEAAGv36iwAAAAS","level":2,"time":"2018-05-09T04:53:34+00:00","remoteAddr":"139.162.116.133","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.116.133\" tried to access using \"109.49.71.205\" as host.","userAgent":"Go-http-client\/1.1","version":"13.0.2.1"}
{"reqId":"WvMaSn8AAQEAAAW5lDMAAABC","level":2,"time":"2018-05-09T15:56:58+00:00","remoteAddr":"184.105.139.69","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.139.69\" tried to access using \"109.49.71.205\" as host.","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WvNgZH8AAQEAAAW4RzQAAAAN","level":2,"time":"2018-05-09T20:56:04+00:00","remoteAddr":"139.162.113.204","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.113.204\" tried to access using \"109.49.71.205\" as host.","userAgent":"Go-http-client\/1.1","version":"13.0.2.1"}
{"reqId":"WvPIz38AAQEAAAW5qwEAAABQ","level":2,"time":"2018-05-10T04:21:35+00:00","remoteAddr":"166.84.6.79","user":"--","app":"core","method":"GET","url":"\/js\/slinkroot.js","message":"Trusted domain error. \"166.84.6.79\" tried to access using \"109.49.71.205\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/42.0.2311.135 Safari\/537.36 Edge\/12.10136","version":"13.0.2.1"}
{"reqId":"WvPVDH8AAQEAAAW5rD4AAABH","level":2,"time":"2018-05-10T05:13:48+00:00","remoteAddr":"107.170.248.183","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"107.170.248.183\" tried to access using \"109.49.71.205\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"13.0.2.1"}
{"reqId":"WvPayX8AAQEAAAW5rNEAAABC","level":2,"time":"2018-05-10T05:38:17+00:00","remoteAddr":"139.162.78.135","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"139.162.78.135\" tried to access using \"109.49.71.205\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"13.0.2.1"}
{"reqId":"WvQx0n8AAQEAAHP6ynEAAAAA","level":2,"time":"2018-05-10T11:49:38+00:00","remoteAddr":"172.104.121.52","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"172.104.121.52\" tried to access using \"109.49.71.205\" as host.","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WvRP2X8AAQEAAHP728oAAABM","level":2,"time":"2018-05-10T13:57:45+00:00","remoteAddr":"5.188.9.100","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"5.188.9.100\" tried to access using \"109.49.71.205\" as host.","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WvRkdn8AAQEAAHP60F4AAAAH","level":2,"time":"2018-05-10T15:25:42+00:00","remoteAddr":"184.105.247.194","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.247.194\" tried to access using \"109.49.71.205\" as host.","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WvRtmH8AAQEAAHP734QAAABX","level":2,"time":"2018-05-10T16:04:40+00:00","remoteAddr":"71.6.202.205","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"71.6.202.205\" tried to access using \"109.49.71.205\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 10.0; WOW64) AppleWebKit\/537.36 (KHTML, like Gecko) Chrome\/65.0.3325.181 Safari\/537.36","version":"13.0.2.1"}
[... cut ...]
{"reqId":"Wvq6Yn8AAQEAAH5FkwwAAABL","level":2,"time":"2018-05-15T10:45:54+00:00","remoteAddr":"181.48.9.82","user":"--","app":"core","method":"GET","url":"\/admin\/\/config.php","message":"Trusted domain error. \"181.48.9.82\" tried to access using \"109.49.71.205\" as host.","userAgent":"curl\/7.29.0","version":"13.0.2.1"}
{"reqId":"WvrPuX8AAQEAAH5FlMAAAABB","level":2,"time":"2018-05-15T12:16:57+00:00","remoteAddr":"141.212.122.81","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"141.212.122.81\" tried to access using \"109.49.71.205\" as host.","userAgent":"Mozilla\/5.0 zgrab\/0.x","version":"13.0.2.1"}

```
</details>

Ricardosgeral, 2018-05-15:
> pi@nextcloudpi:~ $ systemctl status wicd
> ● wicd.service - LSB: Starts and stops Wicd
>    Loaded: loaded (/etc/init.d/wicd; generated; vendor preset: enabled)
>    Active: inactive (dead)
>      Docs: man:systemd-sysv-generator(8)
>
> May 14 19:57:56 nextcloudpi systemd[1]: Starting LSB: Starts and stops
> Wicd...
> May 14 19:57:59 nextcloudpi wicd[376]: Starting Network connection
> manager: wicd
> May 14 19:57:59 nextcloudpi systemd[1]: Started LSB: Starts and stops Wicd.
> May 14 20:00:59 nextcloudpi systemd[1]: Stopping LSB: Starts and stops
> Wicd...
> May 14 20:00:59 nextcloudpi wicd[2570]: Stopping Network connection
> manager: wic
> May 14 20:00:59 nextcloudpi systemd[1]: Stopped LSB: Starts and stops Wicd.
> lines 1-11/11 (END)

Another thing
Not know if it is relevante
This started happening after I have turn off the power accidentally to the
RPi.

nachoparker, 2018-05-27:
couldn't reproduce your problem, so remove the bug tag. There are some changes though that are worth retesting

Ricardosgeral, 2018-05-28:
ok. Nothing else to report from me.

nachoparker, 2018-05-28:
for some reason your `systemctl status` shows that `wicd` is not running/active. It should be.

Did you click on "automatically connect"?

[image]

Maybe there will be some clue in `/var/log/wicd/`

nachoparker, 2018-05-28:
also I have noticed that if you leave the cable plugged in during reboot, it will connect to the cable and not to wifi

Ricardosgeral, 2018-05-28:
For now I give up on the WiFi and I'm using the cable
I will try later the WiFi.
And yes I've putted automatic connect in the settings