nextcloud/nextcloudpi #504: wget canyouseeme.org timeout too small causes many retries

ejsmit, 2018-05-16:
<details>
<summary>NextCloudPlus diagnostics</summary>

```

NextCloudPlus version  v0.56.2
NextCloudPlus image    NextCloudPlus_04-21-18
distribution           Raspbian GNU/Linux 9 \n \l
automount              no
USB devices            sda 
datadir                /media/USBdrive/ncdata
data in SD             no
data filesystem        btrfs
data disk usage        391G/1.9T
rootfs usage           1.7G/7.3G
swapfile               /var/swap
Nextcloud check        ok
Nextcloud version      13.0.1.1
HTTPD service          up
PHP service            up
MariaDB service        up
Redis service          up
Postfix service        up
internet check         ok
port check 80          open
port check 443         open
IP                     192.168.0.250
gateway                192.168.0.1
interface              eth0
certificates           none
NAT loopback           no
uptime                 34min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "5": "nextcloudplus.local",
            "1": "192.168.0.250",
        },
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "http:\/\/localhost",
        "dbtype": "mysql",
        "version": "13.0.1.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
        "dbpassword": "***REMOVED SENSITIVE VALUE***",
        "installed": true,
        "instanceid": "***REMOVED SENSITIVE VALUE***",
        "memcache.local": "\\OC\\Memcache\\Redis",
        "memcache.locking": "\\OC\\Memcache\\Redis",
        "redis": {
            "host": "***REMOVED SENSITIVE VALUE***",
            "port": 0,
            "timeout": 0,
            "password": "***REMOVED SENSITIVE VALUE***"
[... 95 lines cut ...]

Running the same commands manually from the command line without the --quiet flag shows that the wget command times out within 2 seconds, and then retries because it has not yet received a valid response. I think it retries 20 times.    

Setting the timeout to a bigger number works.  In my case I needed at about 10 for wget to actually return a html result containing the required Connection Refused string.  

Hardware is RPi 3B+, fast UK Fibre internet connection, so internet or hardware performance should hopefully not cause performance issues.  Ports are closed and will remain closed.   This nextcloud instance will never be exposed to the internet.    Has never been an issue before after years of running owncloud and nextcloud.

```
$ wget --timeout=2  http://canyouseeme.org --post-data 'IP=<removed>&port=443'
--2018-05-16 23:22:33--  http://canyouseeme.org/
Resolving canyouseeme.org (canyouseeme.org)... 52.202.215.126
Connecting to canyouseeme.org (canyouseeme.org)|52.202.215.126|:80... connected.
HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
Retrying.

--2018-05-16 23:22:37--  (try: 2)  http://canyouseeme.org/
Connecting to canyouseeme.org (canyouseeme.org)|52.202.215.126|:80... connected.
HTTP request sent, awaiting response... Read error (Connection timed out) in headers.
Retrying.

--2018-05-16 23:22:41--  (try: 3)  http://canyouseeme.org/
Connecting to canyouseeme.org (canyouseeme.org)|52.202.215.126|:80... connected.
HTTP request sent, awaiting response... 
^C
```

ejsmit, 2018-05-17:
something else I just realised:  from the canyouseeme.org website I also don't get a Connection Refused message.   I get a timed out message.

[image]

nachoparker, 2018-05-17:
what I remember is that once you open your ports, the service caches that and will keep telling you that they are open even if they are not.

maybe we should find a better service. I had to add the timeout because the information in ncp-web was taking forever to load.

thanks for the feedback

nachoparker, 2018-05-17:
or maybe there's a way of telling wget (or curl) not to retry?

ejsmit, 2018-05-17:
Don't know about caching, but my ports have never been open, and the UI tell me they are.   The retry count should solve the long almost 2 minute delay where the admin website is completely unresponsive if ports are closed.

quick man page search:
``` 
       -t number
       --tries=number
           Set number of tries to number. Specify 0 or inf for infinite retrying.  The default is to retry
           20 times, with the exception of fatal errors like "connection refused" or "not found" (404),
           which are not retried.
```

ejsmit, 2018-05-17:
suggestion:   move the port status away from the main system info page.   Maybe put it behind a button or something similar so that it does not automatically run on page load.   

users tend to be perfectly happy to wait for an operation to finish as long as they expect it.  The button can have clear warnings that it may take up to 20 seconds to fully check the port status.   Even I would be happy to press that button every once in a while to check the port status. 

Unfortunately I am not a developer otherwise I would have helped....

nachoparker, 2018-05-18:
wow canyouseeme.org is superslow now... it takes 26 seconds to load from my browser.

that explains everything... definitely need another provider for this

nachoparker, 2018-05-18:
for now, let's make it only two second timeout, only one attempt

nachoparker, 2018-05-18:
Ok, I changed to portchecker.co and made sure that it won't get stuck more tan two seconds in `v0.56.4`.

Please, reopen if there is still problems

Thanks

closed 2018-05-18 by nachoparker