nextcloud/nextcloudpi #506: ncp-test-updates: command not found

TimmThaler, 2018-05-19:
When executing the cron job for updating ncp there is an error. I get a lot of messages in my user mail for "root" just like this:

> /usr/local/bin/ncp-notify-update: line 7: ncp-test-updates: command not found
NextCloudPlus up to date
no new upgrades

When I run the "ncp-test-updates" command by myself there is no error on the cli ...

Any ideas? :smiley:

nachoparker, 2018-05-19:
what's the output of 

```
ls -l /usr/local/bin/ncp-test-updates
```
?

And also, paste output of

```
sudo ncp-report
```

TimmThaler, 2018-05-19:
> ls -l /usr/local/bin/ncp-test-updates

shows
>-rwxr-xr-x 1 root staff 977 May 19 10:35 /usr/local/bin/ncp-test-updates

and 
> sudo ncp-report

shows this:

<details>
<summary>NextCloudPlus diagnostics</summary>

```

NextCloudPlus version  v0.56.4
NextCloudPlus image    NextCloudPi_03-18-18
distribution           Raspbian GNU/Linux 9 \n \l
automount              yes
USB devices            sda sdb sr0 
datadir                /media/USBdrive/ncdata
data in SD             no
data filesystem        btrfs
data disk usage        20G/466G
rootfs usage           3.2G/15G
swapfile               /var/swap
Nextcloud check        ok
Nextcloud version      13.0.2.1
HTTPD service          up
PHP service            up
MariaDB service        up
Redis service          up
Postfix service        up
internet check         ok
port check 80          open
port check 443         open
IP                     192.168.2.108
gateway                192.168.2.1
interface              eth0
certificates           ***REMOVED SENSITIVE VALUE***
NAT loopback           yes
uptime                 7days

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": [
            "localhost",
            "***REMOVED SENSITIVE VALUE***",
            "***REMOVED SENSITIVE VALUE***"
        ],
        "datadirectory": "***REMOVED SENSITIVE VALUE***",
        "overwrite.cli.url": "https:\/\/***REMOVED SENSITIVE VALUE***",
        "dbtype": "mysql",
        "version": "13.0.2.1",
        "dbname": "***REMOVED SENSITIVE VALUE***",
        "dbhost": "***REMOVED SENSITIVE VALUE***",
        "dbport": "",
        "dbtableprefix": "oc_",
        "mysql.utf8mb4": true,
        "dbuser": "***REMOVED SENSITIVE VALUE***",
[... 74 lines cut ...]
```

{"reqId":"WwBGan8AAQEAAHVFa4oAAABD","level":0,"time":"2018-05-19T15:44:44+00:00","remoteAddr":"74.82.47.4","user":"--","app":"core","method":"GET","url":"\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery-ui-fixes.scss, ignoring","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WwBGan8AAQEAAHVFa4oAAABD","level":0,"time":"2018-05-19T15:44:44+00:00","remoteAddr":"74.82.47.4","user":"--","app":"core","method":"GET","url":"\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/server.scss, ignoring","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WwBGan8AAQEAAHVFa4oAAABD","level":0,"time":"2018-05-19T15:44:44+00:00","remoteAddr":"74.82.47.4","user":"--","app":"core","method":"GET","url":"\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/jquery.ocdialog.scss, ignoring","userAgent":"--","version":"13.0.2.1"}
{"reqId":"WwBGan8AAQEAAHVFa4oAAABD","level":0,"time":"2018-05-19T15:44:44+00:00","remoteAddr":"74.82.47.4","user":"--","app":"core","method":"GET","url":"\/","message":"Scss is disabled for \/var\/www\/nextcloud\/core\/css\/share.scss, ignoring","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Activity\\BackgroundJob\\EmailNotification job with ID 7","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Activity\\BackgroundJob\\EmailNotification job with ID 7 in 0 seconds","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Files\\BackgroundJob\\ScanFiles job with ID 9","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Files\\BackgroundJob\\ScanFiles job with ID 9 in 0 seconds","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\UpdateNotification\\ResetTokenBackgroundJob job with ID 266","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\UpdateNotification\\ResetTokenBackgroundJob job with ID 266 in 0 seconds","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Files\\BackgroundJob\\CleanupFileLocks job with ID 11","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OCA\\Files\\BackgroundJob\\CleanupFileLocks job with ID 11 in 0 seconds","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OCA\\Files_Versions\\BackgroundJob\\ExpireVersions job with ID 12","userAgent":"--","version":"13.0.2.1"}
[... cut ...] .0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Run OC\\Log\\Rotate job with ID 15","userAgent":"--","version":"13.0.2.1"}
{"reqId":"BcQNmcu194anPQRMxrfv","level":0,"time":"2018-05-19T15:45:04+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Finished OC\\Log\\Rotate job with ID 15 in 0 seconds","userAgent":"--","version":"13.0.2.1"}

```
</details>

nachoparker, 2018-05-20:
that's so weird. Does it get fixed by editing `/usr/local/bin/ncp-notify-update` and writing the full path in line 7?

```
/usr/local/bin/ncp-test-updates || { echo "NextCloudPlus up to date"; exit 0; }
```

TimmThaler, 2018-05-21:
Yes, that's working.
Now the error message is gone. :+1: 

That was quite easy :laughing:

closed 2018-05-21 by TimmThaler

TimmThaler, 2018-05-26:
With the path changed, the automated NCP updates are working on my instance :smiley: