nextcloud/nextcloudpi #507: Nextcloud Offline httpd service down Ports 80, 443 closed

pilepalle, 2018-05-19:
Hi guys, 
I am new in the Nextcloudpi environment. I had the NextcloudPI image ready to go. Then I thought I'm going to update to the latest versions. Unfortunately, something went wrong. Since then, I'm no longer on the Nextcloud. It can be seen in the ncp-report that the httpd service is down and the port check 80 and 443 are closed. Does anyone have an idea what that can be?

[image]

Thanks

nachoparker, 2018-05-19:
hi,

what's the output of 

```
systemctl status apache2
```

pilepalle, 2018-05-19:
● apache2.service - The Apache HTTP Server
   Loaded: loaded (/lib/systemd/system/apache2.service; enabled; vendor preset: enabled)
   Active: failed (Result: exit-code) since Sat 2018-05-19 14:03:48 CEST; 3h 5min ago
  Process: 3227 ExecStart=/usr/sbin/apachectl start (code=exited, status=1/FAILURE)

May 19 14:03:48 nextcloudplus systemd[1]: Starting The Apache HTTP Server...
May 19 14:03:48 nextcloudplus apachectl[3227]: (2)No such file or directory: AH02291: Cannot access directory '/var/www/nextcloud/data/' for error log of vhost defined at /etc/apache2/sites-enabled/nextcloud.conf:2
May 19 14:03:48 nextcloudplus apachectl[3227]: AH00014: Configuration check failed
May 19 14:03:48 nextcloudplus apachectl[3227]: Action 'start' failed.
May 19 14:03:48 nextcloudplus apachectl[3227]: The Apache error log may have more information.
May 19 14:03:48 nextcloudplus systemd[1]: apache2.service: Control process exited, code=exited status=1
May 19 14:03:48 nextcloudplus systemd[1]: Failed to start The Apache HTTP Server.
May 19 14:03:48 nextcloudplus systemd[1]: apache2.service: Unit entered failed state.
May 19 14:03:48 nextcloudplus systemd[1]: apache2.service: Failed with result 'exit-code'.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~

pilepalle, 2018-05-19:
I've just set up the nextcloudpi on another raspi. The update to 0.56.4 and the Nextcloud from 13.0.1.1 to 13.02.1 performed. After a reboot same problem.

nachoparker, 2018-05-20:
how did you perform the Nextcloud update? from NC itself, or from ncp-web?

nachoparker, 2018-05-20:
ok, that feature was just added recently. If you could help me fix this... can I get the output of the process?

You can get it from `/var/log/ncp.log`, or from ncp-web/ncp-config right after you launch `nc-update-nextcloud`

nachoparker, 2018-05-20:
you don't need to issue those commands. It should get fixed by updating to `v0.56.6`

thanks!

closed 2018-05-20 by nachoparker

nachoparker, 2018-05-22:
either your pi is really busy doing something right now, or you have hardware issues (bad powersupply/SD card)

you have tried rebooting right?