nextcloud/nextcloudpi #508: nc-scan-auto is not being deactivated

jbeuke, 2018-05-19:
Background: What I am trying to achieve is a folder for scanned documents that is fed from an FTP server running on the same Raspberry Pi 3B+ as the nextcloud instance. The scanner uploads the file via FTP. An inotifywait script then moves the file to the nextcloud file folder and calls `sudo -u www-data php /var/www/nextcloud/occ files:scan --path="<path>"` in order to add the new file to the database.

Problem: For a first approach I activated nc-scan-auto via the web panel. Since I now found the occ solution, I deactivated the scan again via the web panel. But still `journalctl -r` shows, that a scan is being executed every five minutes. Can someone confirm this problem?

[report.txt](https://github.com/nextcloud/nextcloudpi/files/2019400/report.txt)

nachoparker, 2018-05-19:
Does the file `/etc/cron.d/ncp-scan-auto` exist?

jbeuke, 2018-05-19:
According to `ls -all /etc/cron.d/` it doesn't exist.

nachoparker, 2018-05-20:
what's the status of 

```
systemctl status nc-scan
```

nachoparker, 2018-05-20:
Yes, can you try to do it manually with these commands?

```
systemctl stop nc-scan.service
systemctl disable nc-scan.service
rm -f /etc/systemd/system/nc-scan.service
systemctl stop nc-scan.timer
systemctl disable nc-scan.timer
rm -f /etc/systemd/system/nc-scan.timer
```
The explanation to this is that we moved from systemd timers to cron (mainly for the docker container, doesn't have systemd), and we forgot to deal with this situation.

Thanks for your feedback

jbeuke, 2018-05-20:
Yep, that worked. Thanks for the help and this awesome project!

nachoparker, 2018-05-20:
thanks for your feedback! I will release an update to fix the issue for others

nachoparker, 2018-05-24:
fixed `v0.56.9`

thanks!

closed 2018-05-24 by nachoparker