nextcloud/nextcloudpi #510: nc-format-USB counts 5 USB devices when only one is present

magickalJ, 2018-05-20:
After unmounting and detaching the USB-Drive where ncdata was located (to format a bigger USB drive and subsequentially move ncdata to it) and attaching a FAT32-formatted USB drive with automount enabled, nc-format-USB refuses to format the new USB drive with the message 
`ERROR: Counted 5 devices. Please, only plug in the USB drive you want to format. `

Only this one USB drive with one partition is plugged in, however, apart from the SD card.

<details>
<summary>NextCloudPlus diagnostics</summary>

```

PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 136
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":3,"time":"2018-05-20T18:39:42+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#136","userAgent":"--","version":"13.0.2.1"}
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":2,"time":"2018-05-20T18:39:42+00:00","remoteAddr":"","user":"--","app":"no app in context","method":"--","url":"--","message":"Temporary directory \/media\/USBdrive\/ncdata\/tmp is not present or writable","userAgent":"--","version":"13.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 136
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":3,"time":"2018-05-20T18:39:43+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"file_get_contents(\/media\/USBdrive\/ncdata\/news\/config\/config.ini): failed to open stream: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Files\/Storage\/Local.php#214","userAgent":"--","version":"13.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 136
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":3,"time":"2018-05-20T18:39:43+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#136","userAgent":"--","version":"13.0.2.1"}
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":2,"time":"2018-05-20T18:39:43+00:00","remoteAddr":"","user":"--","app":"news","method":"--","url":"--","message":"Configuration invalid. Ignoring values.","userAgent":"--","version":"13.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 136
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":3,"time":"2018-05-20T18:39:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"mkdir(): Permission denied at \/var\/www\/nextcloud\/apps\/news\/lib\/AppInfo\/Application.php#118","userAgent":"--","version":"13.0.2.1"}
PHP Warning:  fileperms(): stat failed for /media/USBdrive/ncdata/nextcloud.log in /var/www/nextcloud/lib/private/Log/File.php on line 136
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":3,"time":"2018-05-20T18:39:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"fileperms(): stat failed for \/media\/USBdrive\/ncdata\/nextcloud.log at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#136","userAgent":"--","version":"13.0.2.1"}
{"reqId":"Ip6o7Qj6iS3SrPgVLW6Z","level":3,"time":"2018-05-20T18:39:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"chmod(): No such file or directory at \/var\/www\/nextcloud\/lib\/private\/Log\/File.php#137","userAgent":"--","version":"13.0.2.1"}
NextCloudPlus version  v0.56.6
NextCloudPlus image    NextCloudPlus_04-21-18
distribution           Raspbian GNU/Linux 9 \n \l
automount              yes
USB devices            sda
datadir                /media/USBdrive/ncdata (doesn't exist)
rootfs usage           2.0G/7.1G
swapfile               /var/swap
Nextcloud check        error
HTTPD service          down
PHP service            up
MariaDB service        up
Redis service          up
Postfix service        up
internet check         ok
port check 80          closed
port check 443         closed
IP                     192.168.2.**
gateway                192.168.2.1
interface              wlan0
certificates           [*******]
NAT loopback           yes
uptime                 37min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

magickalJ, 2018-05-20:
Okay, so I peeked into the shell script for nc-format-USB and decided to disable zram - and indeed, it works, simple as that. maybe add a check and disable zram before trying to format in case it is enabled?
would save a lot of time to (still yet) less savvy people like me. Thanks.

nachoparker, 2018-05-22:
thanks, this should be an easy fix

nachoparker, 2018-05-28:
fixed `v0.56.12`

closed 2018-05-28 by nachoparker