nextcloud/nextcloudpi #511: Web server after update only text mode (Ticket-507)

pilepalle, 2018-05-21:
Hy Guys,

thank you for fixing Problem 507 (nc-datadir not updating apache logs # 507).
After the fix, my web presentation will not work properly anymore. The web server only displays everything in text. Does anyone have an idea what that can be and how can I fix it?

[image]

Thanks

pilepalle, 2018-05-21:
I just noticed that when I open the Nextcloud via the local IP address the web view does not come to its 
full extent (icons and co). The view via DNS Web externally displays only the text version.

[image]

nachoparker, 2018-05-22:
either your pi is really busy doing something right now, or you have hardware issues (bad powersupply/SD card)

you have tried rebooting right?

chiyi4488, 2018-09-28:
Me too. I have the same question.

chiyi4488, 2018-09-28:
I think this is not a "hardware" problem, it SSL.
because if I add a new domain and connection to my ncp.It can normal display.

chiyi4488, 2018-09-28:
but if I update NC to 14.0.0 (13.0.6 to 14.0.0),I can normal display.
and reboot........ text mode :D~~~
why?

TomTurnschuh, 2018-09-28:
@chiyi4488: Could you please try with a different browser?
I once had this issue after a hardware change. With a different browser it worked – and later also with the original browser.

chiyi4488, 2018-09-29:
no, I can't. I change my browser to edge and IE. It same.

chiyi4488, 2018-09-29:
Wait, I used the update software built into the nextcloud to update (previously applied to ncp), the system is back to normal, no longer "text mode", very strange. I think this may be caused by some kind of error that nc-restore and nc-nextcloud-update have not found yet.

I have recovered nc before. (nc-restore)

TomTurnschuh, 2018-09-30:
>  I think this may be caused by some kind of error that nc-restore and nc-nextcloud-update have not found yet.

Well, maybe there is indeed an unresolved issue with nc-restore. However, I rather think it is a matter of browser cache. @chiyi4488, have you connected to your cloud with the mentioned browsers already before the restore, or only after? If you also did before, then deleting the browser cache could be a reasonable solution.

k3tan25, 2018-10-28:
I'm also getting 'text mode'. Not sure why. Every browser. Every device. And no, this has nothing to do with the power supply.

chiyi4488, 2018-10-28:
I am guessing that there may be a problem with the way nc-restore is restored. @k3tan25 Did you show this after restore?

k3tan25, 2018-10-28:
No, I haven't used the restore function at all. Freshly installed. Ran some updates for ncp and nc. Not sure what's triggered it.

chiyi4488, 2018-10-28:
According to my guess, if ncp replaces the nc file, it is very likely that something like this will happen. But we still don't know why this is happening.

@nachoparker Any ideas?

chiyi4488, 2018-10-28:
Maybe we should compare the ncp file with the nc white paper, but I didn't subscribe. I can't view it.

chiyi4488, 2018-10-28:
https://github.com/DecaTec/Nextcloud-Backup-Restore
https://portal.nextcloud.com/article/13/backup-restore-30.html

nachoparker, 2018-10-28:
this has nothing to do with backup and restore