nextcloud/nextcloudpi #515: "NextCloudPlus not yet initialized, trying again in a few seconds ..."

godlich, 2018-05-23:
My setup: NCP on raspberrypi with system and data on 128 GB SD card.

I installed NCP two days before and did a backup yesterday. After the backup, I was asked to create a new database, so I think I did the backup to the sd-card where my data was stored. 

I wanted to reinstall NCP, the setup screen with the passwords appears, but clicking on "activate" leads to:
"NextCloudPlus not yet initialized, trying again in a few seconds ..." 
Refreshing the page generates new passwords, but not to a successfull installation. I copied the image again with etcher to the same and another SD-card, but the error message remains.

raivtash, 2018-05-27:
I had the same error when I installed the package on a fresh debian stretch system. I tried with a 60 ssd and a 2T hard drive, both were installed with a fresh debian stretch and had same error.

nachoparker, 2018-05-27:
Make sure you are using a good power supply and good SD card, double make sure, because this works

godlich, 2018-05-27:
I used two different and new SD cards from Sandisk (128 GB) and a 3A charger.
I tried several times with the NCP image, which did not work. Then I downloaded debian stretch and installed via bash, whichs worked fine.

nachoparker, 2018-05-27:
think about it, thousands of users and this only happens to you?

I see this everyday, people never believe their HW is bad, but it is.

More to read

https://forum.armbian.com/announcement/1-1-check-power-supply-check-sd-card-and-check-other-people-experiences/

jomikn, 2018-06-19:
I had the same problems with a 128 GB Toshiba and a 64 GB Kingston SD-Card (both Class 10).
I tried the installation with a 32 GB Kingston and a 8GB Raspi B Rev.2 SD-Card and it worked instantly fine.

After Installation and Activation on one of the "small" SD cards I cloned it on the "bigger" ones and started the resizing manually (raspi-config --> advanced options --> expand filesystem).
Now everything works fine with the "big" SD cards.

wilz7, 2019-03-04:
Run
sudo ncp-config
to launch the terminal version of the configuration panel.
install nc-nextcloud
Scrolled down to nc-init and ran it.
you should able to login now .

hope this helps for all