nextcloud/nextcloudpi #517: After restart NextCloudPi installed on BerryBoot/external drive can not see data drive

pabaja, 2018-05-24:
After fresh install NextCloudPi 13.0.1 on Raspberry Pi3/BerryBoot/external USB drive, the sever works perfectly (except SSL certificate, but it is not the subject). But after a restart the data drive is not visible to NextCloud.
I made some investigation and I noticed that USB drive is mounted properly, but symbolic links are changed:
```
ls -l /media
drwxr-xr-x  7 root root 4096 Jan  1  1970 berryboot
drwxr-xr-x 21 root root  310 Mar 18 23:06 disk
drwxr-xr-x  1 root root   12 May 22 22:28 myCloudDrive
lrwxrwxrwx  1 root root   16 May 24 08:53 USBdrive -> /media/berryboot
lrwxrwxrwx  1 root root   11 May 24 08:53 USBdrive1 -> /media/disk
lrwxrwxrwx  1 root root   19 May 24 08:53 USBdrive2 -> /media/myCloudDrive
```
while before restart it looked somehow like this:
```
ls -l /media
drwxr-xr-x  1 root root   12 May 22 22:28 myCloudDrive
lrwxrwxrwx  1 root root   12 May 22 22:28 USBdrive -> /media/myCloudDrive
```
When I redirect USBDrive symblolic link back to /media/myCloudDrive everything works again, but restart reverts link to USBdrive -> /media/berryboot.

nachoparker, 2018-05-25:
When you have several drives, you need to use label names, instead of USBdriveX

They are there just for the very simple case of a non technical user with only one drive, so he/she doesn't need to know the label name.

Just use `/media/myCloudDrive` everywhere, instead of `/media/USBdrive`

closed 2018-05-25 by nachoparker

nachoparker, 2018-05-25:
you are more than welcome to improve the documentation ;)