nextcloud/nextcloudpi #518: SPDyn updater not working under Pi HypriotOS/Docker

Darren-Hill, 2018-05-27:
Running on a Pi3 under HypriotOS and docker, using NCP 0.56.10 .

I have a DDNS set up via spdyn.de, which is working fine and set up with their token etc. However if I try to use the updater in the NCP panel to keep the IP address updated I get the following:

`[ spDYN ]
/usr/local/etc/spdnsupdater/spdnsUpdater.sh: line 56: curl: command not found
/usr/local/etc/spdnsupdater/spdnsUpdater.sh: line 23: curl: command not found
[]
Restarting periodic command scheduler: cronStopping periodic command scheduler: cron.
Starting periodic command scheduler: cron.`

The spdyn dashboard also shows that the token has never been used, so it looks like the script (or the container) is missing some libraries or something to allow it to work (curl seems to be missing?).

nachoparker, 2018-05-27:
ideally the script would be adapted to use wget... feeling brave @Darren-Hill ?

xD

Darren-Hill, 2018-05-27:
If I start a terminal into the container I can't find an editor that I can use (and I'm only a noob who likes nano anyway). Otherwise it could be an interesting challenge, at least if I wasn't going away for 2 weeks tomorrow on business too.

So I think I'll have to pass on your kind offer and leave it to your good self ;)