nextcloud/nextcloudpi #523: Armbian Pine64 SD Card Composing Stalled at ncp-update

lovenemesis, 2018-05-31:
Hi,

Recently I migrated my RPi2 based nextcloudpi to Pine64+ 2G. Following the generic instruction on how to install nextcloudpi on debian stretch, it installed well on the next branch Armbian based Pine64.
It works much better than RPi2 with its dedicate USB chip and GBit network, and consumes less power than the similar armv8 based RPi3.

Seeing calling for images for various Armbian SBC in forum, I decided to compose an nextcloudpi ready Pine64 image. However, it seems to be stalled somewhere.

I mostly follow the instruction on below page: https://ownyourbits.com/2017/12/04/nextcloudpi-comes-nc-12-0-4-odroid-hc1-and-armbian-support-btrfs-snapshots-diagnostics-tools-and-an-awesome-case/

The "build-SD-armbian.sh" script is modified a bit to parse "BRANCH=next" instead of "BRANCH=default" as for Pine64 that's the only kernel debian stretch supported. 

Build Environment Info:

```
> docker version
Client:
 Version:      18.03.1-ce
 API version:  1.37
 Go version:   go1.9.5
 Git commit:   9ee9f40
 Built:        Thu Apr 26 07:19:31 2018
 OS/Arch:      linux/amd64
 Experimental: false
 Orchestrator: swarm

Server:
 Engine:
  Version:      18.03.1-ce
  API version:  1.37 (minimum version 1.12)
  Go version:   go1.9.5
  Git commit:   9ee9f40
  Built:        Thu Apr 26 07:23:21 2018
  OS/Arch:      linux/amd64
  Experimental: false
```

The standard armbian image composing looks fine, but when comes to blend nextcloudpi in, it stalls at below step:

```
qemu: Unsupported syscall: 278
--2018-05-31 23:04:50--  https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/bin/ncp-update
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.40.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.40.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 1361 (1.3K) [text/plain]
Saving to: ‘/usr/local/bin/ncp-update’

/usr/local/bin/ncp-update             100%[=======================================================================>]   1.33K  --.-KB/s    in 0s      

2018-05-31 23:04:52 (2.99 MB/s) - ‘/usr/local/bin/ncp-update’ saved [1361/1361]

Downloading updates
qemu: Unsupported syscall: 278
```
It has been there for quite a while and according to system monitor, no obvious bandwidth consumption thus it's NOT downloading anything. This is already the 3rd trial and the last two times it stalled at the same step.

Looking at the bash of ncp-update, there isn't much as the problematic step. I also checked the qemu message but according to https://github.com/armbian/build/issues/572 that's nothing wrong.
Also, since post-composing on already created armbian image succeed, I cannot say much wrong of ncp-update either.

So I'm clueless on how to get around this issue to generate a nextcloudpi ready pine64 image.

nachoparker, 2018-06-01:
hey, thanks for the help!

well, that used to happen to me for a while, and I couldn't understand what whas going on. There is some problem with the chroot and the `git` command gets stuck.

for some reason it doesn't happen to me anymore, and it doesn't happen everytime... does it happen everytime? what happens if you try a couple more times?

I thought it was some armbian issue that went away, but I guess it is still there

Let's get those images going! thanks again!

lovenemesis, 2018-06-01:
In some way it's good to know I'm not the only one who had this problem before.

This is the 3rd time I tried and I guess it's no harm to give it 4th.

Report back tomorrow.

nachoparker, 2018-06-04:
I really would like to know why this happens sometimes to some people. It just stopped happening to me. Are you running Arch Linux?

nachoparker, 2018-06-11:
Any luck so far? what distro are you using?

nachoparker, 2018-06-23:
the qemu warning is normal. It is the `git` command that is getting stuck for whatever reason. Must be some kernel / qemu race condition >_<

so, do you have an image? :D

nachoparker, 2018-06-23:
https://github.com/nextcloud/nextcloudpi/blob/master/bin/ncp-update#L15

nachoparker, 2018-09-23:
Hi, I changed the build system to be fully local, so it doesn't need to use git at all. This has the nice side effect of avoiding this issue.

closed 2018-09-23 by nachoparker

kashkaj, 2018-09-30:
hello people, iam new to nextcloud, i have also one pineA64 board. does any one have here the right nextcloudpi image for pineA64? thanks

nachoparker, 2018-09-30:
@kashkaj I can make you one, if you help me by trying it out every time there is a release (every 3 or 4 months) so we can have it as "supported"

kashkaj, 2018-10-01:
> @kashkaj I can make you one, if you help me by trying it out every time there is a release (every 3 or 4 months) so we can have it as "supported"

Hello, that wold be great ;) . yesterday i tried to build it on my own, but in the end i got this message

**[ Pack Image ]
packing tmp/NextCloudPi_pine64_09-30-18.img → output/NextCloudPi_pine64_09-30-18.tar.bz2
NextCloudPi_pine64_09-30-18.img
/bin/sh: 1: pbzip2: not found
tar: output/NextCloudPi_pine64_09-30-18.tar.bz2: Cannot write: Broken pipe
tar: Child returned status 127
tar: Error is not recoverable: exiting now**

I found the image file in teh temp directory, today i will try it out, i hope the only think that doesnt work was the packing process, so the image is OK. If it doesnt work, it wold be great if you could build the image for me. 

thanks

nachoparker, 2018-10-02:
You are just missing the `pbzip2` tool, used to compress the image, but yes, you have the uncompressed version in `/tmp`

what terminal command did you use to build?

kashkaj, 2018-10-02:
I use the **./build-SD-armbian.sh pine64** to build the image. the only thing i changed was the **BRANCH=default** in the code to **BRANCH=next**, as **LOVENEMESIS** did. as i said iam not veryy experienced in linux. Thank y ou for your help.

nachoparker, 2018-10-02:
You can try `default`, but it will only build if it's Debian 9 Stretch.

What I would do is to do a basic Armbian install (no NCP) and find a combination of branch/board that works for you.

Then using those parameters you can generate a NCP image

https://docs.armbian.com/Developer-Guide_Build-Preparation/