nextcloud/nextcloudpi #524: Blank page on https but webui accessible

Olivl, 2018-06-02:
### PROBLEMS

Hello,
After the last upgrade, i've problems with nextcloudpi, i get an error 500 in the browser but i can access to the web-ui (443).
mysql is running, apache2 too.

My data and database is in an usbdrive

When i run a nc-backup i have error "access denied for user ncadmin"
The password is correct in config and mysql

When i try to run a fix-permission, it return me "data dir not found"

Can you help me to find how to resolve this problem?

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.56.16
NextCloudPi image    NextCloudPi_02-06-18
distribution         Raspbian GNU/Linux 9 \n \l
automount            yes
USB devices          sda 
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      12G/448G
rootfs usage         1,8G/15G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.89
gateway              192.168.1.1
interface            eth0
certificates         cloud.****************.com
NAT loopback         no
uptime               29min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: YES) in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(618): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\\IDBConnecti...')
[... 84 lines cut ...]
2018-06-02 15:25:18 1989001216 [Note] InnoDB: Compressed tables use zlib 1.2.8
2018-06-02 15:25:18 1989001216 [Note] InnoDB: Using Linux native AIO
2018-06-02 15:25:18 1989001216 [Note] InnoDB: Using generic crc32 instructions
2018-06-02 15:25:18 1989001216 [Note] InnoDB: Initializing buffer pool, size = 128.0M
2018-06-02 15:25:18 1989001216 [Note] InnoDB: Completed initialization of buffer pool
2018-06-02 15:25:18 1989001216 [Note] InnoDB: Highest supported file format is Barracuda.
2018-06-02 15:25:19 1989001216 [Note] InnoDB: 128 rollback segment(s) are active.
2018-06-02 15:25:19 1989001216 [Note] InnoDB: Waiting for purge to start
2018-06-02 15:25:19 1989001216 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 471505872
2018-06-02 15:25:19 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-06-02 15:25:19 1989001216 [Note] Plugin 'FEEDBACK' is disabled.
2018-06-02 15:25:19 1989001216 [Note] Server socket created on IP: '127.0.0.1'.
2018-06-02 15:25:19 1989001216 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

```
</details>

nachoparker, 2018-06-04:
what is the output of 

```
systemctl status nc-provisioning
```

Olivl, 2018-06-04:
Hi,

Here is the output : 

`● nc-provisioning.service - Randomize passwords on first boot
   Loaded: loaded (/usr/lib/systemd/system/nc-provisioning.service; enabled; vendor preset: enabled)
   Active: inactive (dead) since Mon 2018-06-04 16:20:23 CEST; 20min ago
  Process: 1296 ExecStart=/bin/bash /usr/local/bin/ncp-provisioning.sh (code=exited, status=0/SUCCESS)
 Main PID: 1296 (code=exited, status=0/SUCCESS)

juin 04 16:20:23 nextcloudpi systemd[1]: Started Randomize passwords on first boot.
juin 04 16:20:23 nextcloudpi bash[1296]: Updating NextCloud config with Redis password
juin 04 16:20:23 nextcloudpi bash[1296]: Updating NextCloud config with MariaDB password`

BatPio, 2018-06-06:
@Olivl 

Have you found the solution? I have the same problem.

nachoparker, 2018-06-11:
For whatever reason, your database access is not right. What happens if you run

```
systemctl start nc-provisioning
systemctl status nc-provisioning
cat /root/.my.cnf
cat /var/www/nextcloud/config/config.php
```

Does it still not work? Can you share the output?

BatPio, 2018-06-11:
Strange, it looks that update changed ncadmin password or config.php .
I changed ncadmin password to same as is set in /root/.my.cnf and now it work.

nachoparker, 2018-06-11:
does it still work after a reboot?

closed 2018-06-11 by nachoparker

BatPio, 2018-06-11:
Yes, I restarted my raspberry and nc still works_