nextcloud/nextcloudpi #527: Configure system limits not working

perryclements, 2018-06-04:
The following error occurs when running Configure system limits for NextCloudPi (Docker)
./nc-limits.sh: line 42: systemctl: command not found

Changing line 42 in nc-limits.sh to this fixes the problem.
-  systemctl restart redis
+  bash -c " systemctl restart redis
+            " &>/dev/null &

nachoparker, 2018-06-11:
Hi,

Indeed, I must have missed that, because the docker container doesn't come with systemctl or systemd.

The solution is to issue the command to restart redis without using systemctl.

This is a good first contribution.

Thanks for your feedback!

perryclements, 2018-06-13:
I think systemctl is included in the docker container because all I did to fix the problem was to put bash -c in front of the command, put quotes around the command, and add the null output at the end. This is how the other commands in the file are executed.

nachoparker, 2018-06-18:
fixed `v0.57.1`

closed 2018-06-18 by nachoparker