nextcloud/nextcloudpi #528: static IP adress with nc-static-ip on an Odroid HC2 failed

clown522, 2018-06-06:
Hi,
thank you for your hard effort! This helped me a lot to create my own NextCloud-Server.

I had some problems with the static ip with your newest nexcloudpi image (v0.56.15), because he did not found eth0.

i decided to search with: 

1. ifconfig -a

my controller is called: enx001e06328ba8 instead of eth0, so i edited this file:

2. sudo nano /etc/network/interfaces

3. i replaced everything with eth0 to enx001e06328ba8.

4. sudo reboot

Maybe somebody else has this problem too with Odroid.

nachoparker, 2018-06-11:
Hi @clown522 

Indeed, eth0 is [hardcoded](https://github.com/nextcloud/nextcloudpi/blob/master/etc/ncp-config.d/nc-static-IP.sh#L78) in the current version.

We would need to detect it for each different system instead. This seems like a good first contribution if anyone is interested ;)

Again, thanks for the valuable feedback!

nachoparker, 2018-06-18:
fixed in `v0.57.2`

closed 2018-06-18 by nachoparker