nextcloud/nextcloudpi #532: ncp-web login

defaultuser8, 2018-06-07:
In an attempt to harden my system i removed the user "pi" from my instance of nextcloud.  Now I can no longer log into the ncp-web login.  Anyone have this same issue?

Paste the output of `sudo ncp-report`

nachoparker, 2018-06-11:
shouldn't be related. The user that ncp-web uses is `ncp`. What error are you seeing?

defaultuser8, 2018-06-11:
Up until i deleted the user pi i was using pi and raspberry to log in. Now no matter what user i try i can't  log in.

Sent via the Samsung Galaxy Tab® 4, an AT&T 4G LTE tablet

"@type": "MessageCard",
"@context": "http://schema.org/extensions",
"hideOriginalBody": "false",
"originator": "AF6C5A86-E920-430C-9C59-A73278B5EFEB",
"title": "Re: [nextcloud/nextcloudpi] ncp-web login (#532)",
"sections": [
{
"text": "",
"activityTitle": "**nachoparker**",
"activityImage": "https://assets-cdn.github.com/images/email/message_cards/avatar.png",
"activitySubtitle": "@nachoparker",
"facts": [

]
}
],
{
"name": "Add a comment",
"@type": "ActionCard",
"inputs": [
{
"isMultiLine": true,
"@type": "TextInput",
"id": "IssueComment",
"isRequired": false
}
],
"actions": [
{
"name": "Comment",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueComment\",\n\"repositoryFullName\": \"nextcloud/nextcloudpi\",\n\"issueId\": 532,\n\"IssueComment\": \"{{IssueComment.value}}\"\n}"
}
]
},
{
"name": "Close issue",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"IssueClose\",\n\"repositoryFullName\": \"nextcloud/nextcloudpi\",\n\"issueId\": 532\n}"
},
{
"targets": [
{
"os": "default",
"uri": "https://github.com/nextcloud/nextcloudpi/issues/532#issuecomment-396180527"
}
],
"@type": "OpenUri",
"name": "View on GitHub"
},
{
"name": "Unsubscribe",
"@type": "HttpPOST",
"target": "https://api.github.com",
"body": "{\n\"commandName\": \"MuteNotification\",\n\"threadId\": 343735052\n}"
}
],
"themeColor": "26292E"
}