nextcloud/nextcloudpi #534: I wanna install nextcloud app passwords requires php 7.1

xelcho, 2018-06-08:
This nextcloud app is dependent on php 7.1, nextcloudplus includes php7.0. Please advise as to when to upgrade to 7.1.

Paste the output of `sudo ncp-report`

ovpc, 2018-06-10:
I think (one) answer to that is [here](https://raspberrypi.stackexchange.com/questions/70388/how-to-install-php-7-1) 
I would not recommend implementing it without testing it first on non-production environment.
Not specifically NCP related tho, so help.nextcloud.com would be more appropriate, thank you for reporting.

nachoparker, 2018-06-11:
Yes, it's better to discuss this in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm).

We **could** try to get PHP 7.1 or PHP 7.2 working in NCP, but there's some work involved, plus risk of unstability / things breaking, so I am not sure about the benefits.

Thanks @ovpc

closed 2018-06-11 by nachoparker