nextcloud/nextcloudpi #535: Docker image - ncp-backup does not work

ushills, 2018-06-09:
With a docker instance created with 

```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /root/newcloudpi:/data --name newcloudpi ownyourbits/nextcloudpi-armhf 192.168.1.20
```

I have created a backup of both the configuration and database & data giving two files, both created in a persistent volume.

```
ncp-config_20180609.tar
nextcloud-bkp_20180609_1528536439.tar 
```
If I stop the running container:
```
docker stop container newcloudpi
```

and launch a new instance, but using the same persistant volume:
```
docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /root/newcloudpi:/data --name newcloudpi2 ownyourbits/nextcloudpi-armhf 192.168.1.20
```

I get the following error from the docker log:

```
root@DietPi:~# docker logs -f newcloudpi2
Making /etc/shadow persistent ...
Starting PHP-fpm
Starting Apache
AH00526: Syntax error on line 5 of /etc/apache2/sites-enabled/ncp.conf:
SSLCertificateFile: file '/etc/letsencrypt/live/nextcloud.raspberryhills.co.uk/fullchain.pem' does not exist or is empty
Action 'start' failed.
The Apache error log may have more information.
Provisioning
Provisioning Redis password
Updating NextCloud config with Redis password
Provisioning MariaDB password
ERROR 2002 (HY000): Can't connect to local MySQL server through socket '/var/run/mysqld/mysqld.sock' (111 "Connection refused")
Updating NextCloud config with MariaDB password
PHP threads set to 4
Starting Redis
Starting Cron
Starting Postfix
An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [2002] Connection refused in /data/app/lib/private/DB/Connection.php:64
Stack trace:
#0 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /data/app/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /data/app/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /data/app/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /data/app/lib/private/Server.php(628): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
#15 /data/app/lib/private/Server.php(1511): OC\ServerContainer->query('DatabaseConnect...')
#16 /data/app/lib/private/Server.php(332): OC\Server->getDatabaseConnection()
#17 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#18 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\\Authenticati...')
#19 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
#20 /data/app/lib/private/Server.php(336): OC\ServerContainer->query('OC\\Authenticati...')
#21 /data/app/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#22 /data/app/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OC\\Authenticati...')
#23 /data/app/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OC\\Authenticati...')
[... 224 lines cut ...]
<summary>Database logs</summary>

2018-05-30  8:27:14 1995669504 [Note] InnoDB: Setting log file ./ib_logfile101 size to 48 MB
2018-05-30  8:27:18 1995669504 [Note] InnoDB: Setting log file ./ib_logfile1 size to 48 MB
2018-05-30  8:27:24 1995669504 [Note] InnoDB: Renaming log file ./ib_logfile101 to ./ib_logfile0
2018-05-30  8:27:24 1995669504 [Warning] InnoDB: New log files created, LSN=1616829
2018-05-30  8:27:24 1995669504 [Note] InnoDB: Highest supported file format is Barracuda.
2018-05-30  8:27:24 1995669504 [Note] InnoDB: Read redo log up to LSN=1682432
2018-05-30  8:27:24 1995669504 [Note] InnoDB: 128 rollback segment(s) are active.
2018-05-30  8:27:24 1995669504 [Note] InnoDB: Waiting for purge to start
2018-05-30  8:27:24 1995669504 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.36-82.1 started; log sequence number 1616908
2018-05-30  8:27:24 1396699968 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-05-30  8:27:24 1995669504 [Note] Plugin 'FEEDBACK' is disabled.
2018-05-30  8:27:24 1995669504 [Note] Recovering after a crash using tc.log
2018-05-30  8:27:24 1995669504 [Note] Starting crash recovery...
2018-05-30  8:27:24 1995669504 [Note] Crash recovery finished.
2018-05-30  8:27:24 1995669504 [Note] Server socket created on IP: '127.0.0.1'.
2018-05-30  8:27:24 1995669504 [ERROR] mysqld: Table './mysql/user' is marked as crashed and should be repaired
2018-05-30  8:27:24 1995669504 [Warning] Checking table:   './mysql/user'
2018-05-30  8:27:24 1995669504 [ERROR] mysql.user: 1 client is using or hasn't closed the table properly
2018-05-30  8:27:24 1995669504 [Note] mysqld: ready for connections.
Version: '10.1.26-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.1

</details>
```

ushills, 2018-06-10:
Secondly, if you create a new instance with a different mount point and try to restore a backup.

```docker run -d -p 4443:4443 -p 443:443 -p 80:80 -v /root/**newcloud2**.raspberryhills/:/data --name **newcloudpi2** ownyourbits/nextcloudpi-armhf 192.168.1.20
```

after copying the backup into the volume you get the following error on restore.

```
[ nc-restore ]
INFO: overwriting old instance
extracting backup file /data/backups/ncp-backups/nextcloud-bkp.tar...
restore files...
redis: unrecognized service
Cleanup...
```

nachoparker, 2018-06-11:
Hi,

Mmmm, the certificate lives at `/etc/letsencrypt/live`, so we have to make that persistent. I am surprised it wasn't already, but it is an easy fix.

Your second message should be a separate issue.

Thanks for your feedback.

testbird, 2018-06-17:
Is this fixed already? https://github.com/nextcloud/nextcloudpi/commit/80492b3370095c769b6d790101c27612de80d5d4#diff-d9150e22ce39c8dd69db531a6110701a

nachoparker, 2018-06-17:
Yes, but still haven't landed in the docker images

nachoparker, 2018-06-17:
@testbird thanks for your recent help with documentation! I rolled back the changes about dnsmasq, because you need to set the IP of the raspi, not the router.

The idea is that you point to the raspi to resolve your domain locally correctly.

Thanks again for the edits!

testbird, 2018-06-17:
Yea, the clients need to query the dnsmasq of NCP (that returns its own local IP for the DDNS name).  

But the NCP may just query the router for all other domains. That IP, and the whole dnsmasq could actually be auto configured, there is not need to enter ISP's or 8888 dns servers.

testbird, 2018-06-17:
I see, it collides with option 2.1

testbird, 2018-06-17:
Make sure to inform the users about the GDPR regulation impacts of the default 8888.

nachoparker, 2018-06-17:
mmm I am not sure about those :S

testbird, 2018-06-17:
Usually, the correct thing to do is simply: Not saving or providing data to additional parties => no problem.
(Fortunately, a principle that stays.)

testbird, 2018-06-17:
Oh, i think if option 2.1 only sets the dns server address that gets distributed by the routers' dhcp to point to NCP (and not setting the "uplink" of the routers dns to the NCP) it should not collide. That should work, and allow autoconfiguring the NCP dnsmask to dns query the router.

nachoparker, 2018-06-18:
Yes, if we have the option to use 2.1 we really don't need to setup dnsmasq on the pi so I guess that the wiki is a bit inaccurate here. I think something like this would be better

- option 1: configure an entry in your routers DNS to point to your pi
- option 2: install dnsmasq, and point all your devices to the pi (primary DNS only).

In option 2, I don't know what is the best external resolver to use. The default is 8.8.8.8 for simplicity, but opinions are welcome. We have 1.1.1.1 and idk if we could extract the router provided ISP DNS somehow.

testbird, 2018-06-18:
Right, first... (option 1), otherwise... (option 2). That is what I intended with the edit.

> extract the router provided ISP DNS

It shouldn't be necessary to extract, simply (and automatically) point the dnsmasq on NCP to the user's router IP for resolving of the router managed and external domains. (No need to get involved with further third-party dns servers)

I mistakenly thought it collides (creates a loop). This would only happen if at the same time directing the router itself to the NCP dnsmask, which should not be done. (Instead of directing only the clients to NCP.)

nachoparker, 2018-06-18:
Yes, so it would be better to change it because option 1 doesn't need to setup dnsmasq and that's not how it reads right now.

So the change would be to automatically point to the default gateway instead of 8.8.8.8 or anything like that, and we would make it easier by not presenting that option to the user. Sounds good to me, even though it is a bit less flexible for people that know what they are doing.

We are highjacking this thread, so maybe open a change request to keep talking about this?

nachoparker, 2018-06-19:
fixed and tested. I will push it soon with the new release, probably within a week. If you can't wait, you can build the docker container yourself ;)

closed 2018-06-19 by nachoparker