nextcloud/nextcloudpi #536: Not working on RPi 3b+

dmmop, 2018-06-09:
The NextCloudPi image have an old boot version, and the rpi 3b+ does not bot without the latest version of this.

ovpc, 2018-06-09:
Hi and thanks for reporting.
Could you let us know which version of NCP-berryboot you are using and which version of berryboot?

dmmop, 2018-06-09:
That's the point, there aren't any output because the raspberry does not bot with the latest image of berryboot.
The problem is the boot system version, in this berryboot is "old". And it is not supported for raspberry pi 3b+.

ovpc, 2018-06-09:
At berrterminal [website](https://www.berryterminal.com/doku.php/berryboot)There are two berryboot versions. Which one did you use? And which NCP image? Which procedure did you follow? Try to be specific. ""Not working on RPi3b+"" is just not enough information to work on.  I have a RPi3b+ that boots fine with [this version](http://downloads.sourceforge.net/project/berryboot/berryboot-20180405-pi2-pi3.zip) of berryboot. Once booted you load NCP, basic Rasbian or an other OS.

nachoparker, 2018-06-19:
If anyone is reading this: we are looking for a new maintainer for the latest berryboot version.

It seems that something changed and the current [build script](https://github.com/nextcloud/nextcloudpi/blob/master/build-SD-berryboot.sh) does not work anymore.

We would need someone to

- fix the script... shouldn't be too hard and we can help
- try every new NCP release, at least superficially... as long as it boots it should be fine because all the code is shared with the regular rpi version.

TechToxic, 2018-06-23:
I can do this.

crossr1, 2018-06-27:
The issue has been duplicated. I am still researching the cause.

crossr1, 2018-07-06:
This issue has been fixed. I have created a new Wiki Page to help: https://github.com/nextcloud/nextcloudpi/wiki/BerryBoot-Instructions-for-NextCloudPi