nextcloud/nextcloudpi #538: automount problem?

sgiannulli, 2018-06-10:
Hello all,
I have ncp instance running on my Rpi3; database and data are on external usb hdd. I just added another usb hdd to store backups mainly but:
first hdd named as /media/usb has been renamed as /media/usb1 while the new hdd is named as /media/usb; of course this cause obvious problems as ncp does not find database and data dirs under /usb/media anymore as they are now under /media/usb1.
is this a problem with automount?
Thanks for your help

nachoparker, 2018-06-11:
Hi, If you are going to use two drives, you have to reference them by **label**, not by USBdrive1, USBdrive2.

Just use each device label everywhere. You can see this by

```
ls -l /media
```

closed 2018-06-11 by nachoparker

sgiannulli, 2018-06-11:
Hi,
Sorry but I cannot get your answer. Both disks have no label assigned. If I set a label what will change with automount? I gave a look at the code but cannot sort it out. 
Thank you

nachoparker, 2018-06-11:
if your drive label is `myCloudDrive` it will be mounted at `/media/myCloudDrive`

use that path in ncp-config, like nc-datadir