nextcloud/nextcloudpi #540: fail2ban email notification

clown522, 2018-06-13:
Hi @nachoparker,
i do not want to annoy you but is it possible that the email notification of fail2ban does not sent any mail?
fail2ban works now also at your web interface but without mailalert. (v 0.56.21)
Could you test it when you have time?
Thanks for fixing problems so fast!

nachoparker, 2018-06-14:
it worked last time I checked it, I'll see if I have time to test it again.

or can anybody confirm that it works? (I have it disabled)

nachoparker, 2018-07-17:
Hi, @crossr1 kindly tried to reproduce this issue, and he was not able to.

fail2ban email notifications work fine for him. I will close this now. Please, reopen with details on how to reproduce if the problem still persists.

Thaks for your feedback

closed 2018-07-17 by nachoparker

sonicwave2020, 2018-08-02:
Had these logs in mail log. 
/etc/aliases.db no such file or directory

nachoparker, 2018-08-04:
thanks for reporting.

Maybe @crossr1 or @ovpc can confirm whether they also have that `aliases.db` error in their logs.

Have you checked fail2ban logs?

ovpc, 2018-08-05:
No aliases.db error here  (curl install on Debian9-NUCBOXi5) and 
I am receiving couple emails a day whenever ip's are banned, but I am in habit of installing ssmtp and bsd-mailx  and that install removes postfix.

sonicwave2020, 2018-08-05:
I have checked fail2ban logs. 
I did see where my test device’s ip was banned when I purposely did a wrong NC login 6 times in a row. 
After 6 attempts it would not let me log in from that device’s ip as I was shown a blank web page.  After the allotted time, that ip was allowed again. 
I also did that test again & was able to unban a test device’s ip manually & that feature worked too. 
I saw fail2ban log entries for both of those tests. I did not see in fail2ban, where it attempted any email, but maybe it doesn’t show that there.

nachoparker, 2018-08-05:
maybe try purging and reinstalling postfix? see if the aliases.db error goes away

sonicwave2020, 2018-08-06:
Ok. Any tips on how to do that? Or would a quick web search be easy enough?

nachoparker, 2018-08-06:
```
apt purge postfix
apt install postfix
```

sonicwave2020, 2018-08-07:
postfix purged & then installed. will test later today

sonicwave2020, 2018-08-07:
Tested by entering wrong password 6 times in a row on a device. Device appears to be banned, because it will not show login page now. 
I wasn’t sure what to put in prompts when doing postfix install. I may have entered wrong options. 
I chose option 2- I think it was “website plus” & put in my email provider’s domain, then changed the “NextCloudPi” to my email address’s user name. 
I went to NCP-config webUI & double checked that fail2ban is active & email is entered & checked to send mail. I can check logs later today.

nachoparker, 2018-08-07:
great, thanks. What about the aliases.db error in the logs. Still there?