nextcloud/nextcloudpi #541: [docker] Panel password doesn't persist

lil5, 2018-06-14:
After setting everything up

docker-compose.yml
```yml
version: '3'

services:
  ncp:
    image: ownyourbits/nextcloudpi-armhf
    restart: unless-stopped
    volumes:
      - /mnt/Data/ncp:/data
    ports:
      - 443:443
      - 4443:4443
```

I run the following:

```
docker-compose down
docker-compose up -d
```

...and I can't log in to the Panel:
(using the previous default user: `ncp`, pass: [44 char random] )

nachoparker, 2018-06-14:
Hi,

can you login using `ownyourbits` instead of your random password?

lil5, 2018-06-14:
Yes :frowning:

nachoparker, 2018-06-14:
Ok, so something must be wrong with [this](https://github.com/nextcloud/nextcloudpi/blob/master/docker-common/nextcloudpi/000ncp#L9).

Do you think you can have a look at what is going on? `/etc/shadow` should be a symlink to `/data/etc/shadow` (or similar) after the container has started, therefore persisting the password.

Thanks for reporting

lil5, 2018-06-14:
@nachoparker 
> Do you think you can have a look at what is going on? /etc/shadow should be a symlink to /data/etc/shadow (or similar) after the container has started, therefore persisting the password.

Yes:

```
ls -alh /etc/shadow
lrwxrwxrwx 1 root root 16 Jun 14 13:51 /etc/shadow -> /data/etc/shadow
```

lil5, 2018-06-14:
```
/etc $ ls -ahl
total 4.0K
drwxr-xr-x 1 root root    62 Jun 14 13:51 .
drwxr-xr-x 1 root root    54 Jun  7 16:39 ..
drwxr-xr-x 1 root root   228 May 30 08:17 apache2
drwxr-xr-x 1 root root    50 May 30 09:35 services-enabled.d
-rw-r----- 1 root shadow 794 May 30 09:34 shadow
```

shadow is owned by `root` could that be blocking any password writing?

nachoparker, 2018-06-15:
Does `/data/etc/shadow` have the original contents corresponding to the password `ownyourbits` or the changed contents?

Thanks!