nextcloud/nextcloudpi #543: Apache default page after reboot

gembutterfly, 2018-06-15:
Hi,

after a reboot of my nextcloupi installation, the cloud service is not available anymore. <my raspberry ip>:4443 does nothing and <my raspberry ip> only shows the Apache default page. Login via SSH throws this error message:

```
NextCloudPi v0.56.17 is outdated
update to v0.56.22 through 'ncp-config' or type 'sudo ncp-update'
Could not chdir to home directory /nonexistent: No such file or directory
This account is currently not available.
Connection to 192.168.0.104 closed.
```

Is there any way to repair my installation without losing my data? This could have happened due to a power outage.

Cheers,
Gem

gembutterfly, 2018-06-15:
Nevermind. I had to login with pi@<my raspberry pi id>.

nachoparker, 2018-06-18:
Seems like you installed another web server and apache2 has no available ports. We cannot support this as it is not a NCP issue. Better try at the [forums]https://help.nextcloud.com/c/support/appliances-docker-snappy-vm).

If I am wrong, just explain ;)