nextcloud/nextcloudpi #547: refined local dns option

testbird, 2018-06-18:
Coming from https://github.com/nextcloud/nextcloudpi/issues/535#issuecomment-397986121
> Yes, so it would be better to change it because option 1 doesn't need to setup dnsmasq and that's not how it reads right now.
> 
> So the change would be to automatically point to the default gateway instead of 8.8.8.8 or anything like that, and we would make it easier by not presenting that option to the user. Sounds good to me, even though it is a bit less flexible for people that know what they are doing.

Ok, I changed the wiki accordingly (to currently set the router IP manually)

testbird, 2018-06-18:
May be just change the defaults to be preset with the DDNS domain and the router IP instead of removing the fields?

nachoparker, 2018-06-18:
I just saw your edits, look great thanks!

> Maybe just change the defaults to be preset with the DDNS domain and the router IP instead of removing the fields?

Unfortunatedly the tool right now cannot propose dynamic defaults, only 'premade' defaults can be presented to the user. Maybe after the overhaul we will be able to do this. I'll write a note in the relevant issue

testbird, 2018-06-18:
Command substitution is not available? A nice one: `read _ _ gateway _ < <(ip route list match 0/0)`
https://stackoverflow.com/questions/1204629/how-do-i-get-the-default-gateway-in-linux-given-the-destination

nachoparker, 2018-06-18:
well, the options presented as a default are taken literal from text and are not 'scriptable'