nextcloud/nextcloudpi #548: Odroid HC2 - No NoIp & letsencrypt via setup + No SSH (putty) acsess  after password change

butterbrot21, 2018-06-18:
Hi, after hours of trying this NCP does not work on my Odroid HC2:
I can't connect to noip with the setup wizard. Ports 443 and 80 are set to 100% and the DNS is also set on the router (Fritbox). With my mobile connection I will also be forwarded correctly to my dyndns. I still can't activate noip in web gui. Furthermore I can't activate Letsencrypt either. After access via ssh I am asked to enter a new PW which I do. After that when I log in with the new PW putty aborts. Everything worked with my Rpi back then, but somehow nothing works with the odroid hc2. I had read the problem with Noip several times now. i would be very pleased about help and suggestions

nachoparker, 2018-06-18:
Hi,

Probably those issues are configuration/network issues. Please, use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) for this kind of help request as there's more people there that will help.

Also you need to provide the output of `sudo ncp-report` so we can help, as indicated in the instructions.

Cheers,

closed 2018-06-18 by nachoparker

butterbrot21, 2018-06-18:
And the next reinstall: Ports are open a and i can connect to my duckdns adress via lan and wan. but unfortunatelly i cant setup letsencrypt again! 

[ letsencrypt ]
Saving debug log to /var/log/letsencrypt/letsencrypt.log
Plugins selected: Authenticator webroot, Installer None
An unexpected error occurred:
ConnectionError: HTTPSConnectionPool(host='acme-v01.api.letsencrypt.org', port=443): Max retries exceeded with url: /directory (Caused by NewConnectionError(': Failed to establish a new connection: [Errno -2] Name or service not known',))
Please see the logfiles in /var/log/letsencrypt for more details.

---> no inet? this is the systeminfo msg:

internet check | no }
-- | --
port check 80 | closed
port check 443 | closed

--> 
There is something wrong with the hc2 image. I have a 100% internet connection. I hope this will change, because the hc2 is headless and not everyone can set up the inernet connection via ssh (I don't either)

butterbrot21, 2018-06-18:
thanks a lot for the answer. i will switch to dietpi or omv first, because i don't have much time for troubleshooting at the moment and i think it has to do with the hc2 image, or do i have to configure the inet/lan connection manually? 

Too bad, I was really excited about ncp with my rpi. 

Keep up the good work. This is a really cool projec!

nachoparker, 2018-06-18:
I checked the image, and the issue is that (for whatever reason) the DNS is not updated through DHCP. This must be an Armbian issue.

just add 1.1.1.1 or your ISP DNS if you know it to resolv.conf

cat "nameserver 1.1.1.1" >> /etc/resolv.conf 

The wizard is there, look at the top bar in ncp-web.

In any case, I understand your frustration but you also have to understand that I am a single volunteer and if we are going to improve NCP for everyone together I need patience and good feedback. If you look at your original message... little can be done from there, there is no debug info whatsoever. That's why we request the output of ncp-report.

I'll have a good look at the image when I have time, pretty busy right now.

Thanks,

nachoparker, 2018-06-18:
update: letsencrypt works, noip works, SSH works. Wizard didn't come up automatically but it's there. I'll investigate that.

nachoparker, 2018-06-18:
created #549

butterbrot21, 2018-06-18:
thank you very much for your help and detailed answer. I agree with you completely, unfortunately my post is written very badly and with no helpful info/logs. In the future I will calm down before posting/writing, I was quite under stress. 

you're doing a greatjob, and you deserve a lot of respect here!!

Unfortunately I couldn't read and post Error-logs, because when I access my hc2 for the first time via ssh and putty.
I should change the pw, here, I took user ncp and then a new pw, after that there is the automatic reboot(putty). If I now try to login with ncp and the new pw, putty closes and the connection breaks. According to the forum others also have the problem with our odroidhc2, do you think this also has to do with the DNS? I used the webUi to aktivate ssh.

Thanks again for your time and effort!

nachoparker, 2018-06-18:
We're here to help... if we are patient everything gets solved eventually.

In the armbian image you have, SSH is active by default (this is the default Armbian behaviour). That will change in the next release, and SSH will be disabled by default, requiring ncp-web to activate it.

The problem is that you chose user `ncp`. That user is only for ncp-web and it is not allowed to use SSH for security reasons. Just choose **any** other username.

Again, in the future, more people will be able to help you in the forums and I will have more time to fix things ;)

butterbrot21, 2018-06-19:
Nice, thnak you! that sounds good! 

I can remember that I tried it first with user pi, but that was rejected (User Pi does not exist).
I will try it again with a completely different name. (not ncp or pi ;))

Is it the standart user and password from armbian ( root + pw=1234)?

I'm going to load ncp again on my HC2 tomorrow.
Thanks again for the quick help!

nachoparker, 2018-06-19:
> Is it the standart user and password from armbian ( root + pw=1234)?

yes, that is the default Armbian credentials

I am preparing the next release... if you can't make it work you might have to wait a few days until I announce the release... if you want I can tell you early and you can have a look at it, that would help me a lot ;D

Kunibert-007, 2018-06-19:
Hi Nacho, thanks for your great work!
I have a Odroid XU4 and took the latest image from your site and have the same problem with no internet connection. The string echo "nameserver 1.1.1.1" >> /etc/resolv.conf  helps until I reboot the system. 
But there is another problem, too. When I try to get a certificate with letsencrypt I get this message:

[ letsencrypt ]
./letsencrypt.sh: line 60: letsencrypt: command not found

You know what I can do?

And if you need a beta tester - I'll be there :-)

butterbrot21, 2018-06-19:
It would be a honour to me! Let me know when the time comes :)
Keep up the good work!

Kunibert-007, 2018-06-20:
Ok, self-fixed the issue with let's encypt.
I installed the certbot with `apt-get install python-certbot-apache -t stretch-backports` and now I could sign a certificate for the webserver. I think the package isn't integrated in the actual image, right?

nachoparker, 2018-06-20:
@butterbrot21 @Kunibert-007 thanks guys!

I am basically alone here, and that's why the image had issues. It would be great to be able to count with some help.

It is a great help to at least running the image and checking that it boots, there are no warnings in ncp-web. I started to write some automated tests (in the `test` folder), but that will have to wait for now, because I don't have time to improve/fix them.

Ideally, I would generate all images, and they would go to ownyourbits.com/downloads/testing so the testers could have a quick look at it. This is once a month / once every two months more or less, shouldn't be too much if there's four or five people available.

If you are interested in being more involved you are welcome to join the telegram group at https://t.me/NextCloudPi ... in that chat we talk a bit about everything, and of course we only help when we can and we have time.

Otherwise, let's just test this image, that's also great =)

Building the image now, after fixing some issues

nachoparker, 2018-06-20:
Ok guys, the image is at 

https://ownyourbits.com/downloads/testing/

everything NCP related seems to work, including `letsencrypt`, but there's two issues

- the `armbian-config` tool wasn't included
- there's only 50MiB of space left. Need to expand the root filesystem manually or there will be trouble, it doesn't expand it automatically

It seems that those two issues are an Armbian problem. They must be changing/breaking the builds lately :| 

I will try to create a new image from zero to see if those two go away, otherwise the image seems like a good candidate

cheers

butterbrot21, 2018-06-20:
Nice, that was fast! I am happy if I can help.
I can  imagine that this alone is almost impossible.

I think it's great that you are also talking with dietpi. ncp and dietpi would be a super horny combo.

I am now looking forward to test the image^^

would the commands form Armbian work for expand the root filesystem ? :

_/etc/init.d/resize2fs start_

`update-rc.d firstrun defaults`
and reboot

What programming languages ​​are helpful/are you looking for help?

nachoparker, 2018-06-20:
hi,

update: created a new image and now it correctly expands the filesystems and all the stuff. This could be the next release, with NC13.0.4. Uploaded to the testing folder.

---

thanks for your help!! 

> I think it's great that you are also talking with dietpi. ncp and dietpi would be a super horny combo.

Let's see if we can make it happen, we have some code to adapt but then we could help each other with maintenance. Also, now NCP comes in the Armbian software menu! ([details](https://ownyourbits.com/2018/05/29/nextcloudpi-brings-nc13-0-2-automatic-nc-upgrades-rock64-and-banana-pi-support-armbian-integration-chinese-language-and-more/))

### what we need 
In the developer side of things: bash, in the webdev side of things, PHP + whatever JS, in the testing side of things: people with time to test + optionally python/bash testing.

We are most active in Telegram (everybody is invited), this is a copy/paste of what I wrote there after detecting this Odroid issue

---
Hello everyone
this is a reminder... we need help!  (in order of importance)

 - testers
 - developers
 - webdevs
 - documentation
 - translations

the release schedule is around once a month, or every other month... it will probably slow down a bit in the future
I would need one or two people per board to do a quick test before I announce the release
I don't have all the boards, nor do I have time to test them all one by one like I have been doing... in the end the last Odroid image had some internet connectivity issues that I did not detect
this should be about one hour a month or less of work to help, and we could then try to support more boards
same goes with docker containers
if I were a bit more free, I would like to go back to fixing/improving the automated testing (in the 'test' folder) and that would make testing even easier for everyone... just see that it boots and run the automated tests
ofc help creating these tests would be great but I would be so happy with just some basic testing once every one or two months
if anyone is interested/available, just tell me and let's see if we can create a procedure for each release, like I create the images, then they go to https://ownyourbits.com/downloads/testing/, then they are tested, then moved to the main downloads folder
ofc, we help when we can and have time, so if there's a couple people per board that would make it even less demanding for everyone

butterbrot21, 2018-06-20:
that's awesome!!

then i try to get fit in bash and Python.

until then I can at least help in testing new images,
and help with translation into German if necessary :)

next month i will buy an odroid and maybe an other cheap board for testing.

i hate to say this because you already have enough work, but i think there are problems with your webhost at the moment. I have not come to ownyourbits.com and nextcloudpi.com since yesterday, every now and then.  Right now, I can't connect to ownyourbits.com
-->"This website is not reachable"

looking forward to download and test the new image^^

nachoparker, 2018-06-20:
that's great, thank you very much!

We have partial translations to german, but only like 10% of it. If you use telegram we can talk there, otherwise we can use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) to  communicate.

I plan to announce the new images for testing before each release, probably on Telegram or in the forums maybe.

cheers,

butterbrot21, 2018-06-20:
Of course I'll help! I'll check out the telegram and the formum tomorrow. 

I have now loaded the image and the wizard has started automatically.
An error occurred in the usb format step:

[ nc-automount ]
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount.service → /usr/lib/systemd/system/nc-automount.service.
Created symlink /etc/systemd/system/multi-user.target.wants/nc-automount-links.service → /usr/lib/systemd/system/nc-automount-links.service.
automount enabled

[ nc-format-USB ]
unexpected error

now i try to connect via putty to perform a `sudo ncp-report`
-first i go to ncp wep-gui and chnage the dafault pw 1234 to something different and activate it
-> looking good:
[ SSH ]
Enter new UNIX password: Retype new UNIX password: passwd: password updated successfully
Synchronizing state of ssh.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable ssh
SSH enabled

-i start putty and connect as root
-.i put my new pw in (changed in web-gui with user root)
->welcome message:

NextCloudPi v0.57.8 is up to date
Last login: Wed Jun 20 19:08:54 2018 from 192.168.178.57
Changing password for root.
(current) UNIX password:

->and again .i put my new pw in
--->Putty closes.

my plan was to format the harddrive manually?
it is a wdred 2tb with ext4 at the moment.

if i can connect via ssh i could give more info

nachoparker, 2018-06-20:
Great feedback!

> [ nc-format-USB ]
unexpected error

What is the output of 

```
lsblk -l -n
```
Wouldn't be surprised if this needs adjustment, it's taylored to the pi.

> ->and again .i put my new pw in (1234567)
--->Putty closes.

In order to use SSH, you 
- go to ncp-web
- set a **one time password** ( `1234567` )
- putty in, it will ask for that password
- then, it will ask for another password which will be the permanent one. Don't use `1234567`, use another one this time.

This is for security reasons.

---
I am already sharing this image instead of the other one, as it had a serious flaw (no internet connectivity)

butterbrot21, 2018-06-20:
I´m back! 
thank you for the steps. now it has worked with ssh :)

in the meantime I formatted the drive again with the help of omv.

After reinstall and first setup wizard,it again stops on the format-usb step.

```
 [ nc-format-USB ]
unexpected error
```

Then i tried 
`lsblk -l -n`

-->after that i  start the setut-wizard again but same issue

```
 [ nc-format-USB ]
unexpected error
```

More useful informtion:

```
root@nextcloudpi:~# lsblk
NAME        MAJ:MIN RM   SIZE RO TYPE MOUNTPOINT
sda           8:0    0   1.8T  0 disk
`-sda1        8:1    0   1.8T  0 part
mmcblk1     179:0    0  14.9G  0 disk
`-mmcblk1p1 179:1    0  14.7G  0 part /
zram0       253:0    0    50M  0 disk /var/log
zram1       253:1    0 249.3M  0 disk [SWAP]
zram2       253:2    0 249.3M  0 disk [SWAP]
zram3       253:3    0 249.3M  0 disk [SWAP]
zram4       253:4    0 249.3M  0 disk [SWAP]
root@nextcloudpi:~#   0 249.3M  0 disk [SWAP]
```

```
root@nextcloudpi:~# sudo ncp-report
<details>
<summary>NextCloudPi diagnostics</summary>

NextCloudPi version  v0.57.8
NextCloudPi image    NextCloudPi_06-20-18
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      3.8G/15G
rootfs usage         3.8G/15G
swapfile             /var/swap
Nextcloud check      ok
Nextcloud version    13.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        closed
port check 443       closed
IP                   192.168.178.42
gateway              192.168.178.1
interface            enx001e06328b56
certificates         none
NAT loopback         no
[... 88 lines cut ...]
2018-06-20 21:13:37 3064115200 [Note] InnoDB: 128 rollback segment(s) are active.
2018-06-20 21:13:37 3064115200 [Note] InnoDB: Waiting for purge to start
2018-06-20 21:13:37 3064115200 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.36-82.1 started; log sequence number 3318864
2018-06-20 21:13:37 2483024704 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-06-20 21:13:37 3064115200 [Note] Plugin 'FEEDBACK' is disabled.
2018-06-20 21:13:37 3064115200 [Note] Server socket created on IP: '127.0.0.1'.
2018-06-20 21:13:37 3064115200 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.26-MariaDB-0+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Debian 9.1

</details>

<details>
<summary>Nextcloud logs</summary>

{"reqId":"NLG33NfvLSOSNU0PIS7x","level":3,"time":"2018-06-20T21:13:38+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"Redis::connect(): connect() failed: No such file or directory at \/var\/www\/nextcloud\/lib\/private\/RedisFactory.php#84","userAgent":"--","version":"13.0.4.0"}

</details>

root@nextcloudpi:~#
```
The hard disk (WD-RED 2TB) is 1 month old and could be used in omv.

I hope that all is not a big issue.

See you tomorrow

nachoparker, 2018-06-21:
Great, I found the problem I think. You activated ZRAM first and that is confusing `nc-format-USB`.

Putty in, then type `sudo ncp-update devel` and try again. Be aware that all contents of the USB drive will be lost.

Thanks

butterbrot21, 2018-06-21:
Nice nice nice,  You did it! :)

After `sudo ncp-update devel` and setup-wizard
its all working know, DNS, Certrbot, inet.. all up and runing smoothly :)

You did a great job again!

the perfect start for the day 👍

nachoparker, 2018-06-21:
Great! That's good news.

@Kunibert-007 yes, just backup and restore

[wiki tutorial](https://github.com/nextcloud/nextcloudpi/wiki/How-to-backup-and-restore-a-NextCloudPi-instance-using-ncp-config)

nachoparker, 2018-06-23:
mmm that's weird. Let's see, what is the output of

```
sudo cat /root/.my.cnf
```

(change the password when you paste it here)

Also, let's try

- reboot
- output of

```
sudo ncp-report
```

Kunibert-007, 2018-06-23:
After `sudo cat /root/.my.cnf` get this output:

```
root@nextcloudpi:~# sudo cat /root/.my.cnf
OPCACHEfe8c305d6d50428e53root@nextcloudpi:~#
```

Reboot doesn't help. Here the ncp-report:
`
root@nextcloudpi:~# sudo ncp-report
<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.57.12
NextCloudPi image    NextCloudPi_06-20-18
distribution         Debian GNU/Linux 9 \n \l
automount            yes
USB devices          sda
datadir              /media/USBdrive/ncdata
data in SD           no
data filesystem      btrfs
data disk usage      127G/932G
rootfs usage         3.8G/29G
swapfile             /var/swap
Nextcloud check      error
HTTPD service        up
PHP service          down
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.0.4
gateway              192.168.0.1
interface            enx001e06329e4b
certificates         none
NAT loopback         no
uptime               3min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

An unhandled exception has been thrown:
Doctrine\DBAL\DBALException: Failed to connect to the database: An exception occured in driver: SQLSTATE[HY000] [1045] Access denied for user 'ncadmin'@'localhost' (using password: NO) in /var/www/nextcloud/lib/private/DB/Connection.php:64
Stack trace:
#0 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(429): OC\DB\Connection->connect()
#1 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(389): Doctrine\DBAL\Connection->getDatabasePlatformVersion()
#2 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(328): Doctrine\DBAL\Connection->detectDatabasePlatform()
#3 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/Connection.php(623): Doctrine\DBAL\Connection->getDatabasePlatform()
#4 /var/www/nextcloud/lib/private/DB/Connection.php(151): Doctrine\DBAL\Connection->setTransactionIsolation(2)
#5 /var/www/nextcloud/3rdparty/doctrine/dbal/lib/Doctrine/DBAL/DriverManager.php(172): OC\DB\Connection->__construct(Array, Object(Doctrine\DBAL\Driver\PDOMySql\Driver), Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#6 /var/www/nextcloud/lib/private/DB/ConnectionFactory.php(152): Doctrine\DBAL\DriverManager::getConnection(Array, Object(Doctrine\DBAL\Configuration), Object(Doctrine\Common\EventManager))
#7 /var/www/nextcloud/lib/private/Server.php(628): OC\DB\ConnectionFactory->getConnection('mysql', Array)
#8 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(113): OC\Server->OC\{closure}(Object(OC\Server))
#9 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('OCP\\IDBConnecti...')
#10 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('OCP\\IDBConnecti...')
#11 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(164): OC\ServerContainer->query('OCP\\IDBConnecti...')
#12 /var/www/nextcloud/3rdparty/pimple/pimple/src/Pimple/Container.php(109): OC\AppFramework\Utility\SimpleContainer->OC\AppFramework\Utility\{closure}(Object(OC\Server))
#13 /var/www/nextcloud/lib/private/AppFramework/Utility/SimpleContainer.php(116): Pimple\Container->offsetGet('DatabaseConnect...')
#14 /var/www/nextcloud/lib/private/ServerContainer.php(132): OC\AppFramework\Utility\SimpleContainer->query('DatabaseConnect...')
[... 122 lines cut ...]
{"reqId":"WyukbH8AAQEAAAOUck8AAABQ","level":2,"time":"2018-06-21T13:13:16+00:00","remoteAddr":"184.105.139.67","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"184.105.139.67\" tried to access using \"79.230.113.181\" as host.","userAgent":"--","version":"13.0.4.0"}
{"reqId":"WyxZjH8AAQEAAAOUiukAAABO","level":2,"time":"2018-06-22T02:06:04+00:00","remoteAddr":"178.201.149.206","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"178.201.149.206\" tried to access using \"79.230.113.181\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 5.1; rv:9.0.1) Gecko\/20100101 Firefox\/9.0.1","version":"13.0.4.0"}
{"reqId":"WyxZjH8AAQEAAAOUiuoAAABW","level":2,"time":"2018-06-22T02:06:04+00:00","remoteAddr":"178.201.149.206","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"178.201.149.206\" tried to access using \"79.230.113.181\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 5.1; rv:9.0.1) Gecko\/20100101 Firefox\/9.0.1","version":"13.0.4.0"}
{"reqId":"WyxZjX8AAQEAAAOUiusAAABX","level":2,"time":"2018-06-22T02:06:05+00:00","remoteAddr":"178.201.149.206","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"178.201.149.206\" tried to access using \"79.230.113.181\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 5.1; rv:9.0.1) Gecko\/20100101 Firefox\/9.0.1","version":"13.0.4.0"}
{"reqId":"WyxZjX8AAQEAAAOUiuwAAABS","level":2,"time":"2018-06-22T02:06:05+00:00","remoteAddr":"178.201.149.206","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"178.201.149.206\" tried to access using \"79.230.113.181\" as host.","userAgent":"Mozilla\/5.0 (Windows NT 5.1; rv:9.0.1) Gecko\/20100101 Firefox\/9.0.1","version":"13.0.4.0"}
[... cut ...] ke Gecko) Chrome\/45.0.2454.101 Safari\/537.36","version":"13.0.4.0"}
{"reqId":"Wy0LnX8AAQEAAGVcwNwAAAAT","level":2,"time":"2018-06-22T14:45:50+00:00","remoteAddr":"74.82.47.3","user":"--","app":"core","method":"GET","url":"\/","message":"Trusted domain error. \"74.82.47.3\" tried to access using \"79.230.113.181\" as host.","userAgent":"--","version":"13.0.4.0"}

```
</details>

root@nextcloudpi:~#
`

I just took the latest image and put it on the m2-card. But Internet (DNS-Entry) works now :-)

nachoparker, 2018-06-23:
We have two issues, 

- your `/root/.my.cnf` is corrupt. No idea how it got that way.
- your PHP is down, let me see the output of 

```
sudo systemctl status php7.0-fpm
```
Let's go through the steps one by one

- flash the image, go through activation and wizard. Does everything work? Does NC work?
- if it works, then restore it and look at `/.my.cnf` and PHP output (command above)

It should look like

```
$ sudo cat /root/.my.cnf
[client]
password=XXXXXXXX
```

Kunibert-007, 2018-06-23:
Ok, did the following:
Flashed the card again with the image from 20th. june.  After booting I can't reach the ncp-website. SSH is disabled. In the image before SSH was enabled. NCP in Version 0.57.8 is installed.
Did the update to version 0.57.12 and actived SSH with ncp-config.
But the problem is the same as before, no access to the ncp- or the nextcloud-website.

`sudo systemctl status php7.0-fpm` brings the following result:
```
`root@nextcloudpi:~# sudo systemctl status php7.0-fpm
* php7.0-fpm.service - The PHP 7.0 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.0-fpm.service; enabled; vendor prese
   Active: active (running) since Wed 2018-06-20 14:56:08 UTC; 2 days ago
     Docs: man:php-fpm7.0(8)
 Main PID: 1137 (php-fpm7.0)
   Status: "Processes active: 0, idle: 2, Requests: 0, slow: 0, Traffic: 0req/se
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/php7.0-fpm.service
           |-1137 php-fpm: master process (/etc/php/7.0/fpm/php-fpm.conf)
           |-1231 php-fpm: pool www
           `-1232 php-fpm: pool www

Jun 20 14:56:08 nextcloudpi systemd[1]: Starting The PHP 7.0 FastCGI Process Man
Jun 20 14:56:08 nextcloudpi systemd[1]: Started The PHP 7.0 FastCGI Process Mana
lines 1-14/14 (END)...skipping...
* php7.0-fpm.service - The PHP 7.0 FastCGI Process Manager
   Loaded: loaded (/lib/systemd/system/php7.0-fpm.service; enabled; vendor preset: enabled)
   Active: active (running) since Wed 2018-06-20 14:56:08 UTC; 2 days ago
     Docs: man:php-fpm7.0(8)
 Main PID: 1137 (php-fpm7.0)
   Status: "Processes active: 0, idle: 2, Requests: 0, slow: 0, Traffic: 0req/sec"
    Tasks: 3 (limit: 4915)
   CGroup: /system.slice/php7.0-fpm.service
           |-1137 php-fpm: master process (/etc/php/7.0/fpm/php-fpm.conf)
           |-1231 php-fpm: pool www
           `-1232 php-fpm: pool www

Jun 20 14:56:08 nextcloudpi systemd[1]: Starting The PHP 7.0 FastCGI Process Manager...
Jun 20 14:56:08 nextcloudpi systemd[1]: Started The PHP 7.0 FastCGI Process Manager.
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
~
lines 1-14/14 (END)
```

And I have this:
`
root@nextcloudpi:~# $ sudo cat /root/.my.cnf
Could not find the database of available applications, run update-command-not-found as root to fix this
$: command not found
`

nachoparker, 2018-06-23:
> In the image before SSH was enabled

Yes, this has changed for security reasons. It can still be activated from ncp-web

> After booting I can't reach the ncp-website.

Do you mean in port 4443? during the first boot it is not available. You just have to go to you IP address (or nextcloudpi.local if you are on Linux) and you will be presented with the activation screen. After activating, then ncp-web will be available on port 4443.

Did you activate? I assume so... or how did you update if you say you have no ncp-web and no SSH?

Please, I need more clear details, it's hard to understand.

Try without sudo
```
cat /root/.my.cnf
```
and always include output of

```
sudo ncp-report
```

This time your PHP seems to be up.
thanks!

Kunibert-007, 2018-06-23:
I'm so sorry, shame on me. I tried reaching the ncp-website with port 4443. My fault, sorry!
With port 443 I get the activation-page.
Could active now and actually I restore the backup. Scanning files is still ongoing. I think it works perfectly. Test all this this evening and give you a feedback. Thanks for your patience.

nachoparker, 2018-06-23:
good news!! :D

Kunibert-007, 2018-06-23:
@nachoparker , everything (including DNS-entry, certificate, ncp-restore) works in the image. Many thanks to you and your work!

nachoparker, 2018-06-26:
Hey guys, we are putting together our testing procedures because this is getting big and we need a bit of help and organization!

If you were interested to help, look at this [wiki article](https://github.com/nextcloud/nextcloudpi/wiki/Testing-NextCloudPi) with details. It should really be only 20 minutes a month of volunteer work for the community.

You can also reach us at [Telegram](https://t.me/NextCloudPi) to discuss anything 

Thanks!

nachoparker, 2018-06-27:
great! so I need you to

- create a user in the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm)
- edit the [wiki](https://github.com/nextcloud/nextcloudpi/wiki/Testing-NextCloudPi) article, and add your forum username.

I will mention you when there is a pre-release, and you will receive an email notification.

thanks!

Kunibert-007, 2018-06-27:
No problem, thanks for your time. Will link my forum-name in the article.