nextcloud/nextcloudpi #554: NCP - Docker error (nextcloud access via :4443 only, Redis & postfix down)

Linecutterx, 2018-06-24:
Paste the output of `sudo ncp-report`

<--! Paste this in GitHub report -->

<details>
<summary>NextCloudPi diagnostics</summary>

```

PHP Warning:  Redis::connect(): connect() failed: Connection refused in /var/www/nextcloud/lib/private/RedisFactory.php on line 84
NextCloudPi version  v0.57.13
NextCloudPi image    NextCloudPi_05-30-18
distribution         Debian GNU/Linux 9 \n \l
USB devices          sda
datadir              /data/app/data
data in SD           no
data filesystem      ext2/ext3
data disk usage      15G/58G
rootfs usage         2.2G/7.0G
swapfile             none
Nextcloud check      error
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        down
Postfix service      down
internet check       ok
port check 80        closed
port check 443       closed
IP                   172.17.0.3
gateway              172.17.0.1
interface            eth0
certificates         linecutter.ddns.net
NAT loopback         no
uptime               52min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

PHP Warning:  Redis::connect(): connect() failed: Connection refused in /var/www/nextcloud/lib/private/RedisFactory.php on line 84
An unhandled exception has been thrown:
RedisException: Redis server went away in /var/www/nextcloud/lib/private/Memcache/Redis.php:54
Stack trace:
#0 /var/www/nextcloud/lib/private/Memcache/Redis.php(54): Redis->get('6e805ea6f8dd581...')
#1 /var/www/nextcloud/lib/private/App/InfoParser.php(57): OC\Memcache\Redis->get('/var/www/nextcl...')
#2 /var/www/nextcloud/lib/private/legacy/app.php(621): OC\App\InfoParser->parse('/var/www/nextcl...')
#3 /var/www/nextcloud/lib/private/AppFramework/App.php(62): OC_App::getAppInfo('files')
#4 /var/www/nextcloud/lib/private/legacy/app.php(220): OC\AppFramework\App::buildAppNamespace('files')
#5 /var/www/nextcloud/lib/private/legacy/app.php(119): OC_App::registerAutoloading('files', '/var/www/nextcl...')
#6 /var/www/nextcloud/lib/base.php(662): OC_App::loadApps(Array)
#7 /var/www/nextcloud/lib/base.php(1081): OC::init()
#8 /var/www/nextcloud/console.php(46): require_once('/var/www/nextcl...')
#9 /var/www/nextcloud/occ(11): require_once('/var/www/nextcl...')
#10 {main}
```
</details>

<details>
<summary>HTTPd logs</summary>

```

[Sun Jun 24 20:10:57.640415 2018] [ssl:warn] [pid 48:tid 1995554816] AH01909: localhost:4443:0 server certificate does NOT include an ID which matches the server name
[Sun Jun 24 20:10:58.002245 2018] [core:warn] [pid 48:tid 1995554816] AH00098: pid file /var/run/apache2/apache2.pid overwritten -- Unclean shutdown of previous Apache run?
[Sun Jun 24 20:10:58.014621 2018] [mpm_event:notice] [pid 48:tid 1995554816] AH00489: Apache/2.4.25 (Debian) OpenSSL/1.0.2l   configured -- resuming normal operations
[... 53 lines cut ...]
```

{"reqId":"WzABqawRAAMAAAA59cgAAABH","level":3,"time":"2018-06-24T20:40:10+00:00","remoteAddr":"51.7.194.161","user":"--","app":"core","method":"GET","url":"\/index.php\/204","message":"Exception: {\"Exception\":\"RedisException\",\"Message\":\"Redis server went away\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Memcache\\\/Redis.php(54): Redis->get('6e805ea6f8dd581...')\\n#1 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/InfoParser.php(57): OC\\\\Memcache\\\\Redis->get('\\\/var\\\/www\\\/nextcl...')\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(621): OC\\\\App\\\\InfoParser->parse('\\\/var\\\/www\\\/nextcl...')\\n#3 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/App.php(62): OC_App::getAppInfo('activity')\\n#4 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(220): OC\\\\AppFramework\\\\App::buildAppNamespace('activity')\\n#5 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(119): OC_App::registerAutoloading('activity', '\\\/var\\\/www\\\/nextcl...')\\n#6 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/util.php(193): OC_App::loadApps(Array)\\n#7 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Filesystem.php(264): OC_Util::setupFS('')\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Server.php(261): OC\\\\Files\\\\Filesystem::getMountManager(NULL)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/pimple\\\/pimple\\\/src\\\/Pimple\\\/Container.php(113): OC\\\\Server->OC\\\\{closure}(*** sensitive parameters replaced ***)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/Utility\\\/SimpleContainer.php(116): Pimple\\\\Container->offsetGet('RootFolder')\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/ServerContainer.php(132): OC\\\\AppFramework\\\\Utility\\\\SimpleContainer->query('RootFolder')\\n#12 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Server.php(283): OC\\\\ServerContainer->query('RootFolder')\\n#13 [internal function]: OC\\\\Server->OC\\\\{closure}(*** sensitive parameters replaced ***)\\n#14 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/LazyRoot.php(61): call_user_func(Object(Closure))\\n#15 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/LazyRoot.php(141): OC\\\\Files\\\\Node\\\\LazyRoot->__call('get', Array)\\n#16 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/AppData\\\/AppData.php(79): OC\\\\Files\\\\Node\\\\LazyRoot->get('appdata_ocfkf8b...')\\n#17 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/AppData\\\/AppData.php(105): OC\\\\Files\\\\AppData\\\\AppData->getAppDataFolder()\\n#18 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/JSCombiner.php(94): OC\\\\Files\\\\AppData\\\\AppData->getFolder('core')\\n#19 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/JSResourceLocator.php(113): OC\\\\Template\\\\JSCombiner->process('\\\/var\\\/www\\\/nextcl...', 'core\\\/js\\\/merged-...', 'core')\\n#20 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/JSResourceLocator.php(69): OC\\\\Template\\\\JSResourceLocator->cacheAndAppendCombineJsonIfExist('\\\/var\\\/www\\\/nextcl...', 'core\\\/js\\\/merged-...')\\n#21 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/ResourceLocator.php(78): OC\\\\Template\\\\JSResourceLocator->doFind('js\\\/merged-templ...')\\n#22 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/TemplateLayout.php(306): OC\\\\Template\\\\ResourceLocator->find(Array)\\n#23 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/TemplateLayout.php(146): OC\\\\TemplateLayout::findJavascriptFiles(Array)\\n#24 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/template.php(207): OC\\\\TemplateLayout->__construct('error', '')\\n#25 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/Base.php(132): OC_Template->fetchPage()\\n#26 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/template.php(351): OC\\\\Template\\\\Base->printPage()\\n#27 \\\/var\\\/www\\\/nextcloud\\\/index.php(70): OC_Template::printExceptionErrorPage(Object(RedisException))\\n#28 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Memcache\\\/Redis.php\",\"Line\":54}","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.2.0","version":"13.0.4.0"}
{"reqId":"WzABqawRAAMAAAA59cgAAABH","level":3,"time":"2018-06-24T20:40:10+00:00","remoteAddr":"51.7.194.161","user":"--","app":"PHP","method":"GET","url":"\/index.php\/204","message":"Redis::connect(): connect() failed: Connection refused at \/var\/www\/nextcloud\/lib\/private\/RedisFactory.php#84","userAgent":"Mozilla\/5.0 (Android) ownCloud-android\/3.2.0","version":"13.0.4.0"}
{"reqId":"WzAFg6wRAAMAAAA59ckAAEkD","level":3,"time":"2018-06-24T20:56:35+00:00","remoteAddr":"192.168.1.71","user":"--","app":"index","method":"GET","url":"\/","message":"Exception: {\"Exception\":\"RedisException\",\"Message\":\"Redis server went away\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Memcache\\\/Redis.php(54): Redis->get('6e805ea6f8dd581...')\\n#1 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/App\\\/InfoParser.php(57): OC\\\\Memcache\\\\Redis->get('\\\/var\\\/www\\\/nextcl...')\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(621): OC\\\\App\\\\InfoParser->parse('\\\/var\\\/www\\\/nextcl...')\\n#3 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/AppFramework\\\/App.php(62): OC_App::getAppInfo('files')\\n#4 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app.php(220): OC\\\\AppFramework\\\\App::buildAppNamespace('files')\\n#5 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/app. [... cut ...]
{"reqId":"2ZuIkDbRpqW4pvFZBQn4","level":3,"time":"2018-06-24T21:02:37+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"Redis::connect(): connect() failed: Connection refused at \/var\/www\/nextcloud\/lib\/private\/RedisFactory.php#84","userAgent":"--","version":"13.0.4.0"}
{"reqId":"GMnCL7m5yR4fkccM1ROm","level":3,"time":"2018-06-24T21:02:44+00:00","remoteAddr":"","user":"--","app":"PHP","method":"--","url":"--","message":"Redis::connect(): connect() failed: Connection refused at \/var\/www\/nextcloud\/lib\/private\/RedisFactory.php#84","userAgent":"--","version":"13.0.4.0"}

```
</details>

Linecutterx, 2018-06-24:
[redis-server.log](https://github.com/nextcloud/nextcloudpi/files/2131234/redis-server.log)

nachoparker, 2018-06-26:
what happens if you try to start redis manually inside the container?

```
sudo -u redis redis-server /etc/redis/redis.conf
```

Linecutterx, 2018-06-26:
If I run that command then I can get back in an a normal user :)

but...

Restarting the container  results in redis not starting until it's run manually.

Restarting (at 19:41) & then running tail -50 on the log gives me
                _._
           _.-``__ ''-._
      _.-``    `.  `_.  ''-._           Redis 3.2.6 (00000000/0) 32 bit
  .-`` .-```.  ```\/    _.,_ ''-._
 (    '      ,       .-`  | `,    )     Running in standalone mode
 |`-._`-...-` __...-.``-._|'` _.-'|     Port: 0
 |    `-._   `._    /     _.-'    |     PID: 346
  `-._    `-._  `-./  _.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |           http://redis.io
  `-._    `-._`-.__.-'_.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |
  `-._    `-._`-.__.-'_.-'    _.-'
      `-._    `-.__.-'    _.-'
          `-._        _.-'
              `-.__.-'

346:M 26 Jun 19:37:57.238 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
346:M 26 Jun 19:37:57.239 # Server started, Redis version 3.2.6
346:M 26 Jun 19:37:57.239 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
346:M 26 Jun 19:37:57.272 * DB loaded from disk: 0.032 seconds
346:M 26 Jun 19:37:57.272 * The server is now ready to accept connections at /var/run/redis/redis.sock
<EOF>

Running redis & then checking tail suggests that it's not logging anything as an error

346:M 26 Jun 19:37:57.272 * The server is now ready to accept connections at /var/run/redis/redis.sock
                _._
           _.-``__ ''-._
      _.-``    `.  `_.  ''-._           Redis 3.2.6 (00000000/0) 32 bit
  .-`` .-```.  ```\/    _.,_ ''-._
 (    '      ,       .-`  | `,    )     Running in standalone mode
 |`-._`-...-` __...-.``-._|'` _.-'|     Port: 0
 |    `-._   `._    /     _.-'    |     PID: 341
  `-._    `-._  `-./  _.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |           http://redis.io
  `-._    `-._`-.__.-'_.-'    _.-'
 |`-._`-._    `-.__.-'    _.-'_.-'|
 |    `-._`-._        _.-'_.-'    |
  `-._    `-._`-.__.-'_.-'    _.-'
      `-._    `-.__.-'    _.-'
          `-._        _.-'
              `-.__.-'

341:M 26 Jun 19:43:18.746 # WARNING: The TCP backlog setting of 511 cannot be enforced because /proc/sys/net/core/somaxconn is set to the lower value of 128.
341:M 26 Jun 19:43:18.746 # Server started, Redis version 3.2.6
341:M 26 Jun 19:43:18.747 # WARNING overcommit_memory is set to 0! Background save may fail under low memory condition. To fix this issue add 'vm.overcommit_memory = 1' to /etc/sysctl.conf and then reboot or run the command 'sysctl vm.overcommit_memory=1' for this to take effect.
341:M 26 Jun 19:43:18.775 * DB loaded from disk: 0.028 seconds
341:M 26 Jun 19:43:18.776 * The server is now ready to accept connections at /var/run/redis/redis.sock

nachoparker, 2018-06-27:
what's the output of 

```
grep "^maxmemory" /etc/redis/redis.conf
```

Linecutterx, 2018-06-28:
$ grep "^maxmemory" /etc/redis/redis.conf
maxmemory 3gb

nachoparker, 2018-07-13:
what are the docker logs for the container? maybe we will get a clue there

closed 2018-07-17 by nachoparker

phryciuk, 2018-07-25:
@nachoparker I have the same problem. Today I upgraded my docker image. And it seems that after doing that, after each restart of docker image redis and postfix are down. When manually launching redis, I can access nextcloud login page. It would be better if redis started automatically - with mysql, apache and other services.