nextcloud/nextcloudpi #555: Rock64 5-29-2018 image not booting to login

xelcho, 2018-06-27:
used the following to write image to sd card:
tar -xvf NextCloudPi-lite_02-12-17.tar.bz2
sudo dd bs=4M if=NextCloudPi-lite_02-12-17.img of=/dev/sdx

attempted twice with two different sd cards

I then used the following to try to build it:
git clone https://github.com/nextcloud/nextcloudpi.git
cd nextcloudpi
./build-SD-armbian.sh rock64

The build failed looking for docker tools.

Paste the output of `sudo ncp-report`

nachoparker, 2018-06-28:
what board do you have?

xelcho, 2018-06-28:
4gb version. I also wrote a different distro to sd and ran it fine, so I am confident that the board is pseudo defect free.

nachoparker, 2018-06-28:
Yes, it seems that that build [does not work](https://help.nextcloud.com/t/nextcloudpi-on-a-rock64/29619/51?u=nachoparker) on the 4GB version.

[It's taking some time](https://forum.armbian.com/topic/5771-rock64-nightly-image/) for the Armbian people to come up with good stable images that can be reproduced... I am having issues myself trying to create a new image :S

I suggest you install Armbian's stable Stretch and use NCP Docker version, or the 'curl installer'

```
# curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | bash
```

nachoparker, 2018-06-29:
ok, [let's give it another try](https://github.com/armbian/build/pull/1026#event-1707701965)

xelcho, 2018-06-29:
Too cool! I will certainly give this a try when I get back from my holiday. I will be back the second week of July.

In response to your first response: is there a performance hit involved in running the docker image vs using the curl installation method?

Oh, and thanks for your response! :) I was convinced that I was missing something.

nachoparker, 2018-06-29:
No performance hit, but it is a bit less featureful and more advanced to manage.

We need help testing in order to support more boards, look at this [wiki article](https://github.com/nextcloud/nextcloudpi/wiki/Testing-NextCloudPi) with details if interested. It should really be only 20 minutes a month of volunteer work for the community.

You can also reach us at [Telegram](https://t.me/NextCloudPi) to discuss anything 

Thanks!

xelcho, 2018-07-14:
I'm back and curious if you have had any luck with the new armbian build. How goes it?

nachoparker, 2018-07-14:
hi, unfortunatedly, I had no luck last time I checked. Definitely the armbian guys are working on it, but they have not come up with anything stable yet.

xelcho, 2018-07-15:
Waaaah! I will hold tight with my raspberry b+ for now. :(

Thanks for the help!

xelcho, 2018-08-07:
@nachoparker Have you had any luck with getting the 4 gb board image working?

nachoparker, 2018-08-08:
Not yet. I have been busy with life but I will try again in a couple weeks.

xelcho, 2018-08-31:
@nachoparker It's not cool to bug you about trying to do this, so... Can I ask what steps I need to do to test the build/image so I can do it independently?

nachoparker, 2018-09-01:
I still don't have a computer, but soon will.

The instructions are on the main github page.

```
./build-SD-armbian.sh rock64
```

xelcho, 2018-09-10:
FYI - I am working with armbian folks to find a stable build. :)
https://forum.armbian.com/topic/8145-rock64-not-booting/

If anyone else has a 4gb rock64 board and wants to assist in getting a stable build, please join us at the url above.

nachoparker, 2018-09-11:
great, that really helps! thank you. Most of my computer is here. Only waiting for the RAM modules

nachoparker, 2018-09-24:
Good news, I was able to generate a good image, seems like the Armbian guys fixed their long standing issues.

https://ownyourbits.com/downloads/NextCloudPi_Rock64_09-23-18/

Please, somebody with a 4GiB version verify that this boots for them.

closed 2018-09-24 by nachoparker

stratemeyerjw, 2018-09-24:
> Good news, I was able to generate a good image, seems like the Armbian guys fixed their long standing issues.
> 
> https://ownyourbits.com/downloads/NextCloudPi_Rock64_09-23-18/
> 
> Please, somebody with a 4GiB version verify that this boots for them.

I am downloading it now, will confirm shortly.

xelcho, 2018-10-02:
I can confirm that it is working swimmingly! Woohoo!!

Thanks!
Not-your-cheese! :)
It is very tasty to see 3.x GB used by php... :)

stratemeyerjw, 2018-10-02:
> > Good news, I was able to generate a good image, seems like the Armbian guys fixed their long standing issues.
> > https://ownyourbits.com/downloads/NextCloudPi_Rock64_09-23-18/
> > Please, somebody with a 4GiB version verify that this boots for them.
> 
> I am downloading it now, will confirm shortly.

sorry i didnt respond. I couldnt figure out how to get it onto my mmc without an mmc to usb adapter, i have one on order.

xelcho, 2018-10-02:
@stratemeyerjw dejavu, that was me last month. The usb adapter is cool and transfer rates are teens of MB/Sec. Enjoy!