nextcloud/nextcloudpi #557: No access to nexcloudpi

Phil1988, 2018-07-03:
I just installed NextcloudPi (NextCloudPi_RPi_06-19-18.img) the first time on my Pi 3B+.

I enabled ssh and wifi threw the SD card befor powering the Pi up.

After starting and using the WebIF to hit the "Activate" Button it needed a long time after I could see more...
then I was redirected to https://PiAdress:4443 and I was asked for access.

I used 'ncp' and 'ownyourbits' but I had no luck.
Also 'pi' and 'raspberry' (or any combination) is not working.
Also I cant Login to the normal Nextcloud interface usind 'ncp' and the generated passwort that was shown on the first page - so I have NO access to my cloud :D 

I also tried to connect via SSH usind 'pi' and 'raspberry' works here.

any ideas?

ovpc, 2018-07-03:
On activation page, you should have newly generated passwords for user ncp.
One for https:// ipadress:4443 and one for https://ipadress

Phil1988, 2018-07-03:
I had 2 generated passwords... 
none of them worked for 4443.
also not 'pi'/'raspberry' or ncp/ownyourbits.

if it was that simple I wouldnt open an issue here