nextcloud/nextcloudpi #559: [question] Self Signed Certificates

lil5, 2018-07-09:
### QUESTIONS

I'm not sure which file I should send to my users or where that file might be located.

I am using the docker image looked threw the `/data` and inside the image but all I can find is the `ca-certificates` dir which has 100s of external CA's

nachoparker, 2018-07-13:
Hi,

Certs are in `/etc/letsencrypt/live`. Please, use the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) to ask questions.

closed 2018-07-13 by nachoparker

lil5, 2018-07-16:
@nachoparker 

> Certs are in `/etc/letsencrypt/live`. 

For Self Signed Certificates (not using the NextcloudPi letsencrypt addon) that is not true :-1:

```
$ sudo docker-compose exec -u www-data ncp bash
// I tried root and www-data
www-data@<docker_image>:/etc/letsencrypt/live$ ls -alh
total 0
drwxr-xr-x 1 root root   0 Jun 20 10:49 .
drwxr-xr-x 1 root root 146 Jul  1 07:43 ..
```

Certificates are not an issue of Nextcloud rather that of NextcloudPi so I didn't think this question would fit on their forums.

nachoparker, 2018-07-16:
Hi,

Sorry you are right, the **self-signed** certificates are in 

```
/etc/ssl/certs/ssl-cert-snakeoil.pem
/etc/ssl/private/ssl-cert-snakeoil.key
```
> Certificates are not an issue of Nextcloud rather that of NextcloudPi so I didn't think this question would fit on their forums.

This section of the [forums](https://help.nextcloud.com/c/support/appliances-docker-snappy-vm) is for NCP. It is very clear in the instructions when you open an issue that github is not for questions.

Please, use the forums in the future, unless you find some kind of bug. 

Thanks!

lil5, 2018-07-16:
@nachoparker I'm sorry, will do so from now on,

Thanks for the answer btw

nachoparker, 2018-07-16:
no problem! thanks for understanding... otherwise this becomes madness and there is much more people available to help in the forums