nextcloud/nextcloudpi #560: suggestion: custom DynDns provider

clown522, 2018-07-10:
hi @nachoparker,
i have one suggestion that you are able to use any dyndns provider with following configuration at your config page:

update-url:  http://www.ddnss.de/upd.php?user=<username>&pwd=<pass>&host=<domain>&ip=<ipaddr>&ip6=<ip6addr>
domain name: example.ddnss.de
user: test
pasword: test
update intervall: 10

advantage would be to use nearly any provider without actively supporting each of them.
We just need and Update-URL function.

Thanks again for your great work!

nachoparker, 2018-07-13:
hi!

could you explain this a little bit? everything is in German >_<

thanks for the ideas!

mschilli87, 2018-07-13:
@nachoparker: I don't think the URL is supposed to work, it just show some error. I *guess* the question is how to set up an update URL that encodes all relevant information (user, password, etc.) in the URL as some kind of generalized scheme enabling the use of any dynamic DNS provider.

clown522, 2018-07-13:
Sorry, it was just an example how a configuration would be if the update url function would exist. @mschilli87 is absolutely right.
When a change in IP address is found or a user alters any of their settings, the client should perform an update. All updates are sent using a well-formed HTTP request.

It is described here: 
https://help.dyn.com/remote-access-api/perform-update/

My feature request is to have such an url update function at your config page, so that nearly every dyndns can be used. I hope it is clearer now.