nextcloud/nextcloudpi #563: 503 Service Unavailable after clean installation

Stefomat, 2018-07-15:
Hi @nachoparker,
I have a problem with my nextcloudpi installation.

Originaly i updated from Nextcloud v13.0.2 to v13.0.4 via the Nextcloud updater.
After the update the webinterface wouldn't load an give me a "503 Service unavailable" error.

To find out if a clean and new installation would solve the problem, i took another raspberry an installed the newest version of nextcloudpi using the install.sh script:
`curl -sSL https://raw.githubusercontent.com/nextcloud/nextcloudpi/master/install.sh | sudo bash`

The installation worked fine, i could activate nextcloudpi and access ncp-web.
However, I could not login to nextcloud itself. It loaded for several minutes, then gave me a 503 error.
After some more minutes, i tried again and it worked.
After I rebooted the raspberry, there where no loading times, i got the 503 error straight away.
I investigated a bit and discovered, that "php7.0-fpm.service" and "mysql.service" weren't running.
After manually starting them, everything worked fine.

Thanks for reading!

Here ist the output of ncp-report:

<details>
<summary>NextCloudPi diagnostics</summary>

```

NextCloudPi version  v0.57.18
distribution         Raspbian GNU/Linux 9 \n \l
automount            no
USB devices          none
datadir              /var/www/nextcloud/data
data in SD           yes
data filesystem      ext2/ext3
data disk usage      1.9G/7.1G
rootfs usage         1.9G/7.1G
swapfile             /var/swap
dbdir                /var/lib/mysql
Nextcloud check      ok
Nextcloud version    13.0.4.0
HTTPD service        up
PHP service          up
MariaDB service      up
Redis service        up
Postfix service      up
internet check       ok
port check 80        open
port check 443       open
IP                   192.168.1.46
gateway              192.168.1.1
interface            eth0
certificates         none
NAT loopback         no
uptime               13min

```
</details>

<details>
<summary>Nextcloud configuration</summary>

```

{
    "system": {
        "passwordsalt": "***REMOVED SENSITIVE VALUE***",
        "secret": "***REMOVED SENSITIVE VALUE***",
        "trusted_domains": {
            "0": "localhost",
            "1": "192.168.1.46",
            "5": "nextcloudpi.local"
        },
[... 80 lines cut ...]
2018-07-15 14:19:06 1989189632 [Note] InnoDB:  Percona XtraDB (http://www.percona.com) 5.6.35-80.0 started; log sequence number 4868666
2018-07-15 14:19:07 1447031616 [Note] InnoDB: Dumping buffer pool(s) not yet started
2018-07-15 14:19:07 1989189632 [Note] Plugin 'FEEDBACK' is disabled.
2018-07-15 14:19:08 1989189632 [Note] Server socket created on IP: '127.0.0.1'.
2018-07-15 14:19:08 1989189632 [Note] /usr/sbin/mysqld: ready for connections.
Version: '10.1.23-MariaDB-9+deb9u1'  socket: '/var/run/mysqld/mysqld.sock'  port: 3306  Raspbian 9.0

```
</details>

<details>
<summary>Nextcloud logs</summary>

```

{"reqId":"MP8NQFquCPhvqQj8TAYD","level":4,"time":"2018-07-15T12:45:22+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"CssavZmmpvkfMFP9Qsgp","level":4,"time":"2018-07-15T13:00:07+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":""}
{"reqId":"KTQ340vzD4BLxxNIfeCT","level":4,"time":"2018-07-15T13:15:03+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":"13.0.4.0"}
{"reqId":"t9zF96VvcdyX9ZZkyI0X","level":4,"time":"2018-07-15T13:30:02+00:00","remoteAddr":"","user":"--","app":"cron","method":"--","url":"--","message":"Not installed","userAgent":"--","version":"13.0.4.0"}
{"reqId":"W0tSd38AAQEAACVMY9AAAAQU","level":3,"time":"2018-07-15T13:58:38+00:00","remoteAddr":"192.168.1.20","user":"ncp","app":"index","method":"GET","url":"\/index.php\/apps\/files\/","message":"Exception: {\"Exception\":\"Doctrine\\\\DBAL\\\\Exception\\\\UniqueConstraintViolationException\",\"Message\":\"An exception occurred while executing 'INSERT INTO `oc_filecache` (`mimepart`,`mimetype`,`mtime`,`size`,`etag`,`storage_mtime`,`permissions`,`parent`,`checksum`,`path_hash`,`path`,`name`,`storage`) SELECT ?,?,?,?,?,?,?,?,?,?,?,?,? FROM `oc_filecache` WHERE `storage` = ? AND `path_hash` = ? HAVING COUNT(*) = 0' with params [\\\"1\\\", \\\"2\\\", 1531663108, -1, \\\"5b4b53063c9c7\\\", 1531663108, 31, 36, \\\"\\\", \\\"979b5999fbe72a99dad2c65ed7dd33a3\\\", \\\"appdata_oc8mmywrs1zn\\\\\\\/css\\\\\\\/core\\\", \\\"core\\\", 2, 2, \\\"979b5999fbe72a99dad2c65ed7dd33a3\\\"]:\\n\\nSQLSTATE[23000]: Integrity constraint violation: 1062 Duplicate entry '2-979b5999fbe72a99dad2c65ed7dd33a3' for key 'fs_storage_path_hash'\",\"Code\":0,\"Trace\":\"#0 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/doctrine\\\/dbal\\\/lib\\\/Doctrine\\\/DBAL\\\/DBALException.php(128): Doctrine\\\\DBAL\\\\Driver\\\\AbstractMySQLDriver->convertException('An exception oc...', Object(Doctrine\\\\DBAL\\\\Driver\\\\PDOException))\\n#1 \\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/doctrine\\\/dbal\\\/lib\\\/Doctrine\\\/DBAL\\\/Connection.php(1015): Doctrine\\\\DBAL\\\\DBALException::driverExceptionDuringQuery(Object(Doctrine\\\\DBAL\\\\Driver\\\\PDOMySql\\\\Driver), Object(Doctrine\\\\DBAL\\\\Driver\\\\PDOException), 'INSERT INTO `oc...', Array)\\n#2 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/DB\\\/Connection.php(216): Doctrine\\\\DBAL\\\\Connection->executeUpdate('INSERT INTO `oc...', Array, Array)\\n#3 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/DB\\\/Adapter.php(114): OC\\\\DB\\\\Connection->executeUpdate('INSERT INTO `oc...', Array)\\n#4 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/DB\\\/Connection.php(254): OC\\\\DB\\\\Adapter->insertIfNotExist('*PREFIX*filecac...', Array, Array)\\n#5 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Cache\\\/Cache.php(273): OC\\\\DB\\\\Connection->insertIfNotExist('*PREFIX*filecac...', Array, Array)\\n#6 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Cache\\\/Cache.php(230): OC\\\\Files\\\\Cache\\\\Cache->insert('appdata_oc8mmyw...', Array)\\n#7 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Cache\\\/Scanner.php(292): OC\\\\Files\\\\Cache\\\\Cache->put('appdata_oc8mmyw...', Array)\\n#8 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Cache\\\/Scanner.php(221): OC\\\\Files\\\\Cache\\\\Scanner->addToCache('appdata_oc8mmyw...', Array, -1)\\n#9 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Cache\\\/Scanner.php(336): OC\\\\Files\\\\Cache\\\\Scanner->scanFile('appdata_oc8mmyw...', 3, 36, false, true)\\n#10 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/View.php(1326): OC\\\\Files\\\\Cache\\\\Scanner->scan('appdata_oc8mmyw...', false)\\n#11 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/View.php(1369): OC\\\\Files\\\\View->getCacheEntry(Object(OCA\\\\Files_Trashbin\\\\Storage), 'appdata_oc8mmyw...', '\\\/appdata_oc8mmy...')\\n#12 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/Root.php(198): OC\\\\Files\\\\View->getFileInfo('\\\/appdata_oc8mmy...')\\n#13 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/Node\\\/Folder.php(133): OC\\\\Files\\\\Node\\\\Root->get('\\\/appdata_oc8mmy...')\\n#14 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Files\\\/AppData\\\/AppData.php(105): OC\\\\Files\\\\Node\\\\Folder->get('core')\\n#15 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/SCSSCacher.php(116): OC\\\\Files\\\\AppData\\\\AppData->getFolder('core')\\n#16 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/CSSResourceLocator.php(109): OC\\\\Template\\\\SCSSCacher->process('\\\/var\\\/www\\\/nextcl...', 'core\\\/css\\\/jquery...', 'core')\\n#17 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/CSSResourceLocator.php(61): OC\\\\Template\\\\CSSResourceLocator->cacheAndAppendScssIfExist('\\\/var\\\/www\\\/nextcl...', 'core\\\/css\\\/jquery...')\\n#18 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/Template\\\/ResourceLocator.php(78): OC\\\\Template\\\\CSSResourceLocator->doFind('css\\\/jquery-ui-f...')\\n#19 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/TemplateLayout.php(270): OC\\\\Template\\\\ResourceLocator->find(Array)\\n#20 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/TemplateLayout.php(186): OC\\\\TemplateLayout::findStylesheetFiles(Array)\\n#21 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/private\\\/legacy\\\/template.php(207): OC\\\\TemplateLayout->__construct('user', 'files')\\n#22  [... cut ...] andler), Array)\\n#28 \\\/var\\\/www\\\/nextcloud\\\/lib\\\/base.php(999): OC\\\\Route\\\\Router->match('\\\/apps\\\/files\\\/')\\n#29 \\\/var\\\/www\\\/nextcloud\\\/index.php(42): OC::handleRequest()\\n#30 {main}\",\"File\":\"\\\/var\\\/www\\\/nextcloud\\\/3rdparty\\\/doctrine\\\/dbal\\\/lib\\\/Doctrine\\\/DBAL\\\/Driver\\\/AbstractMySQLDriver.php\",\"Line\":66}","userAgent":"Mozilla\/5.0 (Windows NT 10.0; Win64; x64; rv:61.0) Gecko\/20100101 Firefox\/61.0","version":"13.0.4.0"}

```
</details>

pehebo, 2018-07-16:
SAme problem here after upgrading from 13.02 to 13.04: mariadb and php7.0-fpm down after reboot. Manual starting succeeded.

miodzicho, 2018-08-02:
second to this issues, I just noticed today. After reboot mysql and php was stopped. Also hardware issue ?

nachoparker, 2018-08-02:
most likely