nextcloud/nextcloudpi #564: rsync backup from ncp panel using non-standard ssh port

burgerkingGH, 2018-07-15:
Hello and thanks for NextcloudPlus.

I am using ncp on Raspberry Pi servers at 3 different locations. In case a house burns down, I would like the data to be backed up at the different sites. The servers are configured to accept ssh connections on non-standard ports to reduce the number of brute-force attacks (fail2ban works great too). 

With the non-standard port, is it possible to use the web panel / nextcloudpi-config (specifically nc-rsync, nc-rsync-auto) to manage the backups? I can't figure out the syntax to include a non-standard port.

I know how to do this using just rsync (see [here](https://stackoverflow.com/questions/4549945/is-it-possible-to-specify-a-different-ssh-port-when-using-rsync)), but it would be great to piggy-back off the panel, and not have to set up a cron job, etc.

Thank you,

BK

nachoparker, 2018-07-15:
hi, well that is easy to add. Do you feel brave to send me a PR?

https://github.com/nextcloud/nextcloudpi/wiki/ncp-app-developer-guide

nachoparker, 2018-07-16:
great! let's start by adding an option to set the port and we can add more stuff later

burgerkingGH, 2018-07-29:
**Successes:**

I think adding a port field just involves adding the following code:

`PORT_=22`

and modifying the rsync line to read:

`rsync -aAx -e "ssh -p $PORT_" --delete "$DATADIR"/ "$DESTINATION_"`

**Issues:**

If the destination ssh ports are password or key protected (which they are), I get the error:

Host key verification failed

when using the correct IP and port. So from the command line, after I type in the password, this would work:

`ssh -p 2222 pi@DESTIP`

To solve this, I believe I would need either to 
A) store the plaintext password in a file, and feed it to the ssh call
B) set up and use ssh keys without a password (probably a better solution) 

I assume this is what you mean by:

`INFO="'user' needs SSH autologin from the NCP 'root' user at 'ip' if we are launching from ncp-web"`

However, to set up ssh keys, I would need to ssh into the docker container, and I cannot figure out how to do that (I am a docker newbie). Every time I try to enable ssh in the panel, weirdly it re-prompts for the panel user/pass, and refuses to log me in with the correct credentials. I haven't figured out how to do it from the command line either.

So I am at an impasse. This is probably the wrong place to ask, but how does one gain ssh access to the command line of the docker container that runs ncp?

Thanks.

nachoparker, 2018-10-24:
> I assume this is what you mean by

that's right

The idea is to rsync/ssh **from** the container to another machine. You don't need the SSH daemon running on the container

Feeling brave about sending a PR?

xelcho, 2019-01-27:
May I add a slight twist to this thread?

If so I would like to add:
1: since ncp version 1.x the nc-rsync.sh has moved from /usr/local/bin/ncp to /usr/local/bin/ncp/BACKUPS

2: I am trying to alter the nc-auto-rsync.sh script to:
2.1: Run etherwake {MAC Add of Backup} and sleep 5s prior to running rsync with the custom port config. This is my first unsuccessful attempt:
${SSH[@]} : || { echo "SSH non-interactive not properly configured"; return 1; }
  }
  etherwake {mac addr}
  sleep 5s
  echo "0  5  */${SYNCDAYS}  *  *  root  /usr/bin/rsync -aAx -e 'ssh -p 5150' --delete \"$DATADIR\" \"$DESTINATION\"" > /etc/cron.d/ncp-rsync-auto
  chmod 644 /etc/cron.d/ncp-rsync-auto
  service cron restart

It errors out on ssh port 22. :(

Please advise.

danielroehrig, 2019-02-05:
I have added a field that asks for a port number. What I am unsure about, is if that might affect the behaviour when someone wants to rsync to a local folder. #793

nachoparker, 2019-02-06:
Many thanks @danielroehrig

closed 2019-02-06 by nachoparker