nextcloud/nextcloudpi #565: ncp-config programs fail to match passwords when they have spaces

JVAQUEROM, 2018-07-16:
I have realized that the programs in ncp-config that change your password (ssh, nc-password,..) they fail when your password uses spaces. It takes the first "word" as first input and the second as the "repeat password". So it fails.

mschilli87, 2018-07-16:
@JVAQUEROM: Seems to me like escaping the spaces as `\ ` or quoting the whole password in `'`...`'` should fix the issue.

JVAQUEROM, 2018-07-16:
I tried using double quotes `"..."` and that didnt work. I'll try the other

nachoparker, 2018-07-16:
hi, I am assuming you are doing this from the terminal, with `ncp-config` right? because ncp-web wouldn't even let you try to use spaces.

JVAQUEROM, 2018-07-16:
yes, from the terminal. By the way, the other options do not work either.

nachoparker, 2018-07-17:
what other options?

mschilli87, 2018-07-17:
@nachoparker: I guess escaping via `\ ` or `'`...`'` instead of `"`..`"`:

@JVAQUEROM 
> I tried using double quotes "..." and that didnt work. I'll try the other

nachoparker, 2018-07-17:
well, I'll see if I can correct this behaviour without need to escape anything

btw, thanks for reporting!!

JVAQUEROM, 2018-07-19:
I thought I had answered. The escaping via `\` didn't work and the `'...'` didn't either, as @mschilli87

nachoparker, 2018-07-20:
I just looked at this, and the way things are coded now, spaces cannot be supported. I have added a check though for spaces as an invalid character.

Thank you for the feedback!

closed 2018-07-20 by nachoparker