nextcloud/nextcloudpi #566: Cannot Load Zend OPcache - its already loaded

crossr1, 2018-07-17:
I have noticed with the release of v0.57.18 that I am seeing this error in multiple locations:

Cannot load Zend OPcache - it was already loaded

Thanks

chiyi4488, 2018-07-19:
I also have this problem when I run occ, but this does not affect my work.

Iolaum, 2018-07-19:
This is a duplicate of #416 
It was then fixed from @nachoparker with [this change](https://github.com/nextcloud/nextcloudpi/commit/4bdbf933ed0b809466e2f9e8b2552c40293d1ecd#diff-25e6ad8ebc5d26ba06d2bcb0b220149c)

And yes it is harmless- but it does fill /var/mail/www-data mailbox :o

Iolaum, 2018-07-22:
I removed the two relevant files (renamed one to keep as backup) but I am still getting the warning when I run `ncp-report`:

```
# pwd
/etc/php/7.0/fpm/conf.d
# ls -la | grep cache
lrwxrwxrwx 1 root root   39 Dec  4  2017 20-opcache.ini.bk -> /etc/php/7.0/mods-available/opcache.ini
```
Maybe I misunderstood the fix?

nachoparker, 2018-07-22:
maybe restart the service?

Iolaum, 2018-07-22:
Instead of restarting the services I reboot the ncp - Is that not achieving the same thing (apart from extra wasted computation)?

nachoparker, 2018-07-22:
yes, should be the same thing

Iolaum, 2018-07-22:
I managed to misconfigure opcache by removing both symlinks (#571) but this warning persisted (e.g. when typing `sudo ncp-report`)! So maybe there's another cause?

nachoparker, 2018-07-22:
maybe you forgot the `/etc/php/7.0/cli/conf.d/` folder?

Iolaum, 2018-07-22:
Correct!! dumb me :(

nachoparker, 2018-07-22:
awesome! this is an easy fix. Do you want to send a PR to devel?

Iolaum, 2018-07-22:
Yep, I ll send one. It's basically your code that I referenced earlier.

commit 681a5385 (2018-07-23): Fixes #566 Remove redundant opcache configuration (#572)
commit 214c187d (2018-07-23): Fixes #566 Remove redundant opcache configuration (#572)
closed 2018-07-23 by nachoparker